mne-tools/mne-python #8149: Custom montage object not loading to NIRS info

Isamaoz, 2020-08-23:
Hello,

I am trying to classify EEG and NIRS data using MNE. The datasets I am using were converted into MATLAB format; therefore, I created my data structures from scratch (raw data, montage files, epochs). I did not have any problems loading and classifying the EEG data, but I realised that raw.set_montage() does not support NIRS channels. When I tried to set the montage file to the info object, the coordinates of the sources and detectors are converted into NaNs. 

Do you know if there is an alternative for me to set a montage using NIRS?

agramfort, 2020-08-24:
ping @rob-luke

rob-luke, 2020-08-25:
Hi @Isamaoz,

Can you provide a minimum working example of how you are setting the montage?

I haven't added added NIRS support to set_montage(), so it doesn't surprise me it doesn't work. Taking a quick look at the code I think the following line is just skipping the nirs channels. But the whole code seems quite EEG specific.

https://github.com/mne-tools/mne-python/blob/master/mne/channels/montage.py#L700

I won't be able to add support for this before 0.21. @Isamaoz are you able to take make an attempt at a PR to fix this issue?

HanBnrd, 2021-04-23:
Hi @larsoner, this issue also is closed by #9141 🙂

larsoner, 2021-04-23:
Indeed, thanks!

closed 2021-04-23 by larsoner