mne-tools/mne-python #8154: BUG: Example uses private function

larsoner, 2020-08-24:
cc @GuillaumeFavelier, can you work out some public API to avoid the `_get_renderer` here:

https://mne.tools/dev/auto_tutorials/source-modeling/plot_background_freesurfer_mne.html#cortical-alignment-spherical

GuillaumeFavelier, 2020-08-24:
I would say that the renderer itself is not intended for public use xD 

The closest alternative to `_get_renderer()` is `_create_3d_figure()`.

I'll see what I can do about that tomorrow :)

larsoner, 2020-08-24:
Maybe we should have a `.. warn` saying the API is not necessarily stable. But we need it at least for our own internal use in examples:

https://github.com/mne-tools/mne-python/pull/7087#issuecomment-679312067

We could even say that's it explicit purpose (and to facilitate our transition), and users writing should probably make use of preferably `pyvista` functions directly.

GuillaumeFavelier, 2020-08-27:
What do you say we add a parameter to `create_3d_figure()` to return an instance of `_Renderer` instead of the figure object?

Creating a new public function just to expose `_get_renderer()` seems to me like just duplicating code while we already have `create_3d_figure()` that we can reuse.

larsoner, 2020-08-27:
Sounds reasonable to me

closed 2020-08-27 by GuillaumeFavelier