mne-tools/mne-python #8159: Replace Travis with Azure or GitHub Actions

cbrnr, 2020-08-25:
Travis seems to be very slow especially with our macOS tests, to the point where the job routinely times out after 50 minutes with only 46% of all tests run so far. I suggest that we abandon Travis and use Azure or GitHub actions instead (somehow I'd prefer GH actions because this seems to be the go-to solution, but I have never actually tried it).

larsoner, 2020-08-25:
We can work from SciPy's macOS GitHub action

https://github.com/scipy/scipy/blob/master/.github/workflows/testing.yml

This combined with our code for Travis would make for a nice, hopefully not so painful CI PR if someone wants to learn more about the process

The one problem with the GitHub action IIRC is that it won't kick in until merged with `master`, so for testing whoever implements it will have to enable it for their fork and iterate/monitor progress there

cbrnr, 2020-08-25:
I probably won't have time to work on this in the next weeks (so if you want to take a stab please go ahead). Did I understand you correctly that the CI won't kick in in a normal PR? This can't be true, because browsing the SciPy PRs shows that the macOS job is run...

larsoner, 2020-08-25:
> Did I understand you correctly that the CI won't kick in in a normal PR? This can't be true, because browsing the SciPy PRs shows that the macOS job is run...

The PR to add the GitHub action itself will not have the action run. Once the changes are in `master`, each PR runs the GitHub actions in master -- or at least this is how it was when I added circleci-redirector-action. This is different from Travis/AppVeyor/Azure/CircleCI, where the config YAML in a given PR is actually used in execution. So it makes iterating on CIs a little bit harder / more annoying, but once things work and are in `master`, it otherwise has fine behavior.

GuillaumeFavelier, 2020-08-25:
I would happy to join the effort on this :)

cbrnr, 2020-08-25:
Ah, got it.

larsoner, 2020-08-25:
@GuillaumeFavelier feel free to take a stab at it !

cbrnr, 2020-08-25:
Would you replace only the Travis macOS job or all Travis jobs? I'd prefer abandoning Travis completely because it has been the major bottleneck, and it's probably better to stick with fewer providers.

GuillaumeFavelier, 2020-08-25:
> I'd prefer abandoning Travis completely 

That's what I had in mind

larsoner, 2020-08-25:
> it's probably better to stick with fewer providers.

On the other hand, the more we spread across providers the less often we should hit bottlenecks. So I don't mind spreading across them.

Let's just migrate macOS for now. Travis has been great for Linux for years and GitHub actions are newer. If over the next few months / year time frame we have no problems with the GitHub action maybe we should think about migrating everything then.

cbrnr, 2020-08-25:
It's certainly less effort now, but we're already using Azure, which is very similar to GH Actions (same company), so we might as well try to move all Travis jobs to Azure (and then migrating to GHA will be almost painless once/if we decide to do so).

larsoner, 2020-08-25:
To me it's not about reducing effort, but rather about accumulating experience with the service. I prefer an incremental approach so we get some experience and see how GitHub actions are in practice (reliability, interface, how easy it is to restart, see results, etc.). For example it's more of a pain to iterate on GitHub actions on PRs, so I'm not sure it's worth moving them all over anyway. I'm still -1 on doing them all now. Let's do macOS because it's actually causing problems for us and leave Linux because it has been reliable and we know it works well (and what doesn't).

cbrnr, 2020-08-25:
I agree, I was referring to Azure Pipelines instead of GitHub Actions though.

closed 2020-08-27 by larsoner