mne-tools/mne-python #8171: plotting crashes for than 1 channel

chapochn, 2020-08-28:
#### Describe the bug
plotting of multiple channels does not work

#### Steps to reproduce
    import numpy as np
    import mne
    data = np.arange(1000).reshape(2, -1)
    info = mne.create_info(ch_names=['a', 'b'], sfreq=1000)
    raw = mne.io.RawArray(data, info)
    raw.plot(n_channels=2)
    #  raw.plot(n_channels=1)  # this works

#### Expected results
get a plot

#### Actual results
Traceback (most recent call last):
  File "<input>", line 8, in <module>
  File "/Users/.../opt/anaconda3/envs/mne/lib/python3.8/site-packages/mne/io/base.py", line 1372, in plot
    return plot_raw(self, events, duration, start, n_channels, bgcolor,
  File "<decorator-gen-138>", line 21, in plot_raw
  File "/Users/.../opt/anaconda3/envs/mne/lib/python3.8/site-packages/mne/viz/raw.py", line 414, in plot_raw
    _prepare_mne_browse_raw(params, title, bgcolor, color, bad_color, inds,
  File "/Users/.../opt/anaconda3/envs/mne/lib/python3.8/site-packages/mne/viz/raw.py", line 722, in _prepare_mne_browse_raw
    ax.set_yticklabels(['X' * max([len(ch) for ch in info['ch_names']])])
  File "/Users/.../opt/anaconda3/envs/mne/lib/python3.8/site-packages/matplotlib/axes/_base.py", line 63, in wrapper
    return get_method(self)(*args, **kwargs)
  File "/Users/.../opt/anaconda3/envs/mne/lib/python3.8/site-packages/matplotlib/cbook/deprecation.py", line 451, in wrapper
    return func(*args, **kwargs)
  File "/Users/.../opt/anaconda3/envs/mne/lib/python3.8/site-packages/matplotlib/axis.py", line 1793, in _set_ticklabels
    return self.set_ticklabels(labels, minor=minor, **kwargs)
  File "/Users/.../opt/anaconda3/envs/mne/lib/python3.8/site-packages/matplotlib/axis.py", line 1714, in set_ticklabels
    raise ValueError(
ValueError: The number of FixedLocator locations (2), usually from a call to set_ticks, does not match the number of ticklabels (1).

#### Additional information
Platform:      macOS-10.14.6-x86_64-i386-64bit

cbrnr, 2020-08-28:
It looks like this has already been fixed with #8031 - it's already in the latest master and will be part of 0.21.

closed 2020-08-28 by cbrnr

chapochn, 2020-08-31:
is there a workaround in the meantime? can I downgrade something to make it work?

larsoner, 2020-09-01:
`pip install https://github.com/mne-tools/mne-python/archive/master.zip` or wait 2 weeks for 0.21 to release then `pip install --upgrade mne`

larsoner, 2020-09-04:
https://pypi.org/project/mne/0.20.8/

`pip install --upgrade mne` should get it installed!