mne-tools/mne-python #8172: find_bads_ecg module picks non ecg artifact

balandongiv, 2020-08-29:
#### Describe the bug

The objective is to use `find_bads_ecg`  get ICs that matches to ECG pattern.

However, it seems the module return non ecg artifact regardless of using either the `Cross-trial phase statistics`  and `Pearson correlation` 

For example, using the `Pearson correlation` , it was suggested the IComponent of 15, 46, 52.

However,`plot_sources(epochs)` the three component shows that the series does not look like a typical  ECG pattern as illustrated in this [figure](https://imgur.com/gallery/RswDHxh).

Also, the topomaps of these 3 IC also reveal it is not an ECG like pattern.

I am not sure whether this issue can defined as bug or not. 

#### Steps to reproduce

#### Expected results
```
import mne
from autoreject import Ransac, get_rejection_threshold  # noqa
from autoreject.utils import interpolate_bads  # noqa
import matplotlib
from autoreject.utils import set_matplotlib_defaults  # noqa
import matplotlib.pyplot as plt  # noqa
import os.path
from pathlib import Path
set_matplotlib_defaults(plt)
matplotlib.use('TkAgg')

cur_path = os.path.dirname(__file__)
data_folder = Path("../data/raw_S17_3_ica_fit.fif")
path = os.path.relpath(data_folder, cur_path)
new_ica = mne.preprocessing.read_ica(path)

data_folder_epoch = Path("../data/raw_S17_3_epoch_drop_bad.fif")
path_bad = os.path.relpath(data_folder_epoch, cur_path)
epochs = mne.read_epochs(path_bad)

#find which ICs match the ECG pattern
#{'ctps', 'correlation'}
ecg_indices, ecg_scores = new_ica.find_bads_ecg(epochs, method='correlation',
                                                threshold='auto')
new_ica.exclude = ecg_indices

#barplot of ICA component "ECG match" scores
new_ica.plot_scores(ecg_scores)

#plot diagnostics
new_ica.plot_properties(epochs, picks=ecg_indices)

#plot ICs applied to raw data, with ECG matches highlighted
new_ica.plot_sources(epochs)
```

#### Additional information

The files  `raw_S17_3_ica_fit.fif` and `raw_S17_3_epoch_drop_bad.fif` are accessible via the [link](https://drive.google.com/drive/folders/1csZ3jrgvmh_5MwC_v1EtLFQLTUnvjZQ8?usp=sharing):

agramfort, 2020-08-30:
I don't see a clear ECG component in your data.

balandongiv, 2020-08-30:
This is rather interesting and lead to more questions than answer.

In that case,  it is not advisable to simply accept the proposed IC outputted by the module. Atleast for this data, eventhought the system suggested 3 IC related to ECG, but in reality there are no clear ecg pattern as validated by the human expert.

drammock, 2020-08-30:
The human expert should probably view the raw data before performing ICA decomposition, and if there are no appreciable ECG artifacts, skip the `find_bads_ecg` step. As @dengemann often says, "always look at your artifacts before trying to remove them. They might be so small that you needn't bother."

closed 2020-08-30 by drammock