mne-tools/mne-python #8173: Reading in Peryst (.lay/.dat) files into Raw

adam2392, 2020-08-30:
#### Describe the problem
I have some files that were output from the Persyst system, which are text-readable .lay/.dat file combination. I would like to read them into MNE Python Raw.

This would be useful since I think it's a major "EEG" software processing system, which many clinical centers have. In addition, it is text-readable, so nothing hard about it. 

#### Describe your solution
A `read_raw_persyst` function that takes in a `lay_fname` and reads in data into a `Raw` object. The `.dat` filename is usually encoded in the `lay_fname`. 

The first solution won't be "perfect", but some more iterations from Persyst users might be warranted.

#### Describe possible alternatives
There is an initial start I found on Github: https://github.com/epiSurg/python-layread/blob/master/layread.py

but they do not read it into a `Raw` object, which I think would be more desirable for robust data flow.

#### Additional context
I have some example files added here.
[Archive.zip](https://github.com/mne-tools/mne-python/files/5146962/Archive.zip)

I can help implement it if perhaps @agramfort  or @larsoner can point me in the right direction? I'm assuming it's just that one function `read_raw_persyst`.

agramfort, 2020-08-30:
seems indeed easy to add.

you should add these testing files in the mne-testing-data repo and then
add the reader in PR to mne-python

you can look at brainvision or nicolet readers for inspiration

>

larsoner, 2020-08-31:
I would look at https://github.com/mne-tools/mne-python/pull/6674 to see the list of files you might need to change. If there are no new channel types, you probably only need to change half as many (or even fewer).

PR #8176 (2020-09-06): [MRG, IO] Persyst reader into Raw object
#### Reference issue
Closes: #8173 

#### What does this implement/fix?
Adds a Persyst reader for .lay + .dat file combinations and unit tests.

Note:
- birthdate and measurement date is not a strict adherence sometimes when output by Persyst. I am not sure why... but things like `16-Nov-199` obviously probably stands for nov 16th, 1992 based on the data, but not sure why it gets output funky. I'm assuming these things can change with Persyst versions without our knowing.

With that being said though, the data format and such seems like it shouldn't cha [... cut]
closed 2020-09-06 by larsoner