mne-tools/mne-python #8183: Topomaps interactive mode - Ticks disappear when zooming in

LeyloJohnny, 2020-09-01:
Dear all, 

during plotting of topomaps in the interactive mode, the ticks on the colorbar disappear when zooming in.

Codes:
```
bandAlpha1 = [
    [1.36144854e-05],
    [1.01637179e-05],
    [1.18642372e-05],
    [1.41590759e-05],
    [1.52691734e-05],
    [1.33787913e-05],
    [1.77864235e-05],
    [1.44647350e-05],
    [2.74915514e-05],
    [3.23384006e-05],
    [2.01845999e-05],
    [7.55948936e-06],
    [9.03805553e-06],
    [5.27009746e-06],
    [8.32830484e-06],
    [8.43511259e-06],
    [1.64113800e-05],
    [1.82010771e-05],
    [2.04712873e-05],
    [6.43561709e-06]]
bandHighBeta = [
    [9.48504150e-05],
    [5.74889957e-05],
    [9.19252640e-05],
    [8.93399030e-05],
    [4.18493759e-05],
    [3.26190943e-05],
    [2.76993552e-05],
    [1.18445736e-05],
    [1.24432218e-05],
    [2.84315098e-05],
    [7.61318171e-05],
    [4.71576660e-05],
    [3.18943913e-05],
    [2.00989813e-05],
    [1.72555372e-05],
    [2.70839355e-05],
    [5.72236646e-05],
    [3.24518943e-05],
    [1.88565039e-05],
    [3.15454230e-05]]
bandGamma = [
    [1.25188644e-06],
    [1.11386832e-06],
    [8.60886848e-07],
    [8.30455560e-07],
    [6.95102232e-07],
    [5.05349294e-07],
    [5.19661342e-07],
    [4.49833399e-07],
    [5.48951741e-07],
    [6.86494014e-07],
    [8.64004756e-07],
    [1.54017029e-06],
    [3.24345057e-06],
    [1.81364017e-06],
    [6.71151020e-07],
    [8.67693066e-07],
    [9.21449764e-07],
    [7.49581188e-07],
    [5.50885141e-07],
    [4.22944226e-07]]

import mne
import matplotlib.pyplot as plt
import numpy as np
freBandNames = ['bandAlpha1', 'bandHighBeta', 'bandGamma']

import matplotlib
print(matplotlib.__version__)
#print(bandDelta+bandHighBeta+bandGamma)
voltageFactor = 1000000
toposcalar = [ min(bandAlpha1+bandHighBeta+bandGamma)[0]*voltageFactor, max(bandAlpha1+bandHighBeta+bandGamma)[0]*voltageFactor ] #calculate topomap scalar
print(toposcalar[0],toposcalar[1])

ch_names_1020sys = ['Fp1', 'Fp2','F3','F4', 'C3', 'C4', 'P3', 'P4', 'O1', 'O2', 'F7', 'F8', 'T3', 'T4', 'T5', 'T6', 'Fz', 'Cz', 'Pz', 'A2']
info = mne.create_info(ch_names=ch_names_1020sys, sfreq=256, ch_types='eeg')

evoked = []
evoked.append( mne.EvokedArray(bandAlpha1, info) )
evoked.append( mne.EvokedArray(bandHighBeta, info) )
evoked.append( mne.EvokedArray(bandGamma, info) )

ten_twenty_montage = mne.channels.make_standard_montage('standard_1020')

fig, ax = plt.subplots(1, 6, gridspec_kw=dict(width_ratios= 3*[12, 1]))

for i in range(len(evoked)):
    evoked[i].set_montage(ten_twenty_montage)

    evoked[i].plot_topomap(times=0, vmin = 0, vmax=toposcalar[1], time_unit='s',time_format=None, 
                                axes = ax[i*2:i*2+2], cmap='Spectral_r', colorbar = True, units = '$\mu V^2$', show=False)
    #evoked[i].plot_topomap(times=0, time_unit='s',time_format=None, 
    #                        axes = ax[i*2:i*2+2], cmap='Spectral_r', colorbar = True, units = '$\mu V^2$', show=False)
    
    # set title for this topomap
    ax[i*2].set_title(str(freBandNames[i]), fontsize=10)

# adjust subplot position
fig.subplots_adjust(top=0.9,bottom=0.1)
# give the figure a main title
plt.suptitle('Power Comparison')

evoked[2].plot_topomap(times=0, vmin=0, time_unit='s',time_format=None, cmap='Spectral_r', colorbar = True, units = '$\mu V^2$', show=False)
plt.show()
```

before zooming in:
[image]

after a bit zooming in:
[image]

after more zooming in:
[image]

Does anyone know how to fix this?

datalw, 2020-09-01:
I also have this problem, does anyone have the solution?

larsoner, 2020-09-01:
We need to make DraggableColorbar update the ticks when it's interacted with. Want to see if you can come up with something that works by playing with https://github.com/mne-tools/mne-python/blob/master/mne/viz/utils.py#L1981-L2052 ? It might just require changing the tick type from fixed (which it probably gets set to on creation) to automatic at the matplotlib end.

larsoner, 2020-09-02:
Fixed by #8197 if you want to try it

PR #8197 (2020-09-02): MRG, ENH: Add cov.plot_topomap
Closes #7384
Closes #8183

- Also adds support for `time_label=''` to avoid having the time printed.
- With a bit of squaring, we get correct units / scale factors / whitening compared to what's in `master`.
closed 2020-09-02 by larsoner