mne-tools/mne-python #8184: STY: Website update options

larsoner, 2020-09-01:
- [pydata-sphinx-theme](https://github.com/pandas-dev/pydata-sphinx-theme) uses Bootstrap 4
- [sphinx-multiversion](https://github.com/Holzhaus/sphinx-multiversion) can maybe save us a bit of version drop-down work

Something for someone to work on when they want a break from "doing science"

larsoner, 2020-09-01:
I played a little bit with each of these and neither is trivial

drammock, 2020-09-01:
+1 for switching (after 0.21).

sappelhoff, 2020-09-01:
what are the reasons to switch the website style?

drammock, 2020-09-01:
> what are the reasons to switch the website style?

1. the current theme doesn't support next/previous buttons on documentation pages by default.
2. current theme has issues with default scroll alignment (target anchors are often just off the top screen or at least obscured by the navigation bar / dev bar)
3. current theme is one person's project, and is not being developed very actively (~1 year since last commit).  Proposed new theme is used by Pandas, Bokeh, etc. and its last commit is yesterday, so I predict it to be maintained / upgraded for a long time.

Note that switching the theme doesn't necessarily mean that the site has to look *drastically* different.  We can still customize the styling to make it look more or less like the current site does (in terms of color palette, typeface, layout, etc).

larsoner, 2020-09-01:
>> what are the reasons to switch the website style?

> Note that switching the theme doesn't necessarily mean that the site has to look drastically different. We can still customize the styling to make it look more or less like the current site does (in terms of color palette, typeface, layout, etc).

To quote the Tim Hoffman on matplotlib gitter (not pinging his handle, doubt he cares about our decision process):

> Overall it’s not too bad. There are some aspects I don’t like as well, but that’s also true for our current theme. Having a similar theme across a number of libraries can be a usability plus (with some color/font variations). From tge maintainer pov, the advantage is a carefully crafted (I think) modern theme. IMHO the builtin sphinx themes (From which ours is forked) are not really modern anymore, neither concerning design, or navigation/usability, or technically (e.g. responsiveness).

I more or less agree with this. So I would say some theme that's recognizable as being "in the scientific Python ecosystem" but (maybe) with some minor deviations to give it some MNE "flair" might be worth considering.

larsoner, 2020-11-25:
See also where `mne-nirs` and @rob-luke changed/is changing: https://github.com/mne-tools/mne-nirs/pull/207, https://github.com/mne-tools/mne-nirs/pull/208

closed 2021-02-16 by larsoner