mne-tools/mne-python #8192: BUG: Arrows gone in mixed norm plotting

larsoner, 2020-09-01:
Glyph in glass brain in : 

https://mne.tools/dev/auto_examples/inverse/plot_time_frequency_mixed_norm_inverse.html#sphx-glr-auto-examples-inverse-plot-time-frequency-mixed-norm-inverse-py

do not show up. I should see 2 sources.

_Originally posted by @agramfort in https://github.com/mne-tools/mne-python/pull/8188#issuecomment-685065216_

larsoner, 2020-09-01:
This has been incorrect going back to 0.18:

https://mne.tools/0.18/auto_examples/inverse/plot_time_frequency_mixed_norm_inverse.html#sphx-glr-auto-examples-inverse-plot-time-frequency-mixed-norm-inverse-py

larsoner, 2020-09-01:
It was broken by 78561338b1ffe376455e8a1324f55e2833979ce1, I'll try to figure out what the problem is...

agramfort, 2020-09-01:
:'(

PR #8194 (2020-09-02): MRG, BUG: Fix sphere glyphs
Closes #8192

In this PR on mayavi:

[image]

pyvista:

[image]

these look different but I think it might just be a subtle angle/perspective difference (?), here's a different angle:

[image]
[image]

@GuillaumeFavelier can you look and see if the difference is meaningful and if necessary push commits to fix it? Also might be a good idea to check the code for any other instances of `if/elif/elif` and change them to `if/elif/else` with an `assert` like I did here (preferably with a user-friendly `_check_option` so that the `assert` is just an internal sanity check).
closed 2020-09-02 by larsoner