mne-tools/mne-python #8199: Issues with sensor coordinates and positions in mne coreg

justinhyon, 2020-09-02:
I have a few problems with setting the sensor coordinates and correctly visualizing them in mne coreg.

I am using the following code to set the sensor locations via a dictionary of coordinates:
```
dig_montage = make_dig_montage(ch_pos=coords, nasion=coords['1-NS'], lpa=coords['1-LE'], rpa=coords['1-RE'], hsp=None,
                               hpi=None, coord_frame='mri')

dig_montage.plot()
raw = mne.io.read_raw_bdf(input_fname='/Path/To/File/T1_Beep.bdf')
raw.load_data().pick_types(meg=False, eeg=True, stim=True)
raw.set_montage(dig_montage)
raw.plot_sensors(show_names=True)
raw.save('MD_raw.fif', overwrite=True)
```
This outputs the following plots:
[image]

The eeg sensors should cover the entire head, but I can't figure out how to adjust the fiducial points to make them do so. When I adjust the numbers, the sensor positions do not change as expected. For example, changing the x coordinate of the left ear by 20 mm moves the sensor locations a certain amount, but changing it an additional 50 results in a smaller change. I am wondering if I might be thinking about the coordinates wrong? The fiducials I have now are:
```    
    'NS': [74.000000, 24.000000, 121.000000],
    'LE': [159.000000, 121.000000, 96.000000],
    'RE': [12.000000, 135.000000, 97.000000]
```

My second issue is with the sensor positions after opening the saved Raw data in mne coreg. The sensors should be arranged in a circular cap on top of the head, but they appear to be automatically redistributing across the head surface.
[image]
[image]
I was wondering how I might be able to make the sensors remain in their original circular orientation?

Thank you!

agramfort, 2020-09-02:
can you share the files to look at your montage?

>

justinhyon, 2020-09-04:
> can you share the files to look at your montage?
> […](#)

Here is the MD_raw.fif file:
[https://drive.google.com/file/d/1IKUNkSpoccbJHunJCeuY3zAqutLTusAa/view?usp=sharing](url)
Should I also send the mri files?

agramfort, 2020-09-05:
it appears you have a unit pb in your input montage file:

```
In [5]: raw.info['dig']
Out[5]:
[<DigPoint |        LPA : (-75588.2, 0.0, 0.0) mm   : head frame>,
 <DigPoint |     Nasion : (-0.0, 107449.6, 0.0) mm  : head frame>,
 <DigPoint |        RPA : (72080.3, -0.0, 0.0) mm   : head frame>,
 <DigPoint |     EEG #1 : (6244.6, 4217.6, 149764.8) mm : head frame>,
 <DigPoint |     EEG #2 : (-48713.7, -78170.0, 56994.5) mm : head frame>,
 <DigPoint |     EEG #3 : (-45339.7, -79639.8, 40079.3) mm : head frame>,
 <DigPoint |     EEG #4 : (-43836.9, -79353.1, 21348.3) mm : head frame>,
 <DigPoint |     EEG #5 : (-17799.1, -93603.2, 25984.0) mm : head frame>,
 <DigPoint |     EEG #6 : (-17809.9, -96583.6, 43190.1) mm : head frame>,
 <DigPoint |     EEG #7 : (-20661.5, -95553.8, 59965.5) mm : head frame>,
 <DigPoint |     EEG #8 : (-22537.0, -91882.0, 75758.1) mm : head frame>,
 <DigPoint |     EEG #9 : (-21817.0, -82563.0, 94741.9) mm : head frame>,
 <DigPoint |    EEG #10 : (-27519.7, -67182.0, 108815.9) mm : head frame>,
 <DigPoint |    EEG #11 : (9494.4, -60536.5, 121464.5) mm : head frame>,
 <DigPoint |    EEG #12 : (6892.5, -12859.8, 148931.0) mm : head frame>,
 <DigPoint |    EEG #13 : (7713.8, -74282.0, 107419.0) mm : head frame>,
 <DigPoint |    EEG #14 : (6763.5, -86177.1, 93481.3) mm : head frame>,
 <DigPoint |    EEG #15 : (7625.2, -94134.9, 76710.3) mm : head frame>,
 <DigPoint |    EEG #16 : (6454.8, -98395.4, 59189.9) mm : head frame>,
 <DigPoint |    EEG #17 : (7591.5, -98078.2, 42072.8) mm : head frame>,
```

I suspect you passed locations in mm and not meters.

justinhyon, 2020-09-09:
Thank you, this was the issue! I have now run into another problem, however. When using stc.plot to plot the source estimate, as described in https://mne.tools/stable/auto_tutorials/source-modeling/plot_eeg_mri_coords.html#sphx-glr-auto-tutorials-source-modeling-plot-eeg-mri-coords-py , the plot disappears as soon as it finishes plotting. The window pops up and everything that should be there is plotted, but it instantly closes as soon as it finishes plotting. This happens also when I use the exact example code above. The program does not finish. How could I make this plot stay open?

agramfort, 2020-09-09:
you should run your code form an open interpreter to keep a command line
prompt when plots are open

>

justinhyon, 2020-09-09:
I am currently running in pycharm, and all other plots, both matplotlib and mayavi, seem to be working except stc.plot. Should I use something else?

agramfort, 2020-09-09:
so it's a pyvista issue? can you check is a simple pyvista example?

larsoner, 2020-09-09:
you might need to do `brain = stc.plot(...)`. It's possible your instance is getting garbage collected immediately

justinhyon, 2020-09-09:
brain = stc.plot worked! thank you both

closed 2020-09-09 by agramfort