mne-tools/mne-python #8203: something wrong in raw.py

alienx-zero, 2020-09-02:
# -*- coding: utf-8 -*-
"""
Created on Sat Aug 29 21:50:53 2020

@author: 99488
"""

import os
import numpy as np
import mne

fname = r'E:\OPT\research\EEG\chb-mit-scalp-eeg-database-1.0.0\chb01\chb01_01.edf'

sample_data_folder = mne.datasets.sample.data_path()
sample_data_raw_file = os.path.join(sample_data_folder, 'MEG', 'sample',
                                    'sample_audvis_filt-0-40_raw.fif')
raw = mne.io.read_raw_fif(sample_data_raw_file)

print(raw)
print(raw.info)

raw.plot_psd(fmax=50)
raw.plot(duration=5, n_channels=30)

I am following the tutorials https://mne.tools/stable/auto_tutorials/intro/plot_10_overview.html#sphx-glr-auto-tutorials-intro-plot-10-overview-py. And in the line _raw.plot(duration=5, n_channels=30)_, console reported that ValueError: The number of FixedLocator locations (10), usually from a call to set_ticks, does not match the number of ticklabels (1). 

And the main problem is from the sentence ax.set_yticklabels(['X' * max([len(ch) for ch in info['ch_names']])]) in raw.py. The number of yticklabels did not match the data dimension. I hope it could be fixed in future~

cbrnr, 2020-09-02:
This has already been fixed in #7811, but it is not available in a release yet. I thought about that problem a couple of minutes ago (#8031), and I think we should really backport the fix and make a 0.20.8 hotfix release even though 0.21 is about to be released soon. @larsoner and @agramfort WDYT?

alienx-zero, 2020-09-03:
Thx, update to 0.21 manually fix it

closed 2020-09-03 by alienx-zero

larsoner, 2020-09-03:
I am not sure how trivial it will be to backport the fix, you want to give it a try @cbrnr ?

larsoner, 2020-09-04:
Opened in #8212, let's see how many failures we get

larsoner, 2020-09-04:
https://pypi.org/project/mne/0.20.8/

`pip install --upgrade mne` should get it installed!