mne-tools/mne-python #8204: Should mne.utils._check_option return the checked value?

cbrnr, 2020-09-02:
Currently, `mne.utils._check_option` returns `True` if a value matches a list of allowed values or else raises a `ValueError`. I'm often using it as follows:

```python
_check_option('component_order', component_order, ['alternate', 'mutual_info'])
self.component_order = component_order
```

It would be useful if `_check_option` returned the value instead of `True`, because this would shorten my example to:

```python
self.component_order = _check_option('component_order', component_order, ['alternate', 'mutual_info'])
```

I don't really see a problem with returning the value, but maybe others rely on the boolean return value (although a quick search didn't turn up any assignment to the return value of this function).

drammock, 2020-09-02:
seems like a good idea to me.  I'd say that since `_check_option` is so widely used, that it might be better to wait until after 0.21 release to make that change though?  It *seems* like a straightforward changeset, but better not to risk a bunch of frustrated users of "stable" who just updated and had their scripts break.

larsoner, 2020-09-03:
```
larsoner@bunk:~/python/mne-python$ git grep "= _check_option"
larsoner@bunk:~/python/mne-python$ git grep ".*=.*_check_option.*"
larsoner@bunk:~/python/mne-python$ 
```
I don't think we have any assignments in our code base so far. To escape these regex it would have to be `a = \ \n whatever` or so and I *really* doubt we've done that. So I think it's safe enough to change, but if it's not holding up any critical development then we might as well wait.

PR #8333 (2020-10-02): Add _check_option return value
Fixes #8204. Instead of just returning `True` if the argument is valid, the function now returns the valid argument for further use. This can simplify some code constructs.
closed 2020-10-02 by hoechenberger