mne-tools/mne-python #8205: Continuous head movement KIT/Ricoh? systems

paulsowman, 2020-09-03:
Hi @larsoner, further to our recent exchange I would like to implement the Maxwell filter correction for head movement in MNE. Our KIT125 system records continuous head position in the .con file. What might be required to develop as API to read your head position information / get it to the correct format?

Regards, P

paulsowman, 2020-09-03:
> Reply to: mne-tools/mne-python <reply@reply.github.com>
>
>
>
> If there is any way you could record ~10 sec with a lot of movement (forward and back, left to right, some rotation) so that we can have a file to add to our testing dataset (public) that would help a lot.
>
> In the meantime, if you have some file with head position information in it that you can share with me (privately or publicly) I can look at it to see what's needed.

No problem. I will do this ASAP. P

larsoner, 2020-12-01:
Ping @paulsowman a gentle reminder to record some continuous-movement data at some point

larsoner, 2021-02-19:
I think this is now working, it would be great to have a nice example at some point though!

closed 2021-02-19 by larsoner