mne-tools/mne-python #8211: 3d plot wrong

alienx-zero, 2020-09-03:
I am following many .py file from MNE tutorials. And in any 3d plot project, eg. in https://mne.tools/stable/auto_tutorials/evoked/plot_20_visualize_evoked.html#sphx-glr-auto-tutorials-evoked-plot-20-visualize-evoked-py, ipython console will report a bug in renderer.py like this:

maps = mne.make_field_map(evks['aud/left'], trans=sample_data_trans_file,
                          subject='sample', subjects_dir=subjects_dir)
evks['aud/left'].plot_field(maps, time=0.1)
Using surface from C:\Users\99488\mne_data\MNE-sample-data\subjects\sample\bem\sample-5120-5120-5120-bem.fif.
Getting helmet for system 306m
Prepare EEG mapping...
Computing dot products for 59 electrodes...
Computing dot products for 2562 surface locations...
Field mapping data ready
    Preparing the mapping matrix...
    Truncating at 21/59 components to omit less than 0.001 (0.00097)
    The map will have average electrode reference
Prepare MEG mapping...
Computing dot products for 305 coils...
Computing dot products for 304 surface locations...
Field mapping data ready
    Preparing the mapping matrix...
    Truncating at 210/305 components to omit less than 0.0001 (9.9e-05)
Traceback (most recent call last):

  File "<ipython-input-21-a12fe6cfd2a7>", line 3, in <module>
    evks['aud/left'].plot_field(maps, time=0.1)

  File "d:\Users\99488\Anaconda3\envs\tensorflow-gpu1.14\lib\site-packages\mne\evoked.py", line 356, in plot_field
    fig=fig, verbose=verbose)

  File "<decorator-gen-127>", line 21, in plot_evoked_field

  File "d:\Users\99488\Anaconda3\envs\tensorflow-gpu1.14\lib\site-packages\mne\viz\_3d.py", line 356, in plot_evoked_field
    renderer = _get_renderer(fig, bgcolor=(0.0, 0.0, 0.0), size=(600, 600))

  File "d:\Users\99488\Anaconda3\envs\tensorflow-gpu1.14\lib\site-packages\mne\viz\backends\renderer.py", line 32, in _get_renderer
    set_3d_backend(get_3d_backend(), verbose=False)

  File "d:\Users\99488\Anaconda3\envs\tensorflow-gpu1.14\lib\site-packages\mne\viz\backends\renderer.py", line 141, in get_3d_backend
    % (VALID_3D_BACKENDS))

RuntimeError: Could not load any valid 3D backend: ['mayavi', 'pyvista']

And I try to update or degrade to other MNE version, but this bug still retain in renderer.py but with different error info

drammock, 2020-09-03:
Please include the output of `mne.sys_info()`

alienx-zero, 2020-09-03:
`mne.sys_info()
Platform:      Windows-10-10.0.18362-SP0
Python:        3.7.4 (default, Aug  9 2019, 18:34:13) [MSC v.1915 64 bit (AMD64)]
Executable:    d:\Users\99488\Anaconda3\envs\tensorflow-gpu1.14\pythonw.exe
CPU:           Intel64 Family 6 Model 158 Stepping 13, GenuineIntel: 8 cores
Memory:        31.9 GB

mne:           0.20.7
numpy:         1.16.5 {blas=mkl_rt, lapack=mkl_rt}
scipy:         1.3.1
matplotlib:    3.1.1 {backend=Qt5Agg}

sklearn:       0.22.2.post1
numba:         0.48.0
nibabel:       Not found
cupy:          Not found
pandas:        0.25.2
dipy:          Not found
mayavi:        Not found
pyvista:       Not found
vtk:           Not found`

drammock, 2020-09-03:
3D plots are not possible unless either mayavi or pyvista are installed. See here: https://mne.tools/dev/install/mne_python.html#for-3d-plotting-and-source-analysis

closed 2020-09-03 by drammock

alienx-zero, 2020-09-03:
@drammock thanks for information~