mne-tools/mne-python #8213: ENH: Try fastcore to reduce boilerplate

larsoner, 2020-09-04:
From @agramfort , see [here](https://twitter.com/hamelhusain/status/1301575601497624577?s=11)

agramfort, 2020-09-04:
well I suggested to have a look but not to put this as an MNE dep :)

drammock, 2020-09-04:
Interesting. I've been wondering if `functools.singledispatch` would be good for our functions that take `inst`. A different problem, but one that fastcore seems to also have a solution for.

larsoner, 2020-09-04:
Maybe it's something we can vendor? Will have to look into it

agramfort, 2020-09-04:
slow down :)

sleep over it for some time first

>

cbrnr, 2020-09-04:
😄 - I saw that on Twitter too and although this certainly looks good, I'd rather not add it as a dependency. I also thought that Julia has multiple dispatch (not limited to two args) by default, but hey.

larsoner, 2020-12-01:
Closing, let's use if we need it someday

closed 2020-12-01 by larsoner