mne-tools/mne-python #8215: confusing (wrong?) description in mne.time_frequency.tfr_array_multitaper

chapochn, 2020-09-04:
The description of mne.time_frequency.tfr_array_multitaper is:

> Compute time-frequency transforms using wavelets and multitaper windows.
> Uses Morlet wavelets windowed with multiple DPSS tapers.

However it is not Morlet wavelet that are windowed by DPSS tapers. In _make_dpss, which is called in _compute_tfr, which is called in tfr_array_multitaper, there is:

            oscillation = np.exp(2.0 * 1j * np.pi * f * (t - t_win / 2.))

            # Get dpss tapers
            tapers, conc = dpss_windows(t.shape[0], time_bandwidth / 2.,
                                        n_taps)

            Wk = oscillation * tapers[m]

This is a DPSS taper multiplied by a complex exponential. A Morlet wavelet is a Gaussian multiplied by a complex exponential. There is no Gaussian and thus no Morlet wavelet in the code from what I can tell, unless I misunderstood something.

agramfort, 2020-09-05:
indeed it seems very wrong... at least the doc does not match the code...

what fix would you suggest?

>

chapochn, 2020-09-05:
Ideally would be great to have the actual mathematical formula. I feel it is often missing in the MNE documentation (I'm new to MNE). However, if one does not want to go into that, one could just write the same as for mne.time_frequency.tfr_multitaper

> Compute Time-Frequency Representation (TFR) using DPSS tapers.

Since from what I understand these 2 functions do the same, but just have a different type of input. tfr_multitaper takes as input epochs whereas tfr_array_multitaper takes NumPy array. It would be helpful to specify that fact in the description (if that's actually the case).

The only difference is that tfr_multitaper has the argument zero_mean but tfr_array_multitaper doesn't. I wonder why that is? It seems to be present in tfr_morlet and tfr_array_morlet.

Best might be to have some consistency between the descriptions of:
tfr_multitaper vs tfr_array_multitaper
tfr_morlet vs tfr_array_morlet
tfr_stockwell vs tfr_array_stockwell

I'm new to all this code and MNE, so I don't have a comprehensive overview of the code at this point...

agramfort, 2020-09-05:
+1 on all you suggest.

so let's update the docstrings to match what the code actually does

> The only difference is that tfr_multitaper has the argument zero_mean but
tfr_array_multitaper doesn't. I wonder why that is? It seems to be present
in tfr_morlet and tfr_array_morlet.

this is an API fix to do. Yes the idea is that tfr_ and tfr_array
functions are the same except
one works with mne objects and the other with numpy arrays

can you send us a PR?

thanks a lot for your help

>

chapochn, 2020-09-08:
I did a PR with updated doc.
Only remains is to update tfr_array_multitaper for it to have the option zero_mean for consistency.

PR #8224 (2020-09-10): Updated doc for time frequency analysis functions
#### Reference issue
Partly fixes #8215

#### What does this implement/fix?
Changed multitaper functions description which were wrongly
described as using Morlet wavelets

Also explained the connection between the functions:
tfr_multitaper vs tfr_array_multitaper
tfr_morlet vs tfr_array_morlet
tfr_stockwell vs tfr_array_stockwell
closed 2020-09-10 by drammock