mne-tools/mne-python #8216: apply_baseline then crop

virvw, 2020-09-05:
If your issue is a usage question, please consider asking on our [Gitter channel](https://gitter.im/mne-tools/mne-python) or on our [mailing list](https://mail.nmr.mgh.harvard.edu/mailman/listinfo/mne_analysis) instead of opening an issue.

#### Describe the problem
apply_baseline on epochs followed by a crop removes the former command apply_baseline
It is counter-intuitive that these two operations, in that specific order, would be dependent on each other
If crop first then baseline ---> obsviously non sense
If baseline first then cropt ---> why not possible currently?

For many statistical contrasts, one may wish to apply a baseline (e.g. [-500.0, 0.0]) and then crop epochs (e.g. to apply statistical tests highly sensitive to the number of samples and decimation - wink) 

Currently, the warning is not strong enough to realize this happens: the sequence of operations applied to the data is essentially not respected and unless one looks at the execution online (aka not run some parallel processing) one really easily misses this is ongoing on. 

#### Describe your solution
Could crop have an option apply_baseline = 'on' /  'off' prior to cropping?

#### Describe possible alternatives
1) Have an easy way to mask portions of the signal but this won't help lower the computational problem
2) bypass apply_baseline but then the info history won't be kept

larsoner, 2020-09-06:
Are you on latest `master`? It sounds like this was probably fixed in https://github.com/mne-tools/mne-python/pull/7629 due to issue https://github.com/mne-tools/mne-python/issues/7628

virvw, 2020-09-08:
This is my current setup

conda 4.8.3
Python 3.8.3
(base) vv221713@is235208:~$  python -c "import mne; mne.sys_info()"
Platform:      Linux-4.4.0-187-generic-x86_64-with-glibc2.10
Python:        3.8.3 (default, Jul  2 2020, 16:21:59)  [GCC 7.3.0]
Executable:    /home/vv221713/anaconda3/bin/python
CPU:           x86_64: 32 cores
Memory:        62.5 GB

mne:           0.20.7
numpy:         1.18.5 {blas=mkl_rt, lapack=mkl_rt}
scipy:         1.5.0
matplotlib:    3.2.2 {backend=Qt5Agg}

sklearn:       0.23.1
numba:         0.50.1
pandas:        1.0.5

virvw, 2020-09-08:
1) OK after simple mne reinstall with pip: 
 I confirm that version mne: 0.20.8  does not warn about not applying baseline. However, it does say "baseline off" 

2) However: 
fname_epochs = (epo_dir +"TF_" + c + '_' + sid + "-epo.fif")         
epochs       = mne.read_epochs(fname_epochs)              
if bsln == 1:
   epochs.apply_baseline(baseline = (bsln_tmin,bsln_tmax))
epochs.crop(tmin=lin_tmin, tmax=lin_tmax)

In[22]: epochs
Out[22]: <EpochsFIF  |   21 events (all good), 0 - 2 sec, baseline off, ~21.6 MB, data loaded,
 '2': 21>

3)Sanity check:
when plotting  the same epochs w/ and w/o baseline preceding the crop I get two distinct outcomes: 
[image]

So perhaps just adjusting the "baseline off" would be helpful?

larsoner, 2020-09-08:
> So perhaps just adjusting the "baseline off" would be helpful?

Maybe to "baseline off (or unknown if cropped out of interval)" or something?

virvw, 2020-09-08:
it should be a known unknown for the user so it should better specify what has been done?
how about 
"[bsln_tmin:bsln_tmax]" (these values will be located outside the current epoch length so it implies to have been done before cropping)
or 
" on [bsln_tmin:bsln_tmax]" 
or
"baseline before crop"
or 
"bsln then crop"
or 
"on before crop"
or 
"on and crop"

larsoner, 2020-09-08:
> it should be a known unknown for the user so it should better specify what has been done?

Once `.crop` is done, the information is removed from the object, so not sure this is an option

agramfort, 2020-09-08:
crop makes you lose provenance. But it's hard to avoid. I would make sure you
crop after the latest stage of your pipeline or even when you call functions.

evoked = read_evoked(...)
my_function(evoked.copy().crop(tmin, tmax), ...)

so you always keep the evoked as untouched as possible

hoechenberger, 2020-09-14:
Couldn't we simply **keep** the baseline attribute, even if the baseline period falls outside of the remaining time points after cropping? And in the `str` or `repr` of the `Epochs` / `Evoked`, we could easily detect this case, and display it sth like this:

```
<Epochs |  108 events (all good), 0 - 1.5 sec, baseline [-0.5, 0]s sec (not applicable: data was cropped), ~206.4 MB, data loaded,
 '1': 108>
```
Isn't that what @virvw suggested? It would make it much clearer that 
- baseline-correction was applied at one point
- but the baseline period (partially) removed bc the data was cropped

This leaves no room for interpretation – unlike the currect "let's set baseline to `None` on cropping".

larsoner, 2020-09-14:
That would be nice, and it would avoid us having to do a bunch of `verbose='error'` to ignore the baseline removed bits. @hoechenberger feel free to try changing the code to do this and let's see if it works. My guess is that it's going to be a challenge and we decided it was easier just to discard the information previously, but I'm not sure.

larsoner, 2020-09-14:
I consider this idea a block for 0.21 release since it affects some warnings we now emit in this cycle...

hoechenberger, 2020-09-14:
@larsoner I can implement this, but I will be busy until including Friday. So no chance I will get this done before the weekend. Let's push the planned release date by a week?

hoechenberger, 2020-09-15:
Note to self: To handle cases where the baseline was specified with an implicit  boundary `tmin=None` or `tmax=None`, we actually need an attribute to keep track of whether we chopped off part of the baseline during cropping. `Epochs._baseline_was_cropped` or something. </end of note to self>

hoechenberger, 2020-09-15:
Note to self 2: When a user specifies an implicit time via `tmin=None` or `tmax=None`, always auto-map it to an explicit time. i.e.,

```python
In [2]: epochs.tmin, epochs.tmax
Out[2]: (-0.9989760657919393, 3.0002581175951244)

In [3]: epochs.apply_baseline(None, 0)
Out[3]: 
<Epochs |  111 events (good & bad), -0.998976 - 3.00026 sec, baseline [-0.998976, 0], ~3.6 MB, data not loaded,
 '1': 111>
```
instead of what we have currently:
```python
In [2]: epochs.tmin, epochs.tmax
Out[2]: (-0.9989760657919393, 3.0002581175951244)

In [3]: epochs.apply_baseline(None, 0)
Out[3]: 
<Epochs |  111 events (good & bad), -0.998976 - 3.00026 sec, baseline [None, 0], ~3.6 MB, data not loaded,
 '1': 111>
```

larsoner, 2020-09-15:
@hoechenberger okay for me to take a stab at this today or tomorrow? It would be nice not to have to delay release a full week...

hoechenberger, 2020-09-15:
> @hoechenberger okay for me to take a stab at this today or tomorrow? It would be nice not to have to delay release a full week...

Yes, by all means, do take a stab! I'm always happy if others do some of my work ;)

hoechenberger, 2020-09-15:
> When a user specifies an implicit time via `tmin=None` or `tmax=None`, always auto-map it to an explicit time

What do you think of this? Would allow us to easily figure out if the baseline is still present on-the-fly.

larsoner, 2020-09-15:
It might end up being nicer/cleaner to do it that way. I think it will break some tests, and either way we're going to need to figure out if the baseline is "pending" or not depending on preload, so not sure it will buy us too much. But we'll see...

hoechenberger, 2020-09-15:
> It might end up being nicer/cleaner to do it that way.

Personally, as a user, I would prefer it that way anyway. Or at least replace the `None` in the time period with `∞` :) `None` is really just an implementation detail that probably shouldn't leak into the user-parsable string representation

larsoner, 2020-09-15:
I am also getting errors like this on `master`:
```
>>> evoked.copy().crop(tmin, tmax)
ValueError: Baseline interval (tmin = -0.05000000074505806) is outside of data range (tmin = 0.3)
```
i.e., instead of a warning I'm getting an error. So I don't think `re`-matching our errors is the safest route here -- it's also not robust to future changings in wordings of these errors.

I'm thinking that we should probably revert #8156 for the release -- it seems to risky to include it at this point. Then we can add it back in or put the solutions described here in instead.

WDYT @hoechenberger ?

hoechenberger, 2020-09-15:
> WDYT @hoechenberger ?

How about just reverting the `re`-matching part? I actually only included it because @agramfort wasn't too happy about the broad exception handling (catching **all** `ValueError`s) which had been there before.

hoechenberger, 2020-09-15:
@larsoner It's these two commits:
https://github.com/mne-tools/mne-python/pull/8156/commits/def384dd2494043950c83d158d93d2959007f532
https://github.com/mne-tools/mne-python/pull/8156/commits/2de7accc85ca12431aff838b61b40a2903fe3500

larsoner, 2020-09-15:
> How about just reverting the re-matching part? I actually only included it because @agramfort wasn't too happy about the broad exception handling (catching all ValueErrors) which had been there before.

It's a bit more than this, though. This catching thing in particular is buggy, but also more generally:

1. The PR changed a lot of core code, so it's risky to change right before release (as evidenced by bugs I just now found by trying to use it in practice)
2. In this issue we talk about trying to at least revert some or all of the warnings anyway by properly tracking baseline in all conditions, so it's possible we will annoy people by forcing them to do a bunch of `verbose='error'`s all over the place for one release

These are two things that we try to avoid.

hoechenberger, 2020-09-15:
Fair enough, let's revert #8156 and try to address these changes ASAP after cutting the release? I'm worried about ending up in rebase hell. 😵

Probably the changes discussed here should then also be added to the  #8156 branch, I can imagine things becoming super messy otherwise, WDYT?

larsoner, 2020-09-15:
It's probably worth starting fresh with making Epochs work with a baseline that's outside the window, then adding the appropriate parts of #8156 on top just by copy-pasting code in a new branch

agramfort, 2020-09-15:
oups... indeed reverting appears much safer

hoechenberger, 2020-09-15:
😭

Go ahead, revert my baby! 😵

virvw, 2020-11-25:
@hoechenberger @larsoner I am bit a lost on the status of that issue. Which version of python can I use to make sure baseline followed by crop will work for now? ty

hoechenberger, 2020-11-26:
Hello @virvw,

the status is that it **already** works in the latest stable version – the baseline correction is not being removed from the **data** when you crop. However, the epochs **do** lose the information as to **which** baseline had been used, and misleadingly state that the baseline has been "removed". So we're mostly talking about a displaying issue here, if you will. In #8444 I'm reimagining this behavior, which is not too easy to change because there are numerous of potential side-effects to account for. @larsoner has provided input and ideas multiple times, and I will work on this again (and hopefully finally be able to resolve it) within the next few hours.

larsoner, 2020-12-16:
Closing as this is fixed in `master` now

closed 2020-12-16 by larsoner