mne-tools/mne-python #8221: RFC: increasing diversity of our contributor pool

drammock, 2020-09-08:
As some of you know, part of our funding proposal to CZI was for funds to increase participation of underrepresented groups at our code sprints (and in our contributor pool more generally).  Due to the global SARS-CoV-2 pandemic, the fall 2020 code sprint in Seattle is postponed, and it is unclear whether a spring 2021 sprint will be possible / whether people will feel comfortable enough with air travel to make the journey.  This issue is for brainstorming ways to recruit and retain a diverse pool of contributors, without relying on (subsidies to attend) in-person code sprints.  @britta-wstnr and I have been chatting a bit about this; here is a summary of our ideas so far:

1. **Prominent public statements that we welcome *all* new contributors.**  Locations might include the main homepage, the GitHub README, the MNE-Analyze email footer, and/or including a slide and verbal statement in any MNE-Python-related talks/trainings done by core devs.
2. **Designated "ambassadors"** who are willing to work directly with new contributors a little bit more than the usual PR review. Ambassadors should be willing to exchange a few emails or videoconference with new contributors to answer questions, give guidance, even do a little screen sharing or pair programming as appropriate.
3. **Designated "mentors"** who would commit to a certain number of hours/week for a defined period, and get matched with an "intern" to work one-on-one with during that period. It may or may not be possible to provide stipends for the mentor and/or the intern for this option.

The goals for new contributors are flexible and expected to vary depending on the new contributors' past experiences and current interests, but generally will include some of:

- developing comfort and fluency using git / GitHub
- understanding Python package installation (i.e., different ways to install MNE-Python and its dependencies, upgrading, etc)
- learning best (or "good enough") practices for scientific programming & scientific software development (documentation, testing, deprecation cycles, etc)

Please weigh in here if you have alternatives or additions to suggest, want to express support or lack of support for any ideas already mentioned, or have other related comments to share. **We are very much open to other ideas.** Also please tag anyone you think should see this but who might not get auto-notified without an @-mention.

hoechenberger, 2020-09-08:
tagging @SophieHerbst and @dnacombo
cc @peircej @opyh

agramfort, 2020-09-08:
an idea... have like office hours 1H every week or every two weeks to onboard
people or ask questions in a video chat. What you know people it's always
easier to work together

britta-wstnr, 2020-09-08:
Thank you for the summary, @drammock.
I think what we should add to the first point (Prominent public statements) is to also make clear that diversity is a value of the MNE-Python community and that we believe it contributes to our joint goals. 
To essentially go beyond "diversity just for the stats" or other negative descriptors that can often be associated with quotas or similar - to show that we believe in the positive effects of a diverse contributor pool and therefore actively work on increasing the representation of underrepresented groups.

drammock, 2020-09-08:
> we believe in the positive effects of a diverse contributor pool

:100:

jasmainak, 2020-09-09:
Just a comment as I was passing by. I think it might be a good time to review what issues are *truly* "easy", which ones are "good first issues" etc. Something I've found helps new contributors is to give them progressively harder issues. If you accidentally take up an issue that's too hard, it's going to discourage future contributions.

virvw, 2020-09-09:
Hi folks,

great to see this happening! :-)

My 3 cents:

- inverse the balance and strongly highlight contributions of the
minorities (even if in the grand scheme of things they are smaller). By
default, biases favor so strongly the majority that the regular pat on the
back for a minority is not effective enough. The bragging needs to be 2 or
3 times louder for minorities...

- What is effective is to be fact-oriented: count the ratio of selected
minorities in MEG/EEG/... and compare with your current pool. Then state
out loud: there is 30% of women in the field, we want 30 % of women
contributors - who is up for it?; there is 5 % of Chinese users, we want 5%
of contributors etc...  In other words, pro-actively seek and invite them.

- Build a system of mentorship: the youngest arrival is paired with a more
senior for a few months (perhaps what Daniel referred to as ambassadors).

Cheers,

Virginie van Wassenhove
*https://brainthemind.com/ <https://brainthemind.com/>*

SophieHerbst, 2020-09-09:
Great initiative and great suggestions above! Something I would think might help too, is to offer small hands-on technical tutorials for new contributors (~2h with an experienced programmer, possibly virtual but interactive) to help people get up to speed and increase their _confidence_ for instance: 
- how to use git / github
- principles for clean coding in python
- introduction to the mne architecture
- logic behind the testing (CIs and stuff)

britta-wstnr, 2020-09-09:
@SophieHerbst do you mean something like a regular little hands-on lecture new contributors can sign up for? I think that's a great idea!

SophieHerbst, 2020-09-09:
@britta-wstnr yes, something shorter than a full workshop (to be integrated in one's daily schedule more easily), which targets the skills specifically needed to contribute to mne python.

drammock, 2020-09-09:
> something shorter than a full workshop, which targets the skills specifically needed to contribute to mne python.

I like this idea a lot. Perhaps we could start with the framework of The Carpentries lessons and tailor then to our needs.

hoechenberger, 2020-12-04:
I just opened my first issue at the NeuroKit2 repo – and got a friendly automated welcome message from a bot! Interesting idea!

https://github.com/neuropsychology/NeuroKit/issues/392#issuecomment-738694192

cbrnr, 2020-12-04:
😄 - except for the meme I like this idea!

hoechenberger, 2020-12-04:
> 😄 - except for the meme I like this idea!

Yes :)
One could then even include a link to our forum and such.

m-macaskill, 2020-12-06:
That's great - even though it's automated, getting an immediate friendly response is a nice touch.

Mike

> 
> 
> 😄 - except for the meme I like this idea!
> 
> Yes :)
> One could then even include a link to our forum and such.

hoechenberger, 2020-12-21:
@peircej recently posted "[on] improving Equality, Diversity and Inclusivity (EDI) in our community"
I believe this is a valuable read:
https://discourse.psychopy.org/t/on-improving-equality-diversity-and-inclusivity-edi-in-our-community/18356

drammock, 2021-01-12:
ICYMI: we've just announced the first **MNE-Python office hours**.  Details [here](https://mne.discourse.group/t/announcing-mne-python-office-hours/2468). 🚀

drammock, 2021-02-15:
Applications are open for the MNE-Python New Developers Code Sprint!  Please share widely.  Stipends are available for accepted applicants, with priority given to participants from groups underrepresented in neuroscience. More info here: https://mne.discourse.group/t/save-the-date-15-19-march-mne-python-code-sprint/2684/5?u=drammock

britta-wstnr, 2021-02-15:
Can mne_news tweet about it?

drammock, 2021-02-15:
I put a tweet request (via slack) earlier today... I don't even know who else has access to that account other than @agramfort

‐‐‐‐‐‐‐ Original Message ‐‐‐‐‐‐‐

ktavabi, 2021-03-05:
In all fairness this sounds like an effort to replicate the software carpentry model into electrophysiology with a smattering of equitable diversity that IMHO already exists in the MNE ecosystem. I am curious as to what the MNE-Analysis-mail-list crowd  thinks of this? 
At the end of the day folks need to work with electrophysiology data that's first and foremost representatively sampled, and yes ideally analyzed by a culture of curiosity as @agramfort suggests rather than some independent monolith based on distributions of race and gender as suggested by @virvw in niche open source sci-computing communities. Seems like putting the horse behind the carriage. That's just my .000000000001 alt coin

hoechenberger, 2021-03-05:
Now this can be a really long and extensive discussion! 👌 
My understanding is that the primary focus of this initiative is to try and encourage more people of all backgrounds to contribute to MNE. The first step to this is try to create a welcoming environment for people of various backgrounds.

I have a couple of female close friends who are working in "tech"; and shared some of their impressions and, indeed, issues when joining male-dominated working environments with me. Without going into too much detail, what they found was that certain "guy-things" were well-established (and never reflected upon), which simply made them feel uncomfortable (and, really – quite a bunch of those were plainly unprofessional!). This is not a great experience, and I believe if we can find ways to identify and reduce such issues, it will be beneficial to everybody. 

I'm not sure what you're referring to when mentioning "diversity in human intelligence"? I don't think we have a lack of human intelligence in the project (what is that in the context of MNE, anyway?); but we all have different sets of skills, preferences, and requirements. Trying to find a common ground, some kind of consensus among us all, helps us create a product that is indeed useful to many more people than one that were tailored only to the needs of a few. By increasing the diversity of our contributor pool, this aspect can be emphasized and strengthened even more – leading to an overall better product.

And that's just about the quality of the end product – I haven't even talked about the valuable inter-personal exchanges that sometimes emerge beyond filing, reviewing, and merging pull requests! Now you can say what you want, but to me, an open-source project is also a sort of a social hub that helps me interact, get to know, and forces me to understand different kinds of people – sometimes pushing me outside of my comfort zone. I very much appreciate that, and I'm looking forward to seeing much, much more of that.

ktavabi, 2021-03-05:
@hoechenberger good luck finding that 🦄 friend

britta-wstnr, 2021-03-05:
Thanks @hoechenberger for your very elaborate answer!
May I add that it furthermore has been shown that diverse groups benefit scientific research and innovation (e.g. AlShebli et al., 2018, Nat Comm; Hofstra et al., 2020, PNAS), so it will lead to a "better product", as @hoechenberger also has hinted at.
I do not believe that diversity in any scientific field has to be a unicorn: but to get there we need change, and I love that MNE-Python is actively taking steps to bring such change!

drammock, 2021-03-05:
thanks (sincerely) for questioning our ideas, @ktavabi (we are scientists, after all!)

> In all fairness this sounds like an effort to replicate the software carpentry model into electrophysiology

Maybe.  SWC offers good infrastructure and support for lesson development, and there are efforts underway on at least [five other carpentries-style lessons related to neuroimaging](https://carpentries.org/community-lessons/#neuroimaging).  Personally I don't think it makes sense right now for MNE-Python to do a user-focused lesson like that, since I think we have if anything an excess of user-facing tutorial material that needs to be streamlined, and it's already nicely formatted.  But I can see the value in what @SophieHerbst suggests: a very short course (~2 hours, instead of 2 days like a typical SWC workshop) that remixes / cherry-picks the parts of the SWC git curriculum that are most relevant for OSS contribution in general --- and for MNE-Python contribution in particular --- and supplements that with additional info like how to run our tests, how our CIs work, etc.

> ...with a smattering of equitable diversity that IMHO already exists in the MNE ecosystem.

I respectfully disagree that it already exists here. The MNE-Python contributor pool is overwhelmingly male and predominantly white/caucasian americans or europeans: `git shortlog -nse` and some googling of contributor names should hopefully convince you...  I don't know how everyone self-identifies but guessing based on appearances I would say that 19 of top 20 MNE-Python contributors (or 46 of top 50) are male, and at least 15 of top 20 are caucasian/white. So to my eyes the imbalance clearly exists, and as @hoechenberger and @britta-wstnr mentioned, such imbalances can reinforce and perpetuate themselves by normalizing practices that make minority participants feel unwelcome, and repairing those imbalances can actually yield better results/products.

> At the end of the day folks need to work with electrophysiology data that's first and foremost representatively sampled, and yes ideally analyzed by a representative culture of curiosity as @agramfort suggests rather than an independent monolith based on distributions of race and gender as suggested by @virvw in what is effectively a niche open source sci-computing community.

I strongly agree that data diversity is a big problem for biomedical science, across the board (because it results in people getting different quality of treatment from healthcare / different levels of reward from their society's investments in science). But the grant that I wrote and was awarded was for supporting open source scientific *software*, not for fixing the biased data problem. I can't legally spend that money on something else.

> To this aging user, honestly this seems like we're putting the horse behind the carriage. Meaning we're considering using moneys from the philanthropic arm of what is essentially a corporate monopoly on AI algorithms to increase diversity in human intelligence? Or am I missing something?

I don't quite follow the horse/carriage analogy here (something about AI making better humans, instead of vice-versa?).  But I do think you are missing at least one point: what we are discussing is not an effort to "increase diversity in human intelligence". We are discussing ways to make this developer community better reflect the communties that we live in and serve.

ktavabi, 2021-03-05:
Fine, fine. "Product" it is then. This is more rationalization than decision making. But we should take some safeguards, against future (n)pandemics, no? Additional biometrics? No, that'd be overkill. The Chinese would hate that. 
Where do I sign up for office hours with Alex?

drammock, 2021-03-05:
> Fine, fine. "Product" it is then. This is more rationalization than decision making. 

I'm not sure I follow you here.

> But we should take some safeguards, against future (n)pandemics, no? Additional biometrics? No, that'd be overkill. The Chinese would hate that.

Again, not sure I understand.

> Where do I sign up for office hours with Alex?

Office hours are twice monthly on our Discord server, and are hosted by whichever MNE-Python core devs are available that day. See [here](https://mne.discourse.group/t/announcing-mne-python-office-hours/2468) for more details.  You just missed one this morning, and the usual office hours on 19 March will be superseded by the code sprint, so the next one will likely be 2 April.

ktavabi, 2021-03-05:
> I'm not sure I follow you here.
Sorry I forgot the qualifier "better product" 

> Again, not sure I understand. Your remarks here could be interpreted to imply something untoward about Chinese people or the Chinese government. If so, this would violoate our code of conduct. Please consider this a gentle reminder to choose your words carefully
Ummm, interpretation is in the eye of the beholder Dan. It's your money, it stipulates you do(x). Good luck. I look forward to witnessing the fruits of your effort. Maybe we'll discord on discord homie. That's all folks.

hoechenberger, 2021-06-18:
I thought this was an interesting thread on Twitter, started by Guido van Rossum, asking 

> How can I say "we would especially love to hear from women, Latinx and Black candidates" without sounding, how do I say this, "bigoted"? This subject is so touchy that the standard phrasing on our career site doesn't even fit in a tweet. :-)

https://twitter.com/gvanrossum/status/1405298970838978563?s=21

Takeaway for me was: it's probably best to reach out to marginalized communities directly.