mne-tools/mne-python #8222: Issue Reading in Channel Names (that have integers and longer then 4 characters?) in EDF File

adam2392, 2020-09-08:
#### Describe the bug
I am converting a dataset into BIDs using the latest version of mne-bids. I read in a raw EDF file using: `read_raw_edf`. Then I try setting the channel types using `set_channel_type` on the Info object. However it errors out on channels `F10F1-6`. I also check if the string exists in the `raw.info['ch_names']` and it returns False!

The downstream error in mne-bids shows that all the channels match up, so I presume its an error on the EDF read operation:

```
RuntimeError: Channels do not correspond between raw data and the channels.tsv file. For MNE-BIDS, the channel names in the tsv MUST be equal and in the same order as the channels in the raw data.

148 channels in tsv file: "/Users/adam2392/OneDrive - Johns Hopkins/sickkids/sub-E5/ses-presurgery/ieeg/sub-E5_ses-presurgery_task-IED_acq-ecog_run-04_channels.tsv"

 --> ['C1', 'C2', 'C3', 'C4', 'C5', 'C6', 'C7', 'C8', 'C9', 'C10', 'C11', 'C12', 'C13', 'C14', 'C15', 'C16', 'C17', 'C18', 'C19', 'C20', 'C21', 'C22', 'C23', 'C24', 'C25', 'C26', 'C27', 'C28', 'C29', 'C30', 'C31', 'C32', 'C33', 'C34', 'C35', 'C36', 'C37', 'C38', 'C39', 'C40', 'C41', 'C42', 'C43', 'C44', 'C45', 'C46', 'C47', 'C48', 'ML1', 'ML2', 'ML3', 'ML4', 'ML5', 'ML6', 'F3C1', 'F3C2', 'F3C3', 'F3C4', 'F3C5', 'F3C6', 'F1OF1', 'F1OF2', 'F1OF3', 'F1OF4', 'F1OF5', 'F1OF6', 'C67', 'C68', 'C69', 'C70', 'C71', 'C72', 'C73', 'C74', 'C75', 'C76', 'C77', 'C78', 'C79', 'C80', 'C81', 'C82', 'C83', 'C84', 'C85', 'C86', 'C87', 'C88', 'C89', 'C90', 'C91', 'C92', 'C93', 'C94', 'C95', 'C96', 'C97', 'C98', 'C99', 'C100', 'C101', 'C102', 'C103', 'C104', 'C105', 'C106', 'C107', 'C108', 'C109', 'C110', 'C111', 'C112', 'C113', 'C114', 'C115', 'C116', 'C117', 'C118', 'C119', 'C120', 'C121', 'C122', 'C123', 'LEMG1', 'LEMG2', 'REMG1', 'REMG2', 'EKG', 'DC1', 'DC2', 'DC3', 'DC4', 'DC5', 'DC6', 'DC7', 'DC8', 'DC9', 'DC10', 'DC11', 'DC12', 'DC13', 'DC14', 'DC15', 'DC16', 'TRIG', 'OSAT', 'PR', 'PLETH']

148 channels in raw file: "sub-E5_ses-presurgery_task-IED_acq-ecog_run-04_ieeg.edf"

 --> ['C1', 'C2', 'C3', 'C4', 'C5', 'C6', 'C7', 'C8', 'C9', 'C10', 'C11', 'C12', 'C13', 'C14', 'C15', 'C16', 'C17', 'C18', 'C19', 'C20', 'C21', 'C22', 'C23', 'C24', 'C25', 'C26', 'C27', 'C28', 'C29', 'C30', 'C31', 'C32', 'C33', 'C34', 'C35', 'C36', 'C37', 'C38', 'C39', 'C40', 'C41', 'C42', 'C43', 'C44', 'C45', 'C46', 'C47', 'C48', 'ML1', 'ML2', 'ML3', 'ML4', 'ML5', 'ML6', 'F3C1', 'F3C2', 'F3C3', 'F3C4', 'F3C5', 'F3C6', 'F1OF1', 'F1OF2', 'F1OF3', 'F1OF4', 'F1OF5', 'F1OF6', 'C67', 'C68', 'C69', 'C70', 'C71', 'C72', 'C73', 'C74', 'C75', 'C76', 'C77', 'C78', 'C79', 'C80', 'C81', 'C82', 'C83', 'C84', 'C85', 'C86', 'C87', 'C88', 'C89', 'C90', 'C91', 'C92', 'C93', 'C94', 'C95', 'C96', 'C97', 'C98', 'C99', 'C100', 'C101', 'C102', 'C103', 'C104', 'C105', 'C106', 'C107', 'C108', 'C109', 'C110', 'C111', 'C112', 'C113', 'C114', 'C115', 'C116', 'C117', 'C118', 'C119', 'C120', 'C121', 'C122', 'C123', 'LEMG1', 'LEMG2', 'REMG1', 'REMG2', 'EKG', 'DC1', 'DC2', 'DC3', 'DC4', 'DC5', 'DC6', 'DC7', 'DC8', 'DC9', 'DC10', 'DC11', 'DC12', 'DC13', 'DC14', 'DC15', 'DC16', 'TRIG', 'OSAT', 'PR', 'Pleth']
```

#### Steps to reproduce
```
raw = read_raw_edf(edf_fpath)

depths = [f'ML{idx}' for idx in range(1, 7)] + \
                 [f'F3C{idx}' for idx in range(1, 7)] + \
                 [f'F10F{idx}' for idx in range(1, 7)]

for ch in depths:
        try:
            raw.set_channel_types({ch: 'seeg'})
        except ValueError as e:
            not_set.append(ch)

print('This shold work!')
print('F10F1' in ch_names)
>>> False
````

#### Additional information
```
>>> mne.sys_info()
Platform:      Darwin-19.6.0-x86_64-i386-64bit
Python:        3.7.8 | packaged by conda-forge | (default, Jul 23 2020, 03:39:37)  [Clang 10.0.0 ]
Executable:    /Users/adam2392/OneDrive - Johns Hopkins/sickkids/sample_scripts/convert_sickkids_repo/.venv/bin/python
CPU:           i386: 8 cores
Memory:        Unavailable (requires "psutil" package)
mne:           0.21.dev0
numpy:         1.19.1 {blas=openblas, lapack=openblas}
scipy:         1.5.2
matplotlib:    Not found

sklearn:       Not found
numba:         Not found
nibabel:       Not found
cupy:          Not found
pandas:        Not found
dipy:          Not found
mayavi:        Not found
pyvista:       Not found
vtk:           Not found
```

[E5_Sz.edf.zip](https://github.com/mne-tools/mne-python/files/5190401/E5_Sz.edf.zip)

agramfort, 2020-09-08:
mne requires all channels to have unique names and is case sensitive (at
least should).

F10F1-6 suggest a -6 was appended due to many channels being called F10F1

agramfort, 2020-09-08:
can you share more than the file to investigate? like the bids dataset

adam2392, 2020-09-08:
> mne requires all channels to have unique names and is case sensitive (at least should). F10F1-6 suggest a -6 was appended due to many channels being called F10F1

Sorry I wasn’t clear. F10F1-6 is 6 channels. I updated the description to be a bit more clear. 

And it errors out when I check if ‘F10F1’ is “in” the Raw channel names. I’m not sure if it’s an issue with String decoding?

agramfort, 2020-09-08:
In [5]: 'F10F1' in raw.ch_names
Out[5]: False

In [6]: 'F1OF1' in raw.ch_names
Out[6]: True

>

adam2392, 2020-09-08:
lol... okay that's embarassing. Sorry @agramfort for the waste of time.

closed 2020-09-08 by adam2392