mne-tools/mne-python #8226: How to solve the problem of TypeError

chenyuan95, 2020-09-09:
Traceback (most recent call last):
  File "D:/mne使用/use01.py", line 15, in <module>
    raw = mne.io.read_raw_eeglab(r'D:\脑电数据原始\ex2\sub1902.set',preload=False)
  File "D:\mne使用\venv\lib\site-packages\mne\io\eeglab\eeglab.py", line 221, in read_raw_eeglab
    return RawEEGLAB(input_fname=input_fname, preload=preload,
  File "<decorator-gen-197>", line 21, in __init__
  File "D:\mne使用\venv\lib\site-packages\mne\io\eeglab\eeglab.py", line 319, in __init__
    eeg = _check_load_mat(input_fname, uint16_codec)
  File "D:\mne使用\venv\lib\site-packages\mne\io\eeglab\eeglab.py", line 59, in _check_load_mat
    eeg = read_mat(fname, uint16_codec=uint16_codec)
  File "D:\mne使用\venv\lib\site-packages\mne\externals\pymatreader\pymatreader.py", line 82, in read_mat
    raw_data = scipy.io.loadmat(fid, struct_as_record=True,
  File "D:\mne使用\venv\lib\site-packages\scipy\io\matlab\mio.py", line 224, in loadmat
    matfile_dict = MR.get_variables(variable_names)
  File "D:\mne使用\venv\lib\site-packages\scipy\io\matlab\mio5.py", line 333, in get_variables
    res = self.read_var_array(hdr, process)
  File "D:\mne使用\venv\lib\site-packages\scipy\io\matlab\mio5.py", line 293, in read_var_array
    return self._matrix_reader.array_from_header(header, process)
  File "mio5_utils.pyx", line 671, in scipy.io.matlab.mio5_utils.VarReader5.array_from_header
  File "mio5_utils.pyx", line 719, in scipy.io.matlab.mio5_utils.VarReader5.array_from_header
  File "mio5_utils.pyx", line 966, in scipy.io.matlab.mio5_utils.VarReader5.read_struct
  File "mio5_utils.pyx", line 669, in scipy.io.matlab.mio5_utils.VarReader5.read_mi_matrix
  File "mio5_utils.pyx", line 713, in scipy.io.matlab.mio5_utils.VarReader5.array_from_header
  File "mio5_utils.pyx", line 876, in scipy.io.matlab.mio5_utils.VarReader5.read_char
TypeError: buffer is too small for requested array

Process finished with exit code 1

agramfort, 2020-09-09:
you have a problem to read an eeglab file? if can you share it after checking you can read it in matlab?

BarryLiu-97, 2020-09-10:
你的中文路径暴露了你。

BarryLiu-97, 2020-09-10:
> > 你的中文路径暴露了你。
> 
> 对我来说，他的中文路径没有问题。

I know, I'm just saying that indicates that he/she is Chinese as the same as me.

drammock, 2020-09-10:
> I know, I'm just saying that indicates that he/she is Chinese as same as me.

Yes, I understood. I just thought it was a little off-topic to point it out, since it's likely not the cause of the `ValueError`

BarryLiu-97, 2020-09-10:
> > I know, I'm just saying that indicates that he/she is Chinese as same as me.
> 
> Yes, I understood. I just thought it was a little off-topic to point it out, since it's likely not the cause of the `ValueError`

understood.

chenyuan95, 2020-09-10:
yeah,I am Chinese student.I very happy to receive your answer.it is not because of the error caused by the Chinses path.and i can open the .set file with EEGLAB successfully,and i can load it in RAM in mne but sometimes it still goes wrong.so it is unreliable.

Mymiku, 2020-11-25:
Same problem, have you solved it?

drammock, 2020-11-25:
@Mymiku can you provide us with a link to a file that opens fine in EEGLAB but gives this error in MNE-Python? We cannot easily debug if we cannot reproduce it.

closed 2021-02-14 by hoechenberger

Mymiku, 2021-02-16:
@drammock acutually i have solved the problem by adding the argument "uint16_codec="latin1" ", thanks a lot

huayueximujun, 2021-08-24:
> @Mymiku can you provide us with a link to a file that opens fine in EEGLAB but gives this error in MNE-Python? We cannot easily debug if we cannot reproduce
Write uint16_codec="latin1" on line 203 of file \lib\site-packages\scipy\io\matlab\mio5.py

huayueximujun, 2021-08-24:
> > I know, I'm just saying that indicates that he/she is Chinese as same as me.
> 
> Yes, I understood. I just thought it was a little off-topic to point it out, since it's likely not the cause of the `ValueError`

Write uint16_codec="latin1" on line 203 of file \lib\site-packages\scipy\io\matlab\mio5.py

huayueximujun, 2021-08-24:
> Traceback (most recent call last):
> File "D:/mne使用/use01.py", line 15, in
> raw = mne.io.read_raw_eeglab(r'D:\脑电数据原始\ex2\sub1902.set',preload=False)
> File "D:\mne使用\venv\lib\site-packages\mne\io\eeglab\eeglab.py", line 221, in read_raw_eeglab
> return RawEEGLAB(input_fname=input_fname, preload=preload,
> File "", line 21, in **init**
> File "D:\mne使用\venv\lib\site-packages\mne\io\eeglab\eeglab.py", line 319, in **init**
> eeg = _check_load_mat(input_fname, uint16_codec)
> File "D:\mne使用\venv\lib\site-packages\mne\io\eeglab\eeglab.py", line 59, in _check_load_mat
> eeg = read_mat(fname, uint16_codec=uint16_codec)
> File "D:\mne使用\venv\lib\site-packages\mne\externals\pymatreader\pymatreader.py", line 82, in read_mat
> raw_data = scipy.io.loadmat(fid, struct_as_record=True,
> File "D:\mne使用\venv\lib\site-packages\scipy\io\matlab\mio.py", line 224, in loadmat
> matfile_dict = MR.get_variables(variable_names)
> File "D:\mne使用\venv\lib\site-packages\scipy\io\matlab\mio5.py", line 333, in get_variables
> res = self.read_var_array(hdr, process)
> File "D:\mne使用\venv\lib\site-packages\scipy\io\matlab\mio5.py", line 293, in read_var_array
> return self._matrix_reader.array_from_header(header, process)
> File "mio5_utils.pyx", line 671, in scipy.io.matlab.mio5_utils.VarReader5.array_from_header
> File "mio5_utils.pyx", line 719, in scipy.io.matlab.mio5_utils.VarReader5.array_from_header
> File "mio5_utils.pyx", line 966, in scipy.io.matlab.mio5_utils.VarReader5.read_struct
> File "mio5_utils.pyx", line 669, in scipy.io.matlab.mio5_utils.VarReader5.read_mi_matrix
> File "mio5_utils.pyx", line 713, in scipy.io.matlab.mio5_utils.VarReader5.array_from_header
> File "mio5_utils.pyx", line 876, in scipy.io.matlab.mio5_utils.VarReader5.read_char
> TypeError: buffer is too small for requested array
> 
> Process finished with exit code 1

Write uint16_codec="latin1" on line 203 of file \lib\site-packages\scipy\io\matlab\mio5.py

akitasawa, 2021-09-27:
uint16_codec="latin1"  this method is too great！ thanks！

> Write uint16_codec="latin1" on line 203 of file \lib\site-packages\scipy\io\matlab\mio5.py

thanks，its  solved to my question！

huayueximujun, 2021-09-27:
> uint16_codec="latin1" this method is too great！ thanks！
>
>
> thanks，its solved to my question！

you are welcome

Python-learing, 2021-11-08:
> uint16_codec="latin1" this method is too great！ thanks！
>
>
> thanks，its solved to my question！

hello, I want to know where to write, thanks! Because line 203 is a comment.

Python-learing, 2021-11-08:
> Traceback (most recent call last): File "D:/mne使用/use01.py", line 15, in raw = mne.io.read_raw_eeglab(r'D:\脑电数据原始\ex2\sub1902.set',preload=False) File [...]

Same question. Have you solved it?

Python-learing, 2021-11-08:
@Mymiku hello, may I ask you where to write it? Thanks！

drammock, 2021-11-08:
nowadays, `read_raw_eeglab` has a parameter `uint16_codec` where you can simply pass the codec to the function.  Try passing 'latin1' as that parameter. If that doesn't work, **please provide a link to a file that loads correctly in MATLAB but fails to load in MNE-Python**.  Without such a file to test with, we cannot debug and therefore cannot fix.

huayueximujun, 2021-11-08:
> Same question. Have you solved it?
A look at drammock's solution just below might solve it

huayueximujun, 2021-11-08:
> nowadays, `read_raw_eeglab` has a parameter `uint16_codec` where you can simply pass the codec to the function. Try passing 'latin1' as that parameter. If that doesn't work, **please provide a link [...]

I will try your method and it seems to work

Python-learing, 2021-11-08:
> nowadays, `read_raw_eeglab` has a parameter `uint16_codec` where you can simply pass the codec to the function. Try passing 'latin1' as that parameter. If that doesn't work, **please provide a link [...]

thanks, I will try it.

Python-learing, 2021-11-09:
> I will try your method and it seems to work

hello， have you solved?
Can I contact you?

drammock, 2021-11-09:
@Python-learing please try it out yourself with your own data and let us know if it works. Something like this:

```py
import mne
fname = 'path/to/my/eeglab/file.set'  # edit this line
raw = mne.io.read_raw_eeglab(fname, uint16_codec='latin1')
```

If it doesn't work, **please provide us with a link to download one of the files that doesn't work** so we can debug the problem.

Python-learing, 2021-11-09:
> @Python-learing please try it out yourself with your own data and let us know if it works. Something like this:
> 
> ```python
> import mne
> fname = 'path/to/my/eeglab/file.set'  # edit this line
> raw = mne.io.read_raw_eeglab(fname, uint16_codec='latin1')
> ```
> 
> If it doesn't work, **please provide us with a link to download one of the files that doesn't work** so we can debug the problem.

ok, thanks.

Python-learing, 2021-11-10:
@drammock 
1. A01T2 has electrode positioning, re-reference, extract epochs and removal of EOG and ICA through eeglab. It can read but say: The number of trials is 288. It must be 1 for raw files. Please use `mne.io.read_epochs_eeglab` if the .set file contains epochs.
2. A01T4 just has channel location, reference and removal of EOG.
The data is BCI Competition IV Data sets 2a subject1. Thanks very much, sorry for the trouble caused to you
[A01T2.zip](https://github.com/mne-tools/mne-python/files/7513155/A01T2.zip)

Python-learing, 2021-11-10:
@drammock 
1. A01T2 performed electrode positioning, re-reference, ocular electrical channel removal ICA and extract epochs. It says: The number of trials is 288. It must be 1 for raw files. Please use `mne.io.read_epochs_eeglab` if the .set file contains epochs.
2. A01T4 just has channel location, reference and removal of EOG. It says buffer is small.

The data is BCI Competition IV Data sets 2a subject1. Thanks very much, sorry for the trouble caused to you. But A01T4 is bigger than the admitted file size. How can I provide it?

[A01T2.zip](https://github.com/mne-tools/mne-python/files/7513264/A01T2.zip)

Python-learing, 2021-11-10:
> After unzipping the file you provided, I tried this:
> 
> ```python
> import mne
> fname = 'A01T2/A01T2.set'
> raw = mne.io.read_raw_eeglab(fname)
> ```
> 
> This resulted in `TypeError: buffer is too small for requested array` (as expected). So then I tried
> 
> ```python
> raw = mne.io.read_raw_eeglab(fname, uint16_codec='latin1')
> ```
> 
> which, as you say, results in `TypeError: The number of trials is 288. It must be 1 for raw files. Please use mne.io.read_epochs_eeglab if the .set file contains epochs.` So, I did exactly what the message suggested:
> 
> ```python
> epo = mne.io.read_epochs_eeglab(fname, uint16_codec='latin1')
> ```
> 
> and it worked fine. There is nothing wrong with the file and no bug with MNE-Python.

thanks very much foryour answer. I used the code in another file, it says buffer is small, but I don't know how to give.

Python-learing, 2021-11-10:
> After unzipping the file you provided, I tried this:
>
> ```python
> import mne
> fname = 'A01T2/A01T2.set'
> raw = mne.io.read_raw_eeglab(fname)
> ```
>
> This resulted in `TypeError: buffer is too small for [...]

Your answer is helpful.

drammock, 2021-11-10:
> I used the code in another file, it says buffer is small, but I don't know how to give.

I'm sorry you can't get it to work with your other file. you're going to have to figure it out yourself.  Since this doesn't seem to be a problem with MNE-Python, I'm going to lock discussion on this issue thread now.  Try to find out what codec is used in that file, and change `'latin-1'` to whatever the correct codec is.  You're welcome to post future usage questions on our forum (https://mne.discourse.group) but we try to use GitHub issues only for feature requests and bugs...  as far as I can tell MNE-Python already has the feature you want (support for opening .set files saved in different codecs) but you just don't know what the correct codec is.