mne-tools/mne-python #8231: KeyError: 'zen_w_delta'

BarryLiu-97, 2020-09-10:
2. When I used mne.viz.plot_raw to draw the image of my own raw data（an eeg data from eeglab '.set'）, I came up with one KeyError: 'zen_w_delta'. The image showed up and seemed OK. While I didn't know what's this error mean.
2. And also, the image's switching was not fluent when I moved the silder. Is there any way to improve that?

agramfort, 2020-09-10:
we cannot help without access to the file. Are you sure the EEGLAB set file
is valid?

>

BarryLiu-97, 2020-09-10:
> we cannot help without access to the file. Are you sure the EEGLAB set file is valid?
> […](#)

Hi,  I think I may know what it means. In  mne.viz.plot_raw image, we can press the key 'z' in the keyboard to stop showing scrollbars. While it is not working to me when I pressed z.

BarryLiu-97, 2020-09-10:
While, I still don't know ho9w to solve this.

agramfort, 2020-09-10:
I cannot solve a problem I cannot replicate

BarryLiu-97, 2020-09-10:
> I cannot solve a problem I cannot replicate

Then do u know how to solve my second question? it is not fluent when I switch the window to different time points.

hoechenberger, 2020-09-10:
> 2\. And also, the image's switching was not fluent when I moved the silder. Is there any way to improve that?

What do you mean, exactly? That you cannot drag the slider and the window will smoothly update, as in other apps? I believe there's nothing we can do about this, as there are technical limitations.

agramfort, 2020-09-10:
> Then do u know how to solve my second question? it is not fluent when I switch the window to different time points.

raw.plot is slow if you have too many sensors and a high sampling rate. You can
limit the number of sensors you see or decimate your data

BarryLiu-97, 2020-09-10:
> What do you mean, exactly? That you cannot drag the slider and the window will smoothly update, as in other apps? I believe there's nothing we can do about this, as there are technical limitations.

Yes, that's what I mean. While the data is just about 54MB and the number of channels is 61. Is that too big?

agramfort, 2020-09-10:
it's small. It should be fluid to browse

BarryLiu-97, 2020-09-10:
> it's small. It should be fluid to browse

I can give u my code and one sample data, if that's OK，can u test it for me because I think it is a little hard to describe the situation as I code following the tutorial. Which email should I send.

agramfort, 2020-09-10:
yes

use a service that allows you to share big files

BarryLiu-97, 2020-09-10:
> yes use a service that allows you to share big files

I could send my code and sample data to you using e-mail, while could u please tell me your address?

agramfort, 2020-09-10:
how big is the file? if it's more than 20MB do not use email

BarryLiu-97, 2020-09-10:
> how big is the file? if it's more than 20MB do not use email

It is about 55MB. Google Drive？

drammock, 2020-09-10:
Yes, please upload the file to Google drive (or a similar file sharing service) and paste the link here. I will take a look (since I'm the one who wrote the "zen mode" feature).

drammock, 2020-09-10:
Please also paste the output from `mne.sys_info()`

BarryLiu-97, 2020-09-10:
> Please also paste the output from `mne.sys_info()`

https://drive.google.com/file/d/1uPHmmIWmUHnAuYjH_QXHKSh8MPaerAei/view?usp=sharing

BarryLiu-97, 2020-09-10:
mne.sys_info()
Platform:      Windows-10-10.0.19041-SP0
Python:        3.7.9 (tags/v3.7.9:13c94747c7, Aug 17 2020, 18:58:18) [MSC v.1900 64 bit (AMD64)]
Executable:    D:\sEEG_Pro\sEEG_Pro\Scripts\python.exe
CPU:           Intel64 Family 6 Model 94 Stepping 3, GenuineIntel: 8 cores
Memory:        Unavailable (requires "psutil" package)
mne:           0.20.8
numpy:         1.19.1 {blas=D:\\a\\1\\s\\numpy\\build\\openblas_info, lapack=D:\\a\\1\\s\\numpy\\build\\openblas_lapack_info}
scipy:         1.5.2
matplotlib:    3.3.1 {backend=TkAgg}
sklearn:       Not found
numba:         Not found
nibabel:       Not found
cupy:          Not found
pandas:        Not found
dipy:          Not found
mayavi:        Not found
pyvista:       Not found
vtk:           Not found

agramfort, 2020-09-10:
it works perfectly for me

>

BarryLiu-97, 2020-09-10:
> it works perfectly for me
> […](#)

no key error? maybe it's some issues about my keyboard.

BarryLiu-97, 2020-09-10:
> > it works perfectly for me
> > […](#)
> 
> no key error? maybe there're some issues about my keyboard.

And it is smooth when dragging the silder?

drammock, 2020-09-10:
I also do not get a `KeyError` when plotting or when pressing `z`.  I'm testing on mne version 0.20.8.  For me, scrolling is as smooth as expected using the left/right arrow keys.  Note that the horizontal slider does not support click-and-drag behavior; it is only possible to single-click to jump to a different part of the recording.  Could you paste the full traceback of the `KeyError`?

BarryLiu-97, 2020-09-10:
Backend TkAgg is interactive backend. Turning interactive mode on.
Exception in Tkinter callback
Traceback (most recent call last):
  File "C:\Users\Barry Liu\AppData\Local\Programs\Python\Python37\lib\tkinter\__init__.py", line 1705, in __call__
    return self.func(*args)
  File "D:\sEEG_Pro\sEEG_Pro\lib\site-packages\matplotlib\backends\_backend_tk.py", line 241, in resize
    self.resize_event()
  File "D:\sEEG_Pro\sEEG_Pro\lib\site-packages\matplotlib\backend_bases.py", line 1767, in resize_event
    self.callbacks.process(s, event)
  File "D:\sEEG_Pro\sEEG_Pro\lib\site-packages\matplotlib\cbook\__init__.py", line 229, in process
    self.exception_handler(exc)
  File "D:\sEEG_Pro\sEEG_Pro\lib\site-packages\matplotlib\cbook\__init__.py", line 81, in _exception_printer
    raise exc
  File "D:\sEEG_Pro\sEEG_Pro\lib\site-packages\matplotlib\cbook\__init__.py", line 224, in process
    func(*args, **kwargs)
  File "D:\sEEG_Pro\sEEG_Pro\lib\site-packages\mne\viz\utils.py", line 798, in _resize_event
    _update_borders(params, new_width, new_height)
  File "D:\sEEG_Pro\sEEG_Pro\lib\site-packages\mne\viz\utils.py", line 604, in _update_borders
    params['zen_w_delta'] *= old_width / new_width
KeyError: 'zen_w_delta'

drammock, 2020-09-10:
OK, I can reproduce the `KeyError` if I use the `TkAgg` matplotlib backend and run in Spyder integrated console on a Windows VM.  Still investigating.

drammock, 2020-09-10:
In a windows VM, using ipython in a regular Anaconda Prompt (not the Spyder integrated console), and using the `TkAgg` matplotlib backend, I do not get a `KeyError` and the scrolling seems fine (although note again that click-drag behavior of the scrollbar is not supported on any platform or backend).  So I think this is a Spyder problem.

BarryLiu-97, 2020-09-10:
> OK, I can reproduce the `KeyError` if I use the `TkAgg` matplotlib backend and run in Spyder integrated console on a Windows VM. Still investigating.

Got it. To be honest, I never noticed the backend of matplotlib.

BarryLiu-97, 2020-09-10:
> In a windows VM, using ipython in a regular Anaconda Prompt (not the Spyder integrated console), and using the `TkAgg` matplotlib backend, I do not get a `KeyError` and the scrolling seems fine [...]

I used pycharm. And I get that drag is not supported, the real meaning is click, I mixed up.

BarryLiu-97, 2020-09-10:
> In a windows VM, using ipython in a regular Anaconda Prompt (not the Spyder integrated console), and using the `TkAgg` matplotlib backend, I do not get a `KeyError` and the scrolling seems fine [...]

Thank u for reminding me.

BarryLiu-97, 2020-09-10:
> In Spyder, I'm able to get this to work by disabling Spyder's automatic capturing of plots into the "plot pane" (by choosing something other than "inline" in `Settings > iPython Console > Graphics > Graphics backend`). Does PyCharm have a similar setting? Does it do something similar where it tries to capture the plots into a dedicated plots pane?

I'll investigate. It's time to sleep. I'll update the progress later.

BarryLiu-97, 2020-09-10:
> In Spyder, I'm able to get this to work by disabling Spyder's automatic capturing of plots into the "plot pane" (by choosing something other than "inline" in `Settings > iPython Console > Graphics > [...]

Thank u for your help!

drammock, 2020-09-10:
I downloaded PyCharm (community edition) and cannot reproduce the `KeyError` (on Linux, running the commands in iPython in the PyCharm integrated terminal).  So I'm only able to reproduce it on Windows VM, using Spyder integrated console, with backend="inline" in the Spyder settings.  Other MPL backend settings within Spyder work fine for me (tested Qt5 and TkAgg).  Using Spyder on Linux, I get no `KeyError` with any backend (inline, Qt5, TkAgg), though interactivity is lost with backend 'inline'. I've been testing wiith the following snippet:

```python
# import matplotlib
# matplotlib.use('TkAgg')  # enable/disable as needed for testing
import mne
from mne import io, viz
fpath = '/home/username/Desktop/sample/sample.set'  # adjust as needed
raw = io.read_raw_eeglab(fpath, preload=True)
events, _ = mne.events_from_annotations(raw)
eeg_channels = raw.ch_names
n_chan = len(eeg_channels)
viz.plot_raw(raw, events, duration=6.0, n_channels=n_chan,
             title='EEG raw data')
```

I'm afraid you'll need to do further debugging yourself @BarryLiu-97.  Or simply run the script in a regular terminal outside of PyCharm.  I'll close this issue for now, but if you discover more information that might help trace the problem back to something in MNE-Python code (rather than a problem with the IDE) then feel free to re-open.

closed 2020-09-10 by drammock

hoechenberger, 2020-09-10:
> I downloaded PyCharm (community edition) and cannot reproduce the `KeyError`

PyCharm supports automatic capturing of plots similar to what Spyder offers, but to my knowledge this feature is part of the "Scientific Mode" and only available in the Professional Edition. This could be the reason why you couldn't reproduce using Community Edition.

> In the Scientific mode, a graph opens in its own tab in the SciView window, allowing you to resize it, zoom in and out, and so on. You can alter this behavior by toggling the `Show plots in tool window` checkbox (`Settings/Preferences | Tools | Python Scientific`).

https://www.jetbrains.com/help/pycharm/matplotlib-support.html#console

BarryLiu-97, 2020-09-11:
Thank u  very much