mne-tools/mne-python #8232: Baseline-correction for source estimates

olafhauk, 2020-09-10:
#### Describe the problem
Baseline-correction for source estimates is recommended/necessary for some processing steps. For example, for unsigned source estimates, differences in noise levels may result in differences in intensities (e.g. for conditions with very different numbers of trials). This can be compenstated for by baseline-correction. 

#### Describe your solution
Add a method 'baseline" to mne.SourceEstimate, equivalent to other structures (Evoked etc.).

#### Describe possible alternatives
Add a function such as mne.baseline_correct_stc().

#### Additional context
In my view the problem of different noise levels or numbers of trials for conventional source space analysis has not been highlighted enough. A while back we discussed this problem for RMS analyses of gradiometer data (https://imaging.mrc-cbu.cam.ac.uk/meg/SensorSpm, section "A note on gradiometer RMS").

agramfort, 2020-09-11:
would you just do a DC offset or a z-score?

DC will not fix the scaling issue

>

agramfort, 2020-09-11:
we can add an stc.apply_baseline method but we need to be aware that will
not
keep provenance of this step in the stc files.

I can live with that

>

hoechenberger, 2020-09-12:
Hm, I thought this had been discussed a couple of years ago, and it was decided that it might not be the best idea to have this:
https://github.com/mne-tools/mne-python/issues/962

agramfort, 2020-09-12:
@olafhauk can you have a look at discussion in https://github.com/mne-tools/mne-python/issues/962 ?

olafhauk, 2020-09-14:
I agree that baseline-correction is not recommended for noise-normalised estimates, because additivity is obviously not valid for ratios anymore. Baseline-corrections also does not remove slow drifts.

However, in my view BC is recommended for unsigned activity estimates, because the mean of rectified signals is biased by their noise levels (as e.g. for power/RMS in signal space). Comparing unsigned source estimates with different noise levels (e.g. due to different numbers of trials) may lead to spurious effects, which can reduced by baseline-correction. This only works perfectly if noise and signal are additive, which is certainly not 100% the case (especially not for long epochs).

This is ultimately up to the user. I would use this method, but it would not have to be active by default, and its documentation should discuss the pros and cons.

agramfort, 2020-09-15:
so you would suggest to use it with free orientation and method='MNE' but
discourage
it with dSPM or sLORETA?

>

olafhauk, 2020-09-17:
Yes - only method='MNE'. It could be useful for all orientation options, but recommended for unsigned estimates.
Looking at appy_inverse() now, is it possible to compute unsigned estimates for fixed orientations?
I'll have to think about beamformers (should be the same logic as long as they are estimating intensities, not variances or SNRs).
What about a function mne.source_estimate.correct_baseline() that users cannot apply by accident?

agramfort, 2020-09-17:
> Yes - only method='MNE'. It could be useful for all orientation options, but recommended for unsigned estimates.
> Looking at appy_inverse() now, is it possible to compute unsigned estimates for fixed orientations?

you can do stc.abs() to get unsigned from signed.

> I'll have to think about beamformers (should be the same logic as long as they are estimating intensities, not variances or SNRs).
> What about a function mne.source_estimate.correct_baseline() that users cannot apply by accident?

stc.apply_baseline method would be more natural with a clear docstring

hoechenberger, 2020-09-17:
> stc.apply_baseline method would be more natural with a clear docstring

+1

olafhauk, 2020-09-22:
Should I have a go?

agramfort, 2020-09-22:
ok for me

>

hoechenberger, 2021-02-01:
Closed via #8452

closed 2021-02-01 by hoechenberger