mne-tools/mne-python #8237: finding EOG/ ECG  artifacts

XPeriment2, 2020-09-10:
trying to use on EEG 60 channels data:
ecg_epochs = create_ecg_epochs(baseline)

without having a specific ecg channel, however it is not able to calculate a ECG channel.

create_eog_epochs(baseline, dict(mag=5e-12, grad=4000e-13),
                               picks='eeg').average()
also, not having an eog channel, no EOG channels are found using this.
When i try to specify a frontal channel in chan_name, same error.
I guess cannot be done if it's not MEG?

thanks

XPeriment2, 2020-09-10:
same thing happens with SSP on
compute_proj_ecg
compute_proj_eog

agramfort, 2020-09-11:
yes these functions do require to have an ECG or EOG channel to identify the artefact automatically.

closed 2020-09-11 by agramfort