mne-tools/mne-python #8241: BUG: CircleCI using too much memory (again)

larsoner, 2020-09-11:
A lot of our `master` builds have failed lately, like:

https://app.circleci.com/pipelines/github/mne-tools/mne-python/3597/workflows/fcb12f01-1f51-4d87-8697-5ac9e30f9411/jobs/22228

@GuillaumeFavelier I suspect it could be some recent-ish (within the last 2 months?) viz update. It was happening before the PyVista 0.26 release. Can you look to see if code from a month or two ago uses the same memory as it does now? Often it's `plot_visualize_stc.py` that breaks things, but in the example above it's `plot_forward.py`.

GuillaumeFavelier, 2020-09-12:
It's an item of #7162

larsoner, 2020-09-18:
Turns out this seems to be related to bad garbage collection. See investigations in https://github.com/mne-tools/mne-python/pull/8266#issuecomment-694561692 and report in https://github.com/pyvista/pyvistaqt/issues/57. It would be good if we could figure out where this is coming from. In the meantime, plotting so many montages is not going to be safe :(

larsoner, 2020-10-20:
@GuillaumeFavelier interestingly enough this does not seem to happen on https://github.com/mne-tools/mne-python/blob/master/examples/visualization/plot_meg_sensors.py#L40-L47, but it does in https://github.com/mne-tools/mne-python/blob/master/examples/visualization/montage.py#L46-L60. Maybe this can help us narrow down where the problem is.

On/after #8379 and https://github.com/sphinx-gallery/sphinx-gallery/pull/759 we can iterate and test hopefully fairly easily using `PATTERN=... make html_dev-pattern-memory`

larsoner, 2020-10-21:
Never mind, it does happen on `plot_meg_sensors.py`...

larsoner, 2023-10-01:
This is an upstream issue so I'll close at our end

PR #8254 (2020-09-14): MRG, MAINT: Reduce memory usage of examples
Closes #8241

The goal is to reduce memory usage -- we'll see what CircleCI reports here vs on `master`. But some relevant test timings locally on master:
```
45.42s total   mne/minimum_norm/tests/
37.23s total   mne/tests/test_source_space.py
21.81s total   mne/forward/tests/
 0.00s total   mne/forward/_make_forward.py
```
This PR:
```
47.28s total   mne/minimum_norm/tests/
37.83s total   mne/tests/test_source_space.py
20.52s total   mne/forward/tests/
 0.00s total   mne/forward/_make_forward.py
```
So it looks like removing the `preload=True` from  [... cut]
closed 2023-10-01 by larsoner