mne-tools/mne-python #8250: mne read_proj Unbound local error 'global_nchan' referenced before assignment

DariusRH, 2020-09-13:
Dear All,

I’m trying to clean any heartbeat related artifacts from an EEG+ECG recording. To do this I’m computing a signal spatial projection for the ECG data recorded in a Bio channel. Then I created a function to save this projection as a fif file then using the saved file, plot the projection as a topomap. Creating the projection and saving it as a fif seem to be working, but an unbound local error is generated in reading the projection file.

Here's the code generating the issue.

import mne; mne.sys_info()
import pandas as pd
import numpy as np
import matplotlib.pyplot as plt
import os
from mne import read_proj

sfreq = 500
l_freq, h_freq = 0.1, 40
tmin, tmax = 0, 60
fmin, fmax = 0.5, 250
n_fft = 2048

channel_names = ['C3', 'C4', 'Cz', 'F3', 'F4', 'F7', 'F8', 'Fp1', 'Fp2', 'Fz', 'O1', 'O2', 'Oz', 'P3', 'P4', 'P7', 'P8', 'Pz', 'T7', 'T8', 'Bio3', 'VSyn', 'ASyn', 'LABEL']
EEG_channels = ['C3', 'C4', 'Cz', 'F3', 'F4', 'F7', 'F8', 'Fp1', 'Fp2', 'Fz', 'O1', 'O2', 'Oz', 'P3', 'P4', 'P7', 'P8', 'Pz', 'T7', 'T8']
channel_types = ['stim', 'meg', 'eeg', 'eeg', 'eog']

os.chdir('C:/Users/6/TEST')

raw = mne.io.read_raw_edf(r'C:/Users/6/TEST/ANFI0001_2020-07-30_M45_Joe_SMITH_EOEC_PRE.edf', eog=None, misc=None, stim_channel='auto', exclude=('VSyn', 'ASyn', 'LABEL'), preload=True, verbose=None)

raw.set_channel_types({'Bio3': 'ecg'})

ecg_projs = mne.preprocessing.compute_proj_ecg(raw, raw_event=None, tmin=- 0.2, tmax=0.4, n_grad=2, n_mag=2, n_eeg=2, l_freq=1.0, h_freq=40.0, average=True, filter_length='10s', n_jobs=1, ch_name=None, reject={'mag': 3e-12}, flat=None, bads=[], avg_ref=False, no_proj=False, event_id=999, ecg_l_freq=5, ecg_h_freq=35, tstart=0.0, qrs_threshold='auto', filter_method='fir', iir_params=None, copy=True, return_drop_log=False, meg='separate', verbose=None)

global ecg_fname
def create_ecg_fif(ecg_fname):
os.path(r'C:/Users/6/TEST/heartbeat-proj.fif')
mne.write_proj(ecg_fname, ecg_projs)
return ecg_fname
ecg_fname = create_ecg_fif

ecg_projs = read_proj(r'C:/Users/6/TEST/heartbeat-proj.fif')
raw.add_proj(ecg_projs)
raw.plot_projs_topomap()

In plotting the projection I'm getting the following error:

Traceback (most recent call last):

File "C:\Users\6\TEST\untitled0.py", line 80, in
ecg_projs = read_proj(r'C:/Users/6/TEST/heartbeat-proj.fif')

File "C:\Anaconda3\envs\mne\lib\site-packages\mne\proj.py", line 47, in read_proj
projs = _read_proj(fid, tree)

File "", line 21, in _read_proj

File "C:\Anaconda3\envs\mne\lib\site-packages\mne\io\proj.py", line 342, in _read_proj
nchan = global_nchan

UnboundLocalError: local variable 'global_nchan' referenced before assignment

Any help would be greatly appreciated. Thanks in advance.

agramfort, 2020-09-13:
can you share with me the edf file so I can replicate the issue?

>

DariusRH, 2020-09-13:
Sure, the edf, proj and py files are in this zip folder. 

[read proj issue.zip](https://github.com/mne-tools/mne-python/files/5215272/read.proj.issue.zip)

Thanks for helping with this.

larsoner, 2020-09-14:
@DariusRH I cannot replicate with this more minimal example:
```
# -*- coding: utf-8 -*-
import mne
raw = mne.io.read_raw_edf('ANFI0001_2020-07-30_M45_Joe_SMITH_EOEC_PRE.edf',
                          eog=None, misc=None, stim_channel='auto',
                          exclude=('VSyn', 'ASyn', 'LABEL'), preload=True)
raw.set_channel_types({'Bio3': 'ecg'})
ecg_projs, _ = mne.preprocessing.compute_proj_ecg(
    raw, raw_event=None, tmin=- 0.2, tmax=0.4, n_grad=2, n_mag=2, n_eeg=2, l_freq=1.0, h_freq=40.0, average=True, filter_length='10s', n_jobs=1, ch_name=None, reject={'mag': 3e-12}, flat=None, bads=[], avg_ref=False, no_proj=False, event_id=999, ecg_l_freq=5, ecg_h_freq=35, tstart=0.0, qrs_threshold='auto', filter_method='fir', iir_params=None, copy=True, return_drop_log=False, meg='separate', verbose=None)
mne.write_proj('test-proj.fif', ecg_projs)
ecg_projs_read = mne.read_proj('test-proj.fif')
```
Can you see if this runs at your end? I think your script had some errors, for example you try to save the output `ecg_projs` from `ecg_projs = mne.preprocessing.compute_proj_ecg(...)`, but there are actually two outputs, so you are saving the wrong thing (you either want to use `ecg_projs, _ = mne.preprocessing.compute_proj_ecg(...)` as I do above or `ecg_projs = mne.preprocessing.compute_proj_ecg(...)[0]`). It makes me wonder if somehow the `write_projs` function got partway done saving the projectors and that's why you end up with a weird problem...

Also, feel free to try https://github.com/mne-tools/mne-python/pull/8253 to see if it fixes your problem in your other script.

PR #8253 (2020-09-14): MRG, MAINT: Better error for nchan
Closes #8250.

It's not a great idea to assign to a var that might not have been created yet, so ensure it's always there. Then if it's still `None`, infer from the channel name list. And either way, check to make sure that the length of the channel name list is consistent with the listed number of channels.

No need for `latest.inc` update I think as this just cleans up some error messages / corner cases.
closed 2020-09-14 by larsoner