mne-tools/mne-python #8251: tmin and tmax is Error if edf_info record_length

ddbbqqpp, 2020-09-13:
**# Get Edf infos**

info, edf_info, orig_units = mne.io.edf.edf._get_info(fname=psg_fpath, stim_channel='auto',
                                                      eog=None, misc=None,
                                                      exclude=(), preload=False)
print(edf_info)

**# Result**

{
    'events': [],
    'units': [1e-06, 1e-06, 1e-06, 1e-06, 1e-06, 1e-06, 1e-06, 1e-06, 1e-06, 1e-06, 1e-06, 1e-06, 1e-06, 1e-06, 1e-06, 1e-06, 1e-06, 1e-06, 1e-06, 1e-06, 1e-06, 1e-06, 1e-06, 1e-06, 1e-06, 1e-06, 1e-06, 1e-06, 1e-06, 1e-06, 1],
    'ch_names': ['EOG Left Horiz', 'EOG Right Horiz', 'EMG Chin1', 'EMG Chin2', 'EMG Chin3', 'EEG F3-CLE', 'EEG F4-CLE', 'EEG C3-CLE', 'EEG C4-CLE', 'EEG O1-CLE', 'EEG O2-CLE', 'ECG ECGI', 'EMG Ant Tibial L', 'EMG Ant Tibial R', 'Resp Thermistor', 'Resp Cannula', 'Resp Belt Thor', 'Resp Belt Abdo', 'EEG F7-CLE', 'EEG F8-CLE', 'EEG T3-CLE', 'EEG T4-CLE', 'EEG T5-CLE', 'EEG T6-CLE', 'EEG P3-CLE', 'EEG P4-CLE', 'EEG Fz-CLE', 'EEG Cz-CLE', 'EEG Pz-CLE', 'EEG A2-CLE', 'SaO2 SaO2'],
    'data_offset': 8448,
    'digital_max': array([32767., 32767., 32767., 32767., 32767., 32767., 32767., 32767.,
            32767., 32767., 32767., 32767., 32767., 32767., 32767., 32767.,
            32767., 32767., 32767., 32767., 32767., 32767., 32767., 32767.,
            32767., 32767., 32767., 32767., 32767., 32767., 32767.]),
    'digital_min': array([-32768., -32768., -32768., -32768., -32768., -32768., -32768.,
            -32768., -32768., -32768., -32768., -32768., -32768., -32768.,
            -32768., -32768., -32768., -32768., -32768., -32768., -32768.,
            -32768., -32768., -32768., -32768., -32768., -32768., -32768.,
            -32768., -32768., -32768.]),
    'highpass': array([], dtype = float64),
    'sel': array([0, 1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15, 16,
            17, 18, 19, 20, 21, 22, 23, 24, 25, 26, 27, 28, 29, 30]),
    'lowpass': array([], dtype = float64),
    'meas_date': datetime.datetime(2000, 1, 1, 22, 3, 21, tzinfo = datetime.timezone.utc),
    'n_records': 15277,
    'n_samps': array([512, 512, 512, 512, 512, 512, 512, 512, 512, 512, 512,
            1024, 512, 512, 256, 256, 256, 256, 512, 512, 512, 512,
            512, 512, 512, 512, 512, 512, 512, 512, 512, 8]),
    'nchan': 32,
    'subject_info': {
        'id': 'X',
        'name': 'X'
    },
    'physical_max': array([730., 728., 272., 267., 272., 712., 696., 700., 697., 700., 690.,
            486., 269., 268., 532., 535., 243., 242., 713., 711., 714., 714.,
            708., 719., 713., 718., 718., 708., 719., 238., 102.]),
    'physical_min': array([-730., -728., -272., -267., -272., -712., -696., -700., -697.,
            -700., -690., -486., -269., -268., -532., -535., -243., -242.,
            -713., -711., -714., -714., -708., -719., -713., -718., -718.,
            -708., -719., -238., -102.]),
    'record_length': array([2., 1.]),
    'subtype': 'edf',
    'tal_idx': array([31], dtype = int64),
    'dtype_byte': 2,
    'dtype_np':  < class 'numpy.int16' > ,
    'stim_channel': None,
    'max_samp': 1024,
    'nsamples': 15643648
}

**Notice**

_'record_length': array([2., 1.])_
the duration seconds twice as much as n_records or nsamples / max_samp

So, can't get the data, Because
raise ValueError(...) about of tmin and tmax

        max_time = (self.n_times - 1) / self.info['sfreq']
        if tmax is None:
            tmax = max_time

        if tmin > tmax:
            raise ValueError('tmin (%s) must be less than tmax (%s)'
                             % (tmin, tmax))
        if tmin < 0.0:
            raise ValueError('tmin (%s) must be >= 0' % (tmin,))
        elif tmax > max_time:
            raise ValueError('tmax (%s) must be less than or equal to the max '
                             'time (%0.4f sec)' % (tmax, max_time))

Docs:
tmin: Start time of the raw data to use in seconds (must be >= 0).
tmax: End time of the raw data to use in seconds (cannot exceed data duration).

ddbbqqpp, 2020-09-13:
base.py
max_time = (self.n_times - 1) / self.info['sfreq']
change to
max_time = self.times[-1]

agramfort, 2020-09-13:
can you send us a pull request with the fix? thanks

cbrnr, 2020-09-14:
@ddbbqqpp I don't understand where the error occurs. Is it after the initial code snippet (`mne.io.edf.edf._get_info`)? Or when loading the EDF file (`mne.io.read_raw_edf`)? Could you please post a minimal example that demonstrates the issue? It might also be necessary to share the specific EDF file, because I haven't come across this error with other EDF files so far.

ddbbqqpp, 2020-09-16:
Maybe I made a mistake, Calculated  the duration

closed 2020-09-16 by ddbbqqpp