mne-tools/mne-python #8257: cannot add eeg channels/data into meg file

ometett, 2020-09-15:
Dear MNE team,

I hope you are doing alright. I am an enthusiastic bioengineering student who is working on a tool for MEG/HD-EEG analysis. I have the eeg and meg recording from the same patient and I want to add the eeg data/channels into the meg file to be able to analyze them together under one .fif file. I have converted my eeg recording(.edf) to .fif format so they are in the same format now.  I am trying mne.concatenate_raws(MEG,EEG) to merge them together but it gives an error message "AttributeError: 'tuple' object has no attribute 'append'".  Is there anyway that I can add eeg data&channels into the meg file even though channel types are different? Any idea and help is very much appreciated.

Best regards,
Omer
[image]

hoechenberger, 2020-09-15:
Hello @ometett,

great to hear you're trying to get started with MNE! It's going to be an amazing, but sometimes bumpy journey ;)

> I am trying mne.concatenate_raws(MEG,EEG) to merge them together but it gives an error message "AttributeError: 'tuple' object has no attribute 'append'

The first argument to `concatenate_raw` should be a list or tuple of Raw, so the correct way to invoke it is sth like:
```python
mne.concatenate_raw([MEG, EEG])
```

But that doesn't really achieve what you want to achieve either.

I **suppose** what you would have to do is:
- Add new "empty" channels channels to your MEG data via [`MEG.add_channels`](https://mne.tools/dev/generated/mne.io.Raw.html#mne.io.Raw.add_channels)
- Change the type of those channels to "meg" via [`MEG.Raw.set_channel_types`](https://mne.tools/dev/generated/mne.io.Raw.html#mne.io.Raw.set_channel_types)
- Copy the EEG data (`EEG.get_data()`) to  `MEG._data` in the correct place

All of this can be a little tricky, but it's doable. 

Or am I missing a simple, obvious alternative here?

larsoner, 2020-09-15:
> Add new "empty" channels channels to your MEG data via MEG.add_channels
> Change the type of those channels to "meg" via MEG.Raw.set_channel_types
> Copy the EEG data (EEG.get_data()) to MEG._data in the correct place

People should not have to touch private attributes like `_data`, it's dangerous. In this case, `MEG.add_channels([EEG])` should work.

Also, make sure `type(MEG)` and `type(EEG)` are both `RawFIF`. I think in your code MEG is probably a `tuple`, which suggests maybe it was obtained from the output of another function and you need to split the output like `MEG, somethingelse = func(...)` or `MEG = func(...)[0]` or so.

hoechenberger, 2020-09-15:
> . In this case, `MEG.add_channels([EEG])` should work.

Wow, really? That's great to know!

ometett, 2020-09-15:
Dear @larsoner  and @hoechenberger,

Thank you for the quick response. I am glad. Both MEG and EEG are the same type(fiff.raw.Raw) however, it's giving me an error "AssertionError: All data dimensions except channels must match, got [632000] != [623660]" I suspect this is because the eeg recording is slightly longer than the meg recording. Could cutting the portion of my eeg recording from its tail possibly solve the problem? I appreciate your time and continuous help.

Best regards,
Omer
[image]

larsoner, 2020-09-15:
Yes, but you need to ensure they are aligned in time. Whether this means cropping the beggining, ending, or resampling to deal with slight clock differences depends on your actual data...

I would use the timing of triggers sent to both systems to determine what you need to do. When working on this sort of problem I've observed linear drifts in the clocks of the two systems, meaning that cropping both files and then resampling one of them on the basis of some shared signal like this is necessary.

hoechenberger, 2020-09-15:
> "AssertionError: All data dimensions except channels must match, got [632000] != [623660]" I suspect this is because the eeg recording is slightly longer than the meg recording. Could cutting the [...]

Were these **actually** simultaneous recordings? Are you sure they started at **precisely** the same time (i.e. with an offset of less than one milli-second)? Then it should be okay to just crop them to the same duration. Otherwise – no. You'd have to find a marker that was sent to both recording systems at the precisely same moment to align them.

Btw could you please try to include error messages as text here? :) It's already great that you **do** include the complete messages, but having them as text would be even better. See https://docs.github.com/en/github/writing-on-github/creating-and-highlighting-code-blocks#syntax-highlighting for details. Thanks!

ometett, 2020-09-15:
Dear @larsoner and @hoechenberger  ,

Thank you for the quick response. I have checked their sampling frequency and both of them are sampled in 1000 Hz. I have somehow made it work by cutting it from the end making their durations equal. At first it gave me an error message.

  File "C:\For Omer\work.py", line 74, in <module>
    MEG.add_channels([EEG3],force_update_info= False)

  File "C:\Users\tanritaniro\Anaconda3\lib\site-packages\mne\channels\channels.py", line 1036, in add_channels
    new_info = _merge_info(infos, force_update_to_first=force_update_info)

  File "<decorator-gen-27>", line 21, in _merge_info

  File "C:\Users\tanritaniro\Anaconda3\lib\site-packages\mne\io\meas_info.py", line 1901, in _merge_info
    raise ValueError(msg)

ValueError: Measurement infos provide mutually inconsistent dev_head_t

I got rid of this error by setting force_update_info = True and it worked. I checked the new .fif file in brainstorm, the eeg channels and data is there. Would forcefully adding the channels like that have any side effects on the data? Thank you so much for enlightening me!

Regards,

larsoner, 2020-09-15:
> I have checked their sampling frequency and both of them are sampled in 1000 Hz. 

I'm not 100% sure this is enough. Quoting an email I sent to someone else about this at some point:

> I am no electronics expert but some [quick searches](https://www.hindawi.com/journals/jcnc/2008/583162/) suggest that something like 1-6 microseconds per second are typical drift rates for computer-grade quartz crystals...
>
> We recorded some basic data a while back -- it had exactly two triggers, each sent to both machines: one at the beginning and one at the end of a ~544 second recording. Having (at least) two reference points allows us to estimate both a fixed offset and a linear drift term. The MEG (FIF) data showed that the two triggers were 544.045 seconds apart, whereas the [EEG EGI] MFF events were 544.052 seconds apart, indicating a linear drift of 7 ms over the 544 second recording, or a ~12 microsecond clock drift between the two devices per second of recording time. With two clocks (each of which could be +/- 6 ms from correct), the ~12 microsecond value is at least in the ballpark of being reasonable.

So I'd advise at least checking for drift, too. It's entirely possible I'm misunderstanding how these things work -- my data only had two triggers, too, which is not great -- but you might save yourself some pain later by ensuring you have some fixed known signal (like a trigger) the same across both devices, and checking that just cropping keeps them synchronized.

ometett, 2020-09-18:
Dear @larsoner ,

You were right. Cropping EEG did not entirely fix the problem. In the EEG, I detected interictal spikes and gathered them under events, when I imported those time points into the EEG that I added inside the MEG file, I found that there was no spikes in those points, more specifically those time points were around 20 seconds behind( if first spike occured at 56 in original EEG, that corresponds to 33s in the cropped EEG which makes sense because EEG recording was 623 seconds and MEG 606 so I technically shifted EEG by 17 seconds.  I have changed the sampling rate of the EEG to overcome this but it did not solve the problem. Is there anyway to overcome the shifting that cropping this recording causes? Thank you so much for the time and ideas!

Regards,

drammock, 2020-09-18:
@ometett you need to do what @larsoner [already suggested above](https://github.com/mne-tools/mne-python/issues/8257#issuecomment-692768824): Work with the uncropped versions of EEG and MEG, find corresponding events in the two files (preferably one near the beginning and one near the end of the recordings), and use those to calculate two values:

- an offset (how much should be cropped from the beginning, and should it be cropped from the EEG or the MEG file?)
- a scaling factor (given 2 events that are the same distance apart *in real clock time*, how many samples apart do they appear in the two files?  The ratio of those two numbers can be used to calculate a resampling rate that should bring the events into alignment)

ometett, 2020-09-18:
@drammock @larsoner  I have done what you guys suggested. I have found the offset and scale value based on the trigger points at both recordings, then I used that value to calculate the sampling rate of the EEG so that they both end up at the same duration which 606 seconds, but I am still having the same results and in addition to that, I have lost data in a lot of channels, there is just noise, I can not view them in brainstorm.  Is there a risk of losing channel data when we manually import them to another file using add.channels?

Regards,

larsoner, 2020-09-18:
> I have found the offset and scale value based on the trigger points at both recordings, then I used that value to calculate the sampling rate of the EEG so that they both end up at the same duration which 606 seconds

The way you are describing this, it sounds like you just resampled. You almost certainly will need to crop the start of one file, the end of the other, and (maybe) resample one of them, unless you were very careful about starting/stopping them at the same times.

Did you check that the "markers" you were using were properly aligned after your alignment procedure?

>  I have lost data in a lot of channels, there is just noise, I can not view them in brainstorm.

I don't understand how this should be possible with just `.crop` and `raw.resample` being used. Did you do something else? Did you look at your instances in MNE-Python with `raw.plot` to make sure they looked okay after each step?

You'll probably need to do a bit of digging with the data yourself to figure out what's going on. Or, if you want to share the data on DropBox or something similar, it's possible someone else could have a look.

hoechenberger, 2020-10-07:
Hello @ometett, did you manage to resolve your issue?

ometett, 2020-10-08:
Hi @hoechenberger , yes I am positive that I solved my problem. I wrote my code and now I am testing on different data files to test it's accuracy. Thank you and all the MNE team for being supportive and helpful. I appreciate it!

closed 2020-10-08 by ometett

moritz-gerster, 2022-07-01:
> Hi @hoechenberger , yes I am positive that I solved my problem. I wrote my code and now I am testing on different data files to test it's accuracy. Thank you and all the MNE team for being supportive and helpful. I appreciate it!

Dear @ometett, I have the same problem. If you still have your code that solved the synchronization between EEG and MEG and if you could paste it here, that would be a great help! :)

larsoner, 2022-07-01:
https://mne.tools/dev/generated/mne.preprocessing.realign_raw.html should work