mne-tools/mne-python #8258: Restrict allowed "verbose" values, or make the verbose kwarg a keyword-only argument?

hoechenberger, 2020-09-15:
Call me stupid but I just spent 5 minutes trying to "debug" why this wasn't doing as I expected:

```python
epochs.apply_baseline(None, 0)
```

Obviously, the baseline period should be specified as a tuple… What happened here is that instead of throwing an error message, I ended up with Epochs without baseline correction. Signature of `apply_baseline` is:
```python
    @verbose
    def apply_baseline(self, baseline=(None, 0), verbose=None):
```
So `verbose` was simply set to `0`.

While you could always argue that of course you should check your results and that you cannot make a software fool-proof, I was still wondering if we couldn't easily prevent some of those "user bugs" from happening. In this particular case, either being more restrictive on the allowed values of `verbose`, or declaring `verbose` to be a keyword-only argument would have successfully mitigated my inaptness :)

```python
    def apply_baseline(self, baseline=(None, 0), *, verbose=None):
```

Thoughts?

cbrnr, 2020-09-15:
I like both suggestions: make `verbose` a kw-only argument and restrict its values.

hoechenberger, 2020-09-15:
There's a third (complementary) solution: at one point, we should start discussing adding type annotations.

cbrnr, 2020-09-15:
Let's try to move this point as far as possible into the future...

larsoner, 2020-09-15:
I'm fine with adding the `*` to functions where `verbose` is close to the other arguments. We can even just start with `apply_baseline` for 0.21 as I agree it's a likely candidate for this problem.

I'm also fine with making the `verbose` decorator more picky, but let's do that early in 0.22

hoechenberger, 2020-09-15:
> I'm fine with adding the `*` to functions where `verbose` is close to the other arguments. We can even just start with `apply_baseline` for 0.21 as I agree it's a likely candidate for this problem.

I can do this later tonight I believe.

agramfort, 2020-09-15:
remind me of

np.zeros(2, 3) :)

>

hoechenberger, 2020-09-15:
> np.zeros(2, 3) :)

I **literally** had to run this code to figure out what's wrong with it. Ouch.

PR #8350 (2020-10-07): MRG, BUG: Force keywords for verbose in short sigs
Closes #8258 
Closes #8263
closed 2020-10-07 by hoechenberger