mne-tools/mne-python #8267: Joint time-interactive evoked plots

flecaignard, 2020-09-16:
Hi everyone,

I need a tool to assess both in time and in space the differences between two (or more) evoked responses. To do so, so far, I've used evoked.plot_topomap(times='interactive') for eahc of my evoked object. This is  really not convenient in the long run because then I need to synch manually the time slider of each map, and I need to adjust colorscales too..
So I was wondering whether it could be possible to augment plot_topomap with joint maps?
here is a screenshot of two figures that I've synch manually, I hope it makes my point clearer ;-)
Many thanks for your help/suggestions!
françoise
[image]

mmagnuski, 2020-11-20:
As far as I know this is not possible currently and I'm not sure if we have public functions available to achieve something like this easily (@drammock should have an idea). You can however create an ERP difference using [`mne.combine_evoked`](https://mne.tools/stable/generated/mne.combine_evoked.html) and ivestigate it interactively as you do for single evokeds.
To specify the same time and color limits for multiple evokeds you can also use code like the one below (assuming you have a list of the evokeds you want to plot in `all_evokeds` variable):
```python
import numpy as np
import matplotlib.pyplot as plt
import mne

# the window and color limits have to be specified:
window_limits = [0.35, 0.45]
vmin, vmax = -5.5, 5.5

# the rest should work well for up to about 8 evokeds in all_evokeds list
window_center = np.mean(window_limits)
window_width = np.diff(window_limits)[0]

n_evokeds = len(all_evokeds)
fig, ax = plt.subplots(ncols=n_evokeds, figsize=(n_evokeds * 2, 3))

for evoked, this_ax in zip(all_evokeds, ax):
    evoked.plot_topomap(times=window_center, average=window_width,
                        axes=this_ax, vmin=vmin, vmax=vmax, show=False,
                        colorbar=False)
```

drammock, 2020-11-20:
Agreed that this is currently not possible in a built-in way.  But it seems like a reasonable use case to consider supporting. I'll add it to the list.  crossref #7751

larsoner, 2023-10-01:
We now have this functionality, see https://mne.tools/dev/auto_tutorials/visualization/20_ui_events.html (thanks @wmvanvliet , found another issue you fixed!)

closed 2023-10-01 by larsoner