mne-tools/mne-python #8270: plot_psd crashes when ch_types is not indicated in info

chapochn, 2020-09-16:
#### Describe the bug
plot_psd does not work when ch_types is not indicated in info

#### Steps to reproduce
```
import numpy as np
import mne
data = np.arange(1000).reshape(2, -1)
info = mne.create_info(ch_names=['a', 'b'], sfreq=1000)
raw = mne.io.RawArray(data, info)
raw.plot_psd(fmax=200)
```

Whereas this works
```
info = mne.create_info(ch_names=['a', 'b'], sfreq=1000, ch_types='ecog')
raw = mne.io.RawArray(data, info)
raw.plot_psd(fmax=200)
```

#### Expected results
expected to see the plot

#### Actual results
Traceback (most recent call last):
  File "<input>", line 6, in <module>
  File "<decorator-gen-156>", line 21, in plot_psd
  File "/Users/.../opt/anaconda3/envs/mne/lib/python3.8/site-packages/mne/io/base.py", line 1388, in plot_psd
    return plot_raw_psd(self, fmin=fmin, fmax=fmax, tmin=tmin, tmax=tmax,
  File "<decorator-gen-139>", line 21, in plot_raw_psd
  File "/Users/.../opt/anaconda3/envs/mne/lib/python3.8/site-packages/mne/viz/raw.py", line 622, in plot_raw_psd
    make_label, xlabels_list = _set_psd_plot_params(
  File "/Users/.../opt/anaconda3/envs/mne/lib/python3.8/site-packages/mne/viz/utils.py", line 3053, in _set_psd_plot_params
    picks = _picks_to_idx(info, picks)
  File "/Users/.../opt/anaconda3/envs/mne/lib/python3.8/site-packages/mne/io/pick.py", line 984, in _picks_to_idx
    picks = _picks_str_to_idx(info, picks, exclude, with_ref_meg,
  File "/Users/.../opt/anaconda3/envs/mne/lib/python3.8/site-packages/mne/io/pick.py", line 1040, in _picks_str_to_idx
    raise ValueError('picks (%s) yielded no channels, consider '
ValueError: picks (NoneNone, treated as "data") yielded no channels, consider passing picks explicitly

#### Additional information
Platform:      macOS-10.14.6-x86_64-i386-64bit
Python:        3.8.5 (default, Aug  5 2020, 03:39:04)  [Clang 10.0.0 ]
Executable:    /Users/.../opt/anaconda3/envs/mne/bin/python
CPU:           i386: 12 cores
Memory:        16.0 GB
mne:           0.20.8
numpy:         1.19.1 {blas=NO_ATLAS_INFO, lapack=lapack}
scipy:         1.5.2
matplotlib:    3.3.1 {backend=Qt5Agg}
sklearn:       0.23.2
numba:         0.51.2
nibabel:       3.1.1
cupy:          Not found
pandas:        1.1.1
dipy:          1.1.0
mayavi:        4.7.2 {qt_api=pyqt5, PyQt5=5.12.3}
pyvista:       0.25.3
vtk:           8.2.0

larsoner, 2020-09-16:
Thanks for the report! However, this is intended behavior. Channel types are important in MNE, and you should set them to the correct types. In this case, the error message about "no data channels" seems to have steered you in the right direction, namely of setting the channel type of your data to ECoG.

closed 2020-09-16 by larsoner

chapochn, 2020-09-16:
This crashes too:

```
info = mne.create_info(ch_names=['a', 'b'], sfreq=1000, ch_types='misc')
raw = mne.io.RawArray(data, info)
raw.plot_psd(fmax=200)
```
Is that also intended? With exactly the same error message.

larsoner, 2020-09-16:
Yes, because there are no data channels. If you really want plot_psd of your misc channels pass `plot_psd(fmax=200, picks='all')` or `plot_psd(fmax=200, picks='misc')` or so. The default (None) picks "good data channels":

https://mne.tools/dev/generated/mne.io.Raw.html#mne.io.Raw.plot_psd

And there are no "data channels" at all if all of your channels are MISC type

larsoner, 2020-09-16:
See also 

https://mne.tools/dev/glossary.html#term-data-channels

chapochn, 2020-09-16:
Thanks for the links and explanations!
Neither `plot_psd(fmax=200, picks='all')` nor `plot_psd(fmax=200, picks='misc')` works actually... Gives the error: 'No data channels found'
I was wondering if there would be some channel with just "arbitrary units" or something like that?

drammock, 2020-09-17:
This works for me:

```python
import numpy as np
import mne
data = np.arange(1000).reshape(2, -1)
info = mne.create_info(ch_names=['a', 'b'], sfreq=1000)
raw = mne.io.RawArray(data, info)
raw.plot_psd(fmax=200, picks='misc')
```

[image]

If it doesn't work for you than please report the results of `mne.sys_info()`

chapochn, 2020-09-17:
It doesn't work for me. I still get a "RuntimeError('No data channels found')". Here is my config:

Platform:      Linux-5.4.0-47-generic-x86_64-with-glibc2.10
Python:        3.8.5 | packaged by conda-forge | (default, Aug 29 2020, 01:22:49)  [GCC 7.5.0]
Executable:    /home/.../Programs/anaconda3/envs/mne/bin/python
CPU:           x86_64: 4 cores
Memory:        15.4 GB
mne:           0.20.8
numpy:         1.19.1 {blas=NO_ATLAS_INFO, lapack=lapack}
scipy:         1.5.2
matplotlib:    3.3.1 {backend=module://backend_interagg}
sklearn:       0.23.2
numba:         0.51.2
nibabel:       3.1.1
cupy:          Not found
pandas:        1.1.1
dipy:          1.1.0
mayavi:        4.7.2 {qt_api=pyqt5, PyQt5=5.12.3}
pyvista:       0.25.3
vtk:           8.2.0

bloyl, 2020-09-17:
This works on master, but does not work on 0.20.8

There is a new release scheduled for the next few weeks, which should solve
the problem, alternatively you could upgrade to the development version
using

pip install -U https://github.com/mne-tools/mne-python/archive/master.zip

chapochn, 2020-09-17:
ok thanks!

chapochn, 2020-09-20:
So, just curious, before reporting an issue, should I always test on master first instead of the current stable version?

larsoner, 2020-09-21:
Yeah usually that's a good idea