mne-tools/mne-python #8271: 'NoneType' object has no attribute '_Renderer'

anupdas777, 2020-09-17:
I am using the following code to plot ECoG electrodes on a brain surface using MNE toolbox.

montage = mne.channels.make_dig_montage(elec_dict, coord_frame='head')
info = mne.create_info(ch_names, 1000., 'ecog').set_montage(montage)
subjects_dir = '/home1/anup.das/mne_data/MNE-sample-data/subjects/'
fig = plot_alignment(info, subject='fsaverage', subjects_dir=subjects_dir, surfaces=['pial'])

However I get the error "'NoneType' object has no attribute '_Renderer'" while executing the last line "fig = plot_alignment(info, subject='fsaverage', subjects_dir=subjects_dir, surfaces=['pial'])". 

ch_names is as follows:

['LFPG1',
 'LFPG2',
 'LFPG3',
 'LFPG4',
 'LFPG5',
 'LFPG6',
 'LFPG7',
 'LFPG8',
 'LFPG9',
 'LFPG10',
 'LFPG11',
 'LFPG12',
 'LFPG13',
 'LFPG14',
 'LFPG15',
 'LFPG16',
 'LFPG17',
 'LFPG18',
 'LFPG19',
 'LFPG20',
 'LFPG21',
 'LFPG22',
 'LFPG23',
 'LFPG24',
 'LFPG25',
 'LFPG26',
 'LFPG27',
 'LFPG28',
 'LFPG29',
 'LFPG30',
 'LFPG31',
 'LFPG32',
 'LSPIS1',
 'LSPIS2',
 'LSPIS3',
 'LSPIS4',
 'LIPIS1',
 'LIPIS2',
 'LIPIS3',
 'LIPIS4',
 'LAIS1',
 'LAIS2',
 'LAIS3']

elec_dict is as follows:

{'LFPG1': [-0.0211204, 0.0518157, 0.0425739356174474],
 'LFPG2': [-0.021700700000000003, 0.0444696, 0.0497241092479105],
 'LFPG3': [-0.0231085, 0.0357866, 0.0537190762100455],
 'LFPG4': [-0.0215652, 0.028994100000000002, 0.059433985242819405],
 'LFPG5': [-0.017928200000000002, 0.0202426, 0.06282722560521581],
 'LFPG6': [-0.0128325, 0.00806245, 0.06988643950838799],
 'LFPG7': [-0.009182560000000001, -0.000170408, 0.07066406845412461],
 'LFPG8': [-0.0148465, -0.00717425, 0.0744570765099918],
 'LFPG9': [-0.0297575, 0.0494856, 0.0371787602006774],
 'LFPG10': [-0.029237100000000002, 0.04269680000000001, 0.0453629682230858],
 'LFPG11': [-0.0288688, 0.034545400000000004, 0.049347715501969],
 'LFPG12': [-0.0325394, 0.0269882, 0.0547113948325137],
 'LFPG13': [-0.0311134, 0.0179123, 0.060000938258775796],
 'LFPG14': [-0.0297766, 0.00610056, 0.0660969066265409],
 'LFPG15': [-0.030526800000000003, -0.00986761, 0.0683945391257418],
 'LFPG16': [-0.0319256, -0.0153364, 0.0721121740764809],
 'LFPG17': [-0.0374236, 0.0464388, 0.0312989921755522],
 'LFPG18': [-0.038046300000000005, 0.038786900000000006, 0.04151989065957121],
 'LFPG19': [-0.0382915, 0.033554299999999995, 0.0454499399952208],
 'LFPG20': [-0.0421091, 0.0229929, 0.0505558727308064],
 'LFPG21': [-0.0414722, 0.013149300000000001, 0.0567791186691879],
 'LFPG22': [-0.04165870000000001, 0.00303469, 0.0600087223847562],
 'LFPG23': [-0.0440802, -0.00777111, 0.061366063652951504],
 'LFPG24': [-0.044979399999999996, -0.017612600000000003, 0.06481629719808021],
 'LFPG25': [-0.0459443, 0.04392240000000001, 0.0259704470121004],
 'LFPG26': [-0.0460278, 0.0357858, 0.034095457751498005],
 'LFPG27': [-0.045699, 0.0280037, 0.0414376209107973],
 'LFPG28': [-0.046341299999999995, 0.0171388, 0.0480353317908815],
 'LFPG29': [-0.0485106, 0.00963697, 0.050873368845506695],
 'LFPG30': [-0.0500475, 0.00046404400000000004, 0.054828320908896305],
 'LFPG31': [-0.0525009, -0.00852335, 0.055103119660809906],
 'LFPG32': [-0.053044100000000004, -0.019564, 0.0573054222573997],
 'LSPIS1': [-9.31769e-05, -0.0744192, 0.0385639418886536],
 'LSPIS2': [-4.51567e-05, -0.0628905, 0.0487602096610609],
 'LSPIS3': [0.0013493, -0.0516644, 0.056598275392439],
 'LSPIS4': [0.00162881, -0.04233870000000001, 0.062016586796788606],
 'LIPIS1': [0.0049590500000000004, -0.06371840000000001, 0.0426646026901921],
 'LIPIS2': [0.00171951, -0.051365400000000005, 0.0504772296053725],
 'LIPIS3': [0.0018220600000000001, -0.046999400000000004, 0.0586447670523916],
 'LIPIS4': [0.00178181, -0.0385381, 0.0594603833106994],
 'LAIS1': [0.0018628499999999999, 0.00114678, 0.0244707713740858],
 'LAIS2': [0.00117872, 0.00855424, 0.0320050649689319],
 'LAIS3': [7.57824e-05, 0.0172445, 0.040820480774000106]}

Can anyone resolve this issue? Thanks.

agramfort, 2020-09-17:
this suggests that you have neither mayavi nor pyvista working on your system to be able to do 3D plots

anupdas777, 2020-09-17:
> 
> 
> this suggests that you have neither mayavi nor pyvista working on your system to be able to do 3D plots

anupdas777, 2020-09-17:
> 
> 
> this suggests that you have neither mayavi nor pyvista working on your system to be able to do 3D plots

I have both of these installed, still getting the error.

GuillaumeFavelier, 2020-09-17:
Could you share the results of: 

```
python -c "import mne; mne.sys_info()"
```

And also:

```
python -c "import pyvista; print(pyvista.Report())"
```

anupdas777, 2020-09-17:
While executing "mne.sys_info()", I am getting an error "Kernel Restarting The kernel for Test.ipynb appears to have died. It will restart automatically." Do you know what I am doing wrong? Why the kernel is dying? Thanks.

anupdas777, 2020-09-17:
> 
> 
> Could you share the results of:
> 
> ```
> python -c "import mne; mne.sys_info()"
> ```
> 
> And also:
> 
> ```
> python -c "import pyvista; print(pyvista.Report())"
> ```

While executing "mne.sys_info()", I am getting an error "Kernel Restarting The kernel for Test.ipynb appears to have died. It will restart automatically." Do you know what I am doing wrong? Why the kernel is dying? Thanks.

agramfort, 2020-09-17:
you have an install pb certainly of vtk

anupdas777, 2020-09-17:
> 
> you have an install pb certainly of vtk

Can you please send commands to uninstall pb of vtk? Or, uninstall entire vtk? Thanks.

drammock, 2020-09-17:
you (probably) need to uninstall and then reinstall vtk.  `pip uninstall vtk` followed by `pip install --no-cache-dir vtk>=9.0.1` is a good first thing to try. Make sure if you are using conda that you are actually in the correct conda environment (the one with MNE-Python in it) when you do this.

anupdas777, 2020-09-17:
> 
> 
> you (probably) need to uninstall and then reinstall vtk. `pip uninstall vtk` followed by `pip install --no-cache-dir vtk>=9.0.1` is a good first thing to try. Make sure if you are using conda that you are actually in the correct conda environment (the one with MNE-Python in it) when you do this.

Did exactly what you suggested, but kernel is still dying when I type mne.sys_info() or pyvista.Report(). Thanks.

larsoner, 2020-09-17:
For now uninstall pyvista, it's probably the culprit. You can do `pip uninstall pyvista`. Then let us know what `mne.sys_info()` says. If that still dies, try `pip uninstall mayavi` then do it again.

anupdas777, 2020-09-17:
> 
> 
> For now uninstall pyvista, it's probably the culprit. You can do `pip uninstall pyvista`. Then let us know what `mne.sys_info()` says. If that still dies, try `pip uninstall mayavi` then do it again.

Uninstalled both pyvista and mayavi and it worked. Here is the output of "mne.sys_info()":

Platform:      Linux-3.10.0-693.17.1.el7.x86_64-x86_64-with-centos-7.4.1708-Core
Python:        3.7.7 (default, May  7 2020, 21:25:33)  [GCC 7.3.0]
Executable:    /home1/anup.das/.conda/envs/py3_env/bin/python
CPU:           x86_64: 40 cores
Memory:        Unavailable (requires "psutil" package)
mne:           0.20.8
numpy:         1.19.2 {blas=openblas, lapack=openblas}
scipy:         1.5.2
matplotlib:    3.3.1 {backend=module://ipykernel.pylab.backend_inline}

sklearn:       0.23.1
numba:         Not found
nibabel:       3.1.1
cupy:          Not found
pandas:        1.1.0
dipy:          Not found
mayavi:        Not found
pyvista:       Not found
vtk:           9.0.1

Please suggest next step to correct my original error.

Thanks.

Anup

larsoner, 2020-09-17:
What do you get for `glxinfo | grep enderer` and `glxinfo | grep "OpenGL version string"`? I get for example:
```
$ glxinfo | grep enderer
OpenGL renderer string: GeForce GTX 650 Ti BOOST/PCIe/SSE2
$ glxinfo | grep "OpenGL version string"
OpenGL version string: 4.6.0 NVIDIA 440.100
```
showing that I have a system with modern OpenGL. If you don't have OpenGL version 3+ 3D rendering is not going to work.

anupdas777, 2020-09-17:
> What do you get for `glxinfo | grep enderer` and `glxinfo | grep "OpenGL version string"`? I get for example:
>
> ```
> $ glxinfo | grep enderer
> OpenGL renderer string: GeForce GTX 650 Ti BOOST/PCIe/SSE2 [...]

I get the following.

$ glxinfo | grep enderer
    GLX_MESA_multithread_makecurrent, GLX_MESA_query_renderer,
    GLX_MESA_copy_sub_buffer, GLX_MESA_query_renderer, GLX_OML_swap_method,
OpenGL renderer string: Gallium 0.4 on llvmpipe (LLVM 3.9, 256 bits)

$ glxinfo | grep "OpenGL version string"
OpenGL version string: 2.1 Mesa 17.0.1

anupdas777, 2020-09-18:
> What do you get for `glxinfo | grep enderer` and `glxinfo | grep "OpenGL version string"`? I get for example:
>
> ```
> $ glxinfo | grep enderer
> OpenGL renderer string: GeForce GTX 650 Ti BOOST/PCIe/SSE2 [...]

What's the next step that I should take to fix this? Thanks.

larsoner, 2020-09-18:
2.1 is too old. Can you try the fix suggested here with MESA_GL_VERSION_OVERRIDE to see if setting it makes the version number higher?

https://mne.tools/dev/install/advanced.html?highlight=mesa_gl_version_override#off-screen-rendering-in-mne-python-on-linux-with-mesa

If it does, then hopefully rendering works for you. If it doesn't, you'll have to look into upgrading your MESA 3D rendering libraries.

anupdas777, 2020-09-18:
> 2.1 is too old. Can you try the fix suggested here with MESA_GL_VERSION_OVERRIDE to see if setting it makes the version number higher? [...]

Did the override as "$ export MESA_GL_VERSION_OVERRIDE=3.3" and I now get the following:

$ glxinfo | grep "OpenGL version string"
OpenGL version string: 3.3 (Core Profile) Mesa 17.0.1

Please suggest next step. Thanks.

larsoner, 2020-09-18:
Now try reinstalling pyvista and pyvistaqt, and re-running `mne.sys_info`. If it works, try re-running your plot_alignment code

anupdas777, 2020-09-18:
> 
> 
> Now try reinstalling pyvista and pyvistaqt, and re-running `mne.sys_info`. If it works, try re-running your plot_alignment code

I reinstalled pyvista and pyvistaqt and "mne.sys_info" works fine now. "mne.sys_info" shows the following information:

Platform:      Linux-3.10.0-693.17.1.el7.x86_64-x86_64-with-centos-7.4.1708-Core
Python:        3.7.7 (default, May  7 2020, 21:25:33)  [GCC 7.3.0]
Executable:    /home1/anup.das/.conda/envs/py3_env/bin/python
CPU:           x86_64: 40 cores
Memory:        Unavailable (requires "psutil" package)
mne:           0.20.8
numpy:         1.19.2 {blas=openblas, lapack=openblas}
scipy:         1.5.2
matplotlib:    3.3.1 {backend=module://ipykernel.pylab.backend_inline}

sklearn:       0.23.1
numba:         Not found
nibabel:       3.1.1
cupy:          Not found
pandas:        1.1.0
dipy:          Not found
mayavi:        Not found
pyvista:       0.26.0
vtk:           9.0.1

However, plot_alignment shows me the following error:

QtDeprecationError: `BackgroundPlotter` has moved to pyvistaqt.
    You can install this from PyPI with: `pip install pyvistaqt`
    See https://github.com/pyvista/pyvistaqt

What should I do? Thanks.

larsoner, 2020-09-18:
Update to latest master branch, or wait until Monday when we cut a new release

anupdas777, 2020-09-18:
> Update to latest master branch

Can you please send commands to do this? Thanks.

drammock, 2020-09-18:
> > Update to latest master branch
> 
> Can you please send commands to do this? Thanks.

https://mne.tools/dev/install/advanced.html#using-the-development-version-of-mne-python-latest-master

anupdas777, 2020-09-18:
> 
> 
> > > Update to latest master branch
> > 
> > 
> > Can you please send commands to do this? Thanks.
> 
> https://mne.tools/dev/install/advanced.html#using-the-development-version-of-mne-python-latest-master

Did $ pip install --upgrade --no-deps https://github.com/mne-tools/mne-python/archive/master.zip, but I am back to the "kernel has died" problem. What to do? Thanks.

anupdas777, 2020-09-21:
> Did $ pip install --upgrade --no-deps https://github.com/mne-tools/mne-python/archive/master.zip, but I am back to the "kernel has died" problem. What to do? Thanks.

Any updates here? I am still not able to solve this problem. Need this to work asap. Thanks.

larsoner, 2020-09-21:
It's possible that MESA<18.4 has problems, see https://github.com/mne-tools/mne-python/issues/7977#issuecomment-673623894. Not sure if there is an option other than upgrading MESA or switching to a computer with a graphics card with better OpenGL support :(

anupdas777, 2020-09-21:
> 
> 
> It's possible that MESA<18.4 has problems, see [#7977 (comment)](https://github.com/mne-tools/mne-python/issues/7977#issuecomment-673623894). Not sure if there is an option other than upgrading MESA or switching to a computer with a graphics card with better OpenGL support :(

Can you please send commands to upgrade MESA? Thanks.

drammock, 2020-09-21:
> 1. Can you please send commands to uninstall pb of vtk? Or, uninstall entire vtk? Thanks.
> 2. Please suggest next step to correct my original error.
> 3. What's the next step that I should take to fix this? Thanks.
> 4. Please suggest next step. Thanks.
> 5. What should I do? Thanks.
> 6. Can you please send commands to do this? Thanks.
> 7. I am back to the "kernel has died" problem. What to do? Thanks.
> 8. Any updates here? I am still not able to solve this problem. Need this to work asap. Thanks.

Dear @anupdas777, it would be easier for us to remain motivated to help you if we saw some evidence that you are at least *trying* to find answers or figure out solutions on your own.  Perhaps you are, and you're just not telling us what you've already tried and what the results were...  but if that's the case, it is considered polite to tell us what you've already tried. http://whathaveyoutried.com/ gives some context for why this approach to asking for help is an important aspect of learning to program.  So, regarding your last question:

> 9. Can you please send commands to upgrade MESA? Thanks.

Did you try using a search engine to look for "how to upgrade mesa on centOS"?  If so, what did you find?  Did you try it?  What happened?  What else did you try?

anupdas777, 2020-09-21:
> 
> 
> > 1. Can you please send commands to uninstall pb of vtk? Or, uninstall entire vtk? Thanks.
> > 2. Please suggest next step to correct my original error.
> > 3. What's the next step that I should take to fix this? Thanks.
> > 4. Please suggest next step. Thanks.
> > 5. What should I do? Thanks.
> > 6. Can you please send commands to do this? Thanks.
> > 7. I am back to the "kernel has died" problem. What to do? Thanks.
> > 8. Any updates here? I am still not able to solve this problem. Need this to work asap. Thanks.
> 
> Dear @anupdas777, it would be easier for us to remain motivated to help you if we saw some evidence that you are at least _trying_ to find answers or figure out solutions on your own. Perhaps you are, and you're just not telling us what you've already tried and what the results were... but if that's the case, it is considered polite to tell us what you've already tried. http://whathaveyoutried.com/ gives some context for why this approach to asking for help is an important aspect of learning to program. So, regarding your last question:
> 
> > 1. Can you please send commands to upgrade MESA? Thanks.
> 
> Did you try using a search engine to look for "how to upgrade mesa on centOS"? If so, what did you find? Did you try it? What happened? What else did you try?

Many people say many things on the internet to install mne and have tried all those things but failed, so I posted here. And I do not have any other ideas regarding how to install mne. If my behavior is against the policies of this forum, then you do not have to help further. Thanks.

drammock, 2020-09-21:
> If my behavior is against the policies of this forum, then you do not have to help further.

You have not violated our [Code of Conduct](https://github.com/mne-tools/.github/blob/master/CODE_OF_CONDUCT.md). But there are norms that are also relevant here, which you're not following. Because those norms are not always clear or well documented, I took this opportunity to tell you that I think you've violated a norm, explain what that norm is, and offer some further reading to give context for why that norm exists and how it benefits the community.  Specifically, the norm is that people requesting help should try to solve the problem themselves first, and when asking for help they should document what they've already tried.

> Many people say many things on the internet to install mne and have tried all those things but failed

I'm glad to hear that you tried several things yourself before posting here. But your original post doesn't say "here are the 3 things I've tried so far to fix this, and I'm still stuck".  Similarly, your most recent question about MESA doesn't say "You suggested I should upgrade MESA, so I tried to upgrade MESA by following instructions on such-and-such webpage, but it didn't work." Instead, you only asked us to tell you the answer. That is what I'm calling attention to here, and why I enumerated your 9 questions in my last response: to draw attention to the pattern your questions conform to, which is a pattern of "just tell me the answer".  *Asking for help in that way comes across as lazy* even if you've actually tried very hard already to solve the problem yourself.

> Can you please send commands to upgrade MESA? Thanks.

Pre-built binary packages for MESA are not distributed by the MESA developers.  MESA comes with Linux, and usually new versions of MESA are provided through the package manager fairly promptly after a new MESA release.  The normal system upgrade commands `sudo apt update; sudo apt upgrade` will update to the latest available version *that is compatible with your system*; if you want something newer than that, you can try following [the instructions for building MESA from source](https://docs.mesa3d.org/install.html).  If you have problems getting that to work, you should seek help from the MESA community.  As @larsoner suggested, sometimes the only remedy is to use a system with more modern graphics card that is capable of supporting newer versions of MESA.

anupdas777, 2020-09-21:
> You have not violated our [Code of Conduct](https://github.com/mne-tools/.github/blob/master/CODE_OF_CONDUCT.md). But there are norms that are also relevant here, which you're not following. Because [...]

I have documented everything in my previous posts here that I have tried and I have also tried my best to get this thing to work, so your comments like "violated a norm", "comes across as lazy" etc. are obviously wrong. It looks like you do not want to help, so I will just go ahead and close this issue. Thanks.

closed 2020-09-21 by anupdas777