mne-tools/mne-python #8273: Speed up CIs with mamba?

cbrnr, 2020-09-17:
I recently tried [mamba](https://github.com/mamba-org/mamba) as a drop-in replacement for conda. As advertised, it seems to be much faster, so this might be an easy way to speed up our CIs (not sure how much time is spent on bootstrapping envs and installing packages, but I guess every bit matters).

There's even [micromamba](https://gist.github.com/wolfv/fe1ea521979973ab1d016d95a589dcde), which enables the creation of new envs without having to install a base miniconda env first - I haven't tried that myself, but [this tweet](https://twitter.com/wuoulf/status/1306530602087313408) says it's really fast.

larsoner, 2020-09-17:
Seems worth trying, a bit of time is spent in the conda stages

cbrnr, 2020-09-17:
As far as I can see we currently use `conda` only in one Windows job (disregarding the notebook job) on Azure Pipelines. Currently, micromamba doesn't work on Windows though, so maybe something for another time. I didn't know that Linux jobs use `pip`.

larsoner, 2020-09-17:
> I didn't know that Linux jobs use pip.

Only one, the [others](https://github.com/mne-tools/mne-python/blob/master/.travis.yml#L54-L62) use [ci-helpers](https://github.com/astropy/ci-helpers/) which uses miniconda

cbrnr, 2020-09-17:
Thanks for pointing that out - I didn't know these scripts. It seems like they already support mamba (but not micromamba), so maybe the best thing would be to try to get micromamba support upstream.

larsoner, 2020-09-17:
Sure, and in the meantime I'll get a run going with `MAMBA=true` to see if it helps

cbrnr, 2020-09-17:
Looks like they're already on it: https://github.com/astropy/ci-helpers/issues/458

larsoner, 2021-02-19:
I think we tried it and it didn't help but we can keep this in mind and try it again in the future. I'm not sure it will, though -- based on what I've seen, most of the time is actually spent downloading and installing packages rather than getting `conda` running at all. And now that we use GitHub actions and Azure conda seems to be better supported out of the box anyway

closed 2021-02-19 by larsoner