mne-tools/mne-python #8277: 3D Mayavi - Mac Os X problems

RickLicona, 2020-09-17:
Hi,

The Mayavi tool is too slow In Mac Os X. I am programming some MNE examples (head model and forward computation), and some times the 3D models does not appear (TVTK white window). Also, when the 3D model appears, it is impossible to interact with the models. 

The fast code solution to see at least a image of the 3D simulation is: 

```
from mayavi import mlab 
#mlab.options.offscreen = True
mlab.init_notebook() 

# The transformation file obtained by coregistration
trans = data_path + '/MEG/sample/sample_audvis_raw-trans.fif'

info = mne.io.read_info(raw_fname)
# Here we look at the dense head, which isn't used for BEM computations but
# is useful for coregistration.
coregistration = mne.viz.plot_alignment(info, trans, subject=subject, dig=True,
                       meg=["helmet", "sensors"], subjects_dir=subjects_dir,
                       surfaces='head-dense')
coregistration
```
 I would like to have a solution for this issue. 

Some recommendations? 

Best.

larsoner, 2020-09-17:
What is `mne.sys_info()`? I wonder if your OpenGL support is too old.

You could try wrapping the `plot_alignment` call in a `with mne.viz.use_3d_backend('pyvista')` to see if PyVista does any better, but these both use VTK under the hood so I'd be suprised.

RickLicona, 2020-09-17:
> What is `mne.sys_info()`? I wonder if your OpenGL support is too old.
> 
> You could try wrapping the `plot_alignment` call in a `with mne.viz.use_3d_backend('pyvista')` to see if PyVista does any better, but these both use VTK under the hood so I'd be suprised.

Platform:      Darwin-19.6.0-x86_64-i386-64bit
Python:        3.6.0 (v3.6.0:41df79263a11, Dec 22 2016, 17:23:13)  [GCC 4.2.1 (Apple Inc. build 5666) (dot 3)]
Executable:    /Users/ricklicona/PycharmProjects/RExam/bin/python
CPU:           i386: 8 cores
Memory:        Unavailable (requires "psutil" package)
mne:           0.20.8
numpy:         1.15.2 {blas=openblas, lapack=openblas}
scipy:         1.5.2
matplotlib:    3.0.2 {backend=module://ipykernel.pylab.backend_inline}

sklearn:       0.21.3
numba:         0.43.1
nibabel:       3.1.1
cupy:          Not found
pandas:        0.25.1
dipy:          Not found
mayavi:        4.7.2 {qt_api=pyqt5, PyQt5=5.15.1}
pyvista:       0.26.0
vtk:           8.1.0

agramfort, 2020-09-18:
+1 to try using pyvista

>

hoechenberger, 2020-09-18:
I also wonder if VTK 9 would improve things here? We're going to release MNE-Python 0.21 very soon hopefully, with support for VTK 9.

RickLicona, 2020-09-18:
I tried this:
```
# The transformation file obtained by coregistration
trans = data_path + '/MEG/sample/sample_audvis_raw-trans.fif'

info = mne.io.read_info(raw_fname)
# Here we look at the dense head, which isn't used for BEM computations but
# is useful for coregistration.
with mne.viz.use_3d_backend('pyvista'):

        mne.viz.plot_alignment(info, trans, subject=subject, dig=True,
                       meg=["helmet", "sensors"], subjects_dir=subjects_dir,
                       surfaces='head-dense')
```

**Out:**

Read a total of 3 projection items:
        PCA-v1 (1 x 102)  idle
        PCA-v2 (1 x 102)  idle
        PCA-v3 (1 x 102)  idle
Using pyvista 3d backend.

Using lh.seghead for head surface.
Getting helmet for system 306m
Using mayavi 3d backend.

**However, I got the next error message:**

QtDeprecationError: `BackgroundPlotter` has moved to pyvistaqt.
    You can install this from PyPI with: `pip install pyvistaqt`
    See https://github.com/pyvista/pyvistaqt 

It seems that the solution is here:  QtDeprecationError is unclear #826 (https://github.com/pyvista/pyvista/issues/826), but it is not a MNE case.  

Also I tried a Eric's code:
```

import mne
from pyvistaqt import BackgroundPlotter
fname = mne.datasets.sample.data_path() + '/MEG/sample/sample_audvis_raw.fif'
raw = mne.io.read_raw_fif(fname)
with mne.viz.use_3d_backend('pyvista'):
    mne.viz.plot_alignment(raw.info, show_axes=True, eeg=False, surfaces=[])
```
I got the same error message.

larsoner, 2020-09-18:
Did you try the solution suggested by the error, namely `pip install pyvistaqt`? It should install, then it should not error on these lines (that way at least) anymore.

hoechenberger, 2020-09-26:
Hello @RickLicona, have you been able to resolve this?

RickLicona, 2020-09-27:
> Hello @RickLicona, have you been able to resolve this?

Hi,

Thanks for the interest!. Unfortunately, I cannot resolve this issue, despite the error suggest of install pyvistaqt (0.2.0). I added this tool in my Pycharm IDE, however, the error remains equal. When I come back again to this part of MNE tutorial and found the solution, I will publish it ;). 

[image]

Best.

RickLicona, 2020-10-05:
Dear all, 

This part of the code (**Visualizing the coregistration**) from the tutorial named **Head model and forward computation** does no longer generate any error messages.  Now I have the correct 3D simulations :-).

[image]
[image]

On the other hand, it seems that is not appropiate the next code for the future, the 3D simulation is wrong. 
[image]
[image]

Here is another example with correct 3D simulation:
[image]
[image]

This is my` mne.sys_info()`:
[image]

 **Also, I have installed pvistaqt (0.2.0), FreeSurfer 6.0 and the Mac OS X patch for 6.0 version (https://drive.google.com/file/d/1ezjiNIbGNrNAiYD5ciYxMO2SK-R5vz9A/view)**

**OS version: macOS Catalina 10.15.7 (the last version)**

THANKS FOR THE HELP AND UPDATE :-) !!

Best wishes,
Ricardo.

closed 2020-10-05 by RickLicona

larsoner, 2020-10-05:
> On the other hand, it seems that is not appropiate the next code for the future, the 3D simulation is wrong.

What's wrong with that image?

RickLicona, 2020-10-05:
> > On the other hand, it seems that is not appropiate the next code for the future, the 3D simulation is wrong.
> 
> What's wrong with that image?

It does not generate the image as the original MNE tutorial (https://mne.tools/dev/auto_tutorials/source-modeling/plot_forward.html, , see the below image). In resume, the code from the original tutorial (**visualizing the coregistration** ) is enough, there is not more error messages as I had before.  

[image]

larsoner, 2020-10-05:
That example code has `surfaces=['head-dense']`, the code that precedes the image below has `surfaces=[]`, so it should not show any surfaces (i.e., it looks correct)