mne-tools/mne-python #8280: A question in "Compute MxNE with time-frequency sparse prior"

zhchsh19, 2020-09-20:
I tried to use my own data to use this algorithm，but no matter how I change the value of alpha, I get the following ONE result：
IndexError: index 4224 is out of bounds for axis 0 with size 4098

The same fault occurs in the algorithm "Compute a sparse inverse solution using the Gamma-Map empirical Bayesian method" and "Compute sparse inverse solution with mixed norm: MxNE and irMxNE"

**Here is my own code：**
cov = mne.read_cov(fname_cov)

# Handling average file

evoked = mne.read_evokeds(fname_evoked , condition='tool', baseline=(None, 0))
evoked = mne.pick_channels_evoked(evoked)
# We make the window slightly larger than what you'll eventually be interested
# in ([-0.05, 0.3]) to avoid edge effects.
labels_vol = ['Left-Amygdala',
              'Left-Thalamus-Proper',
              'Left-Cerebellum-Cortex',
              'Brain-Stem',
              'Right-Amygdala',
              'Right-Thalamus-Proper',
              'Right-Cerebellum-Cortex']

src = mne.setup_source_space(subject, spacing='oct6',
                             add_dist=False, subjects_dir=subjects_dir)

vol_src = mne.setup_volume_source_space(
    subject, mri=fname_aseg, pos=10.0, bem=fname_model,
    volume_label=labels_vol, subjects_dir=subjects_dir,
    add_interpolator=False,  # just for speed, usually this should be True
    verbose=True)

src += vol_src
# Handling forward solution
forward = mne.make_forward_solution(
    fname_evoked, fname_trans, src, fname_bem,
    mindist=5.0,  # ignore sources<=5mm from innerskull
    )

# alpha parameter is between 0 and 100 (100 gives 0 active source)
alpha =60.  # general regularization parameter
# l1_ratio parameter between 0 and 1 promotes temporal smoothness
# (0 means no temporal regularization)
l1_ratio = 0.01  # temporal regularization parameter

loose, depth = 1, 0.9  # loose orientation & depth weighting

# Compute dSPM solution to be used as weights in MxNE
inverse_operator = make_inverse_operator(evoked.info, forward, cov,
                                         loose=loose, depth=depth)
stc_dspm = apply_inverse(evoked, inverse_operator, lambda2=1. / 9.,
                         method='dSPM')

# Compute TF-MxNE inverse solution with dipole output
dipoles, residual = tf_mixed_norm(
    evoked, forward, cov, alpha=alpha, l1_ratio=l1_ratio, loose=loose,
    depth=depth, maxit=200, tol=1e-6, weights=stc_dspm, weights_min=8.,
    debias=True, wsize=16, tstep=4, window=0.05, return_as_dipoles=True,
    return_residual=True)

# Crop to remove edges
for dip in dipoles:
    dip.crop(tmin=-0.1, tmax=0.5)
evoked.crop(tmin=-0.1, tmax=0.5)
residual.crop(tmin=-0.1, tmax=0.5)

stc = make_stc_from_dipoles(dipoles, forward['src'])

Looking forward to your reply

agramfort, 2020-09-20:
can you make available the 3 files : evoked, forward and cov?

zhchsh19, 2020-09-21:
> can you make available the 3 files : evoked, forward and cov?

zhchsh19, 2020-09-21:
> can you make available the 3 files : evoked, forward and cov?

[cov.zip](https://github.com/mne-tools/mne-python/files/5253052/cov.zip)
[evoked.zip](https://github.com/mne-tools/mne-python/files/5253053/evoked.zip)

fwd is bigger than 10MB, how can I uplosd it?

agramfort, 2020-09-21:
use dropbox, google drive, https://wetransfer.com/ etc.

>

zhchsh19, 2020-09-21:
> use dropbox, google drive, https://wetransfer.com/ etc.
> […](#)

https://we.tl/t-uAn1lxOUYZ

agramfort, 2020-09-21:
```
import mne
from mne.inverse_sparse import tf_mixed_norm

forward = mne.read_forward_solution('sub01wkl-fwd.fif')
evoked = mne.read_evokeds('sub01wkl-ave.fif')[0]
cov = mne.read_cov('sub01wkl-cov.fif')

# alpha parameter is between 0 and 100 (100 gives 0 active source)
alpha = 40.  # general regularization parameter
# l1_ratio parameter between 0 and 1 promotes temporal smoothness
# (0 means no temporal regularization)
l1_ratio = 0.03  # temporal regularization parameter

loose, depth = 'auto', 0.9  # loose orientation & depth weighting

dipoles, residual = tf_mixed_norm(
    evoked, forward, cov, alpha=alpha, l1_ratio=l1_ratio, loose=loose,
    depth=depth, maxit=200, tol=1e-6,
    debias=True, wsize=16, tstep=4, window=0.05, return_as_dipoles=True,
    return_residual=True, rank="info")
```

I do not get the error you report. What did you do differently,

zhchsh19, 2020-09-21:
> ```
> import mne
> from mne.inverse_sparse import tf_mixed_norm
>
> forward = mne.read_forward_solution('sub01wkl-fwd.fif')
> evoked = mne.read_evokeds('sub01wkl-ave.fif')[0]
> cov = [...]
I got the error when I run :
stc = make_stc_from_dipoles(dipoles, forward['src'])

agramfort, 2020-09-21:
```
import mne
from mne.inverse_sparse import tf_mixed_norm

forward = mne.read_forward_solution('sub01wkl-fwd.fif')
evoked = mne.read_evokeds('sub01wkl-ave.fif')[0]
cov = mne.read_cov('sub01wkl-cov.fif')

# alpha parameter is between 0 and 100 (100 gives 0 active source)
alpha = 90.  # general regularization parameter
# l1_ratio parameter between 0 and 1 promotes temporal smoothness
# (0 means no temporal regularization)
l1_ratio = 0.03  # temporal regularization parameter

loose, depth = 'auto', 0.9  # loose orientation & depth weighting

dipoles, residual = tf_mixed_norm(
    evoked, forward, cov, alpha=alpha, l1_ratio=l1_ratio, loose=loose,
    depth=depth, maxit=200, tol=1e-6,
    debias=True, wsize=16, tstep=4, window=0.05, return_as_dipoles=True,
    return_residual=True, rank="info")

stc = mne.inverse_sparse.make_stc_from_dipoles(dipoles, forward['src'])
```

works fine with current master and release 0.21 from today.

zhchsh19, 2020-09-24:
I think there is somthing wrong with the parameter"alpha",  If it is too big, then I can only get three rows' data, such as: 

//**117: RuntimeWarning: source space does not contain any vertices for 66 labels:
['bankssts-lh', 'bankssts-rh', 'caudalanteriorcingulate-lh', 'caudalanteriorcingulate-rh', 'caudalmiddlefrontal-lh', 'caudalmiddlefrontal-rh', 'cuneus-lh', 'cuneus-rh', 'entorhinal-lh', 'entorhinal-rh', 'frontalpole-lh', 'frontalpole-rh', 'fusiform-lh', 'fusiform-rh', 'inferiorparietal-lh', 'inferiorparietal-rh', 'inferiortemporal-lh', 'insula-lh', 'insula-rh', 'isthmuscingulate-lh', 'isthmuscingulate-rh', 'lateraloccipital-lh', 'lateralorbitofrontal-lh', 'lateralorbitofrontal-rh', 'lingual-lh', 'lingual-rh', 'medialorbitofrontal-lh', 'medialorbitofrontal-rh', 'middletemporal-lh', 'middletemporal-rh', 'paracentral-lh', 'paracentral-rh', 'parahippocampal-lh', 'parahippocampal-rh', 'parsopercularis-lh', 'parsopercularis-rh', 'parsorbitalis-lh', 'parsorbitalis-rh', 'parstriangularis-lh', 'parstriangularis-rh', 'pericalcarine-lh', 'pericalcarine-rh', 'postcentral-lh', 'postcentral-rh', 'posteriorcingulate-lh', 'posteriorcingulate-rh', 'precentral-lh', 'precentral-rh', 'precuneus-lh', 'precuneus-rh', 'rostralanteriorcingulate-lh', 'rostralanteriorcingulate-rh', 'rostralmiddlefrontal-lh', 'rostralmiddlefrontal-rh', 'superiorfrontal-lh', 'superiorfrontal-rh', 'superiorparietal-lh', 'superiorparietal-rh', 'superiortemporal-lh', 'superiortemporal-rh', 'supramarginal-lh', 'supramarginal-rh', 'temporalpole-lh', 'temporalpole-rh', 'transversetemporal-lh', 'transversetemporal-rh']
Assigning all-zero time series.
  label_ts = mne.extract_label_time_course(**

**If I set "alphs"  too small, such as** 
alpha =30.  # general regularization parameter
# l1_ratio parameter between 0 and 1 promotes temporal smoothness
# (0 means no temporal regularization)
l1_ratio = 0.03  # temporal regularization parameter
snr = 3.0            # use smaller SNR for raw data
inv_method = 'dSPM'  # sLORETA, MNE, dSPM
parc = 'aparc'   
depth =  None  # loose orientation & depth weighting
loose = dict(surface=0.2, volume=1.)
lambda2 = 1.0 / snr ** 2
inverse_operator = make_inverse_operator(evoked.info, forward, cov,
                                         loose=loose, depth=depth)
stc_dspm = apply_inverse(evoked, inverse_operator, lambda2,
                         inv_method,pick_ori=None)
src = inverse_operator['src']
dipoles, residual = tf_mixed_norm(
    evoked, forward, cov, alpha=alpha, l1_ratio=l1_ratio, loose=loose,
    depth=depth, maxit=200, tol=1e-6,weights=stc_dspm, weights_min=8.,
    debias=True, wsize=16, tstep=4, window=0.05, return_as_dipoles=True,
    return_residual=True, rank="info")
stc = mne.inverse_sparse.make_stc_from_dipoles(dipoles, forward['src'])

**Then I will get:**
Traceback (most recent call last):

  File "E:\chengxu\liucheng\hunhe4-tf.py", line 86, in <module>
    stc = mne.inverse_sparse.make_stc_from_dipoles(dipoles, forward['src'])

  File "<decorator-gen-365>", line 21, in make_stc_from_dipoles

  File "D:\anzhuang\anzhuanganaconda\envs\mne\lib\site-packages\mne\inverse_sparse\mxne_inverse.py", line 279, in make_stc_from_dipoles
    rh_vertno.append(src[1]['vertno'][idx - n_lh_points])

IndexError: index 4224 is out of bounds for axis 0 with size 4098

agramfort, 2020-09-24:
please a well formatted code. You can use https://gist.github.com/ if it
helps.
Then I will look into it

>

zhchsh19, 2020-09-26:
> please a well formatted code. You can use https://gist.github.com/ if it helps. Then I will look into it
> […](#)

There are my codes and data, thank you very much! 
https://we.tl/t-9oSNoKe2eU

agramfort, 2020-09-27:
this https://gist.github.com/agramfort/56b5c0d605cdc136cf3a7d31aad74d67
runs fine for me.

I see that you can recompute src while you should use forward['src'] to
avoid issues

>

larsoner, 2021-01-07:
Sounds like this is a `src` vs `fwd['src']` issue that can be fixed in the user scripts, so I'll go ahead and close

closed 2021-01-07 by larsoner