mne-tools/mne-python #8281: Read EDF files with dummy zero year/month/day

wolfgang-ganglberger, 2020-09-20:
#### Describe the bug
EDF file that has dummy year/month/day of 0's cannot be read by mne.io.read_raw_edf() and exits with error "month must be in 1..12"

#### Expected results
I except the code still be able to read the data in the .edf file, and I would suggest that another dummy date is used instead of the 0's in the file, i propose using 1900-01-01.

Current code in mne/io/edf/edf.py:
```
# Recording ID
        meas_id = {}
        meas_id['recording_id'] = fid.read(80).decode('latin-1').strip(' \x00')

        day, month, year = [int(x) for x in
                            re.findall(r'(\d+)', fid.read(8).decode())]
        hour, minute, sec = [int(x) for x in
                             re.findall(r'(\d+)', fid.read(8).decode())]
        century = 2000 if year < 50 else 1900

```
Proposal (same as above, only an additional if clause):
```
# Recording ID
        meas_id = {}
        meas_id['recording_id'] = fid.read(80).decode('latin-1').strip(' \x00')

        day, month, year = [int(x) for x in
                            re.findall(r'(\d+)', fid.read(8).decode())]
        hour, minute, sec = [int(x) for x in
                             re.findall(r'(\d+)', fid.read(8).decode())]
        century = 2000 if year < 50 else 1900

        if any([day == 0, month == 0, year == 0]):
            year = 1900
            month = 1
            day = 1

```

If you think this is a solution you'd accept, I can also do a Pull request, although I am happy to hand it over to someone else too, as it is just a simple if clause add-on.

Best,
Wolfgang

cbrnr, 2020-09-21:
I'm fine with this change - @wolfgang-ganglberger it would be great if you could submit a PR. I suggest to also issue a warning if a date component is zero, e.g.

```python
        if any([day == 0, month == 0, year == 0]):
            year, month, day = 1900, 1, 1
            warn('Invalid date encountered ({}-{}-{}), setting it to '
                 '1900-01-01.'.format(year, month, day))
```

cbrnr, 2020-09-21:
Since MNE 0.21 now requires Python >= 3.6, you could also use an f-string:

```python
            warn(f'Invalid date ({year:04d}-{month:02d}-{day:02d}), setting '
                 'it to 1900-01-01.')
```

agramfort, 2020-09-21:
+1 for a PR but it should deal with any date formatting pb not just 0
placeholders

>

cbrnr, 2020-09-21:
Agreed, but this just means checking if the values are within valid ranges, because we already convert to int after reading from the header. So I guess checking `1 <= day <= 31`, `1 <= month <= 12`, and `year >= 1900` (or some other year) would be sufficient, or do you want a true check for a valid date (e.g. is February 29 2017 valid)?

agramfort, 2020-09-21:
I would use a try trying to construct a datetime

wolfgang-ganglberger, 2020-09-21:
Thanks @agramfort  and @cbrnr! Super awesome!

PR #8283 (2020-09-22): Gracefully handle invalid EDF dates
Fixes #8281.
closed 2020-09-22 by larsoner