mne-tools/mne-python #8290: MNE Python Plotting for Time Series Data: Issue with mne.read_epochs_eeglab

agquinn, 2020-09-22:
**READ THIS FIRST:** If you are having trouble getting MNE-Python to work with
your own data, you should probably ask for help on our
[Gitter channel](https://gitter.im/mne-tools/mne-python) or on our
[mailing list](https://mail.nmr.mgh.harvard.edu/mailman/listinfo/mne_analysis)
before opening an issue; you will get help faster that way. See
[the FAQ entry on filing bug reports](https://mne.tools/dev/overview/faq.html#i-think-i-found-a-bug-what-do-i-do)
for more guidance. If you're pretty sure your problem is a bug in MNE-Python,
**delete this section** and fill in the headings below, replacing the
placeholder text with your own responses. Surround any code samples with triple
backticks above and below the code block, or put code in a
[public gist](https://gist.github.com) and link to it in the issue.

#### Describe the bug
〈There seems to be an issue with mne.read_epochs_eeglab modifying data slightly. When using MNE to plot time series data for the purpose of comparison to time series data in Matlab, the underlying data used in MNE is different than the actual data in Matlab. I plotted my time series data using Matplotlib, and was able to successfully reproduce the time series plot for my data in Matlab, but could not do so using MNE python, due to this issue.〉

#### Steps to reproduce
〈In Matlab, to plot the time series data for analysis:

(Google drive links to Subject1-csd_epo.set and Subject1-csd_epo.fdt files) 
https://drive.google.com/file/d/1DmPQ_jSFkK2gnkMpz17ieitZjYhWSTe6/view?usp=sharing
https://drive.google.com/file/d/1qkvSuqNyTrj-6yhaXXP-iL3d6qI3gZIM/view?usp=sharing

EEG = pop_loadset('Subject1-csd_epo.set');
plot(EEG.times, EEG.data(5, :, 1));

In MNE Python to plot the time series data:
import mne
import matplotlib.pyplot as plt
epochs = mne.read_epochs_eeglab('Subject1-csd_epo.set')
epochs.load_data()
plt.plot(epochs.times, epochs._data[0, 5])
plt.show()

In Matplotlib:
plt.plot(epochs._data[0, 0, :])
plt.show()〉

[MWE]: https://en.wikipedia.org/wiki/Minimal_Working_Example
[built-in datasets]: https://mne.tools/dev/overview/datasets_index.html
[sample]: https://mne.tools/dev/overview/datasets_index.html#sample

#### Expected results
〈I expected the time series data plotted in Matlab, for subject 1, channel 5 (the C3 electrode channel), and epoch 1 to match the time series data plotted using MNE Python.〉

#### Actual results
〈I had the following time series plots using Matlab, Matplotlib and MNE Python, and they do not match. There is an issue between the underlying data MNE python used to create the time series plots, and the underlying data used in Matlab and in Matplotlib:

[image]
[image]
[image]
)

#### Additional information
〈Replace this text with the output of `mne.sys_info()`.〉

agquinn, 2020-09-22:
@alexrockhill

agramfort, 2020-09-22:
if you want channel 5 in python you need to do:

plt.plot(epochs.times, epochs._data[0, 4])

not

plt.plot(epochs.times, epochs._data[0, 5])

>

agquinn, 2020-09-23:
Thank you; I missed that earlier. I think that might be the issue I was actually having.

closed 2020-09-23 by agquinn