mne-tools/mne-python #8291: Create local topographic representations. Don't connect distant sensors in topomap.

rob-luke, 2020-09-22:
#### Describe the problem
When creating topoplots for montages that have clearly separated clusters of sensors the plotting connects these regions, often creating some visual distortion between regions of interest.

For example in the fNIRS example [here](https://mne.tools/stable/auto_tutorials/preprocessing/plot_70_fnirs_processing.html#compare-tapping-of-left-and-right-hands) I would like the left and right regions repeated. Here is what I would like removed.

[image]

And here is another example with a dataset with 4 seperate regions that I would prefer plotted independently

[image]

#### Describe your solution
I am not sure of the optimal solution, but my initial thoughts are I would like to be able to either specify how many independent regions I want plotted. Or perhaps specify a maximum distance between sensors, and anything above this distance is not connected in the plot. 

#### Additional context
I tried varying the existing parameters in the topo plot function, but nothing quite achieves my aim.

agramfort, 2020-09-23:
I would not use kmeans but
https://docs.scipy.org/doc/scipy/reference/generated/scipy.cluster.hierarchy.linkage.html
with single option to find the clusters here.

>

rob-luke, 2020-09-27:
Thanks for the pointers. As this seems like it will involve digging in to the lower level code I will put this on hold till I have some more time.

In the meantime, if anyone lands here with the same issue I have got around this by plotting each ROI independently and specifying the same axes. 

This is a different dataset and analysis than above, but it ends up looking like 
[image]

Using something like...
```python
fig, axes = plt.subplots(nrows=1, ncols=1, figsize=(6, 6))
for r in rois:
    raw_subset = raw().pick(picks= rois[r])
    your_plotting_function(raw_subset, axes=axes)
```

larsoner, 2020-09-27:
You can probably pass `outlines=dict()` or something similar to get it to not re-plot the outlines in the subsequent calls, too. I say we make sure we can avoid re-plotting the head outline somehow, then close, as this is actually a nice solution! +1 for adding this to an fNIRS tutorial at some point, too.

rob-luke, 2020-09-28:
Nice tip! I will try that out and update the fNIRS tutorial to demo it. Thanks mate