mne-tools/mne-python #8292: persyst annotations error

infinitejest, 2020-09-23:
**READ THIS FIRST:** If you are having trouble getting MNE-Python to work with
your own data, you should probably ask for help on our
[Gitter channel](https://gitter.im/mne-tools/mne-python) or on our
[mailing list](https://mail.nmr.mgh.harvard.edu/mailman/listinfo/mne_analysis)
before opening an issue; you will get help faster that way. See
[the FAQ entry on filing bug reports](https://mne.tools/dev/overview/faq.html#i-think-i-found-a-bug-what-do-i-do)
for more guidance. If you're pretty sure your problem is a bug in MNE-Python,
**delete this section** and fill in the headings below, replacing the
placeholder text with your own responses. Surround any code samples with triple
backticks above and below the code block, or put code in a
[public gist](https://gist.github.com) and link to it in the issue.

#### Describe the bug
The persyst raw file function is able to open the persyst files and annotations with no issues. The issue I am having. For my particular use case we need to get all annotations regardless if they are repeated. When some of the EEG systems generate annoations they use xlevents or xlspikes and sometimes they can be multiple of these in the files. however the parser is only taking unique annotations. So if a human annotator or machine annotator uses same annotations then the parser is not picking up all of the annotations but rather unique names or the first one with that name. 

#### Steps to reproduce
Open any lay/dat files that have annotations with different times and same name/description

[MWE]: https://en.wikipedia.org/wiki/Minimal_Working_Example
[built-in datasets]: https://mne.tools/dev/overview/datasets_index.html
[sample]: https://mne.tools/dev/overview/datasets_index.html#sample

#### Expected results
The expected behavior is that the parser should be able to pick up all annotations even if they have the same name. 

#### Actual results
only picks up the first annotation with that name. 

#### Additional information
If I have time I can try to submit a suggestion on how to fix it.

agramfort, 2020-09-23:
ping @adam2392

adam2392, 2020-09-23:
Yes that totally sounds like a bug to me :P.

> #### Additional information
> If I have time I can try to submit a suggestion on how to fix it.
> 
> I think the issue is that you are using the comment texts as keys for your dictionary of annotations. So when your script finds a new comment with the same text is overwriting the previous one. Perhaps one suggestions to append the values to your keys as a list if the keys are already in the dictionary.

Ah I see this sounds like a good fix to me! Would you be willing to submit a PR and I can help review and get it merged? It seems like a few line fix to adapt the dictionary to dictionary of lists.

infinitejest, 2020-09-23:
I was able to get that fixed , but it is breaking the part of the code that
adds annotation to the annotation object because It is expecting single
value Instead of a list. I need to give this some thought.

Andres Rodriguez

infinitejest, 2020-09-26:
> Yes that totally sounds like a bug to me :P.
>
>
> Ah I see this sounds like a good fix to me! Would you be willing to submit a PR and I can help review and get it merged? It seems like a few line fix [...]

I think i got issue resolved . Now there will be objects with same description. I needed to change some of the way you unpacked you annotations to account for the lists of annotation. This way now wach annotation has an object. An alternative way to do it is to pass the whole list to a repeated annotation element but was not sure if that will be more desirable. One thing I dont know if need fixing is that the now when you transform annotations to events it has trouble to recognize the repeated annotation instances. Let see how I can show you the code and you can test.

adam2392, 2020-09-26:
@infinitejest  awesome! can you make the PR into actual mne-python? I think the current one you have is to your local copy.

infinitejest, 2020-09-26:
Check if it went through. I am not very good with git.