mne-tools/mne-python #8300: Reorder channels stopped working when preceded with add_reference_channels

mmagnuski, 2020-09-23:
I have a file reader for some of my files, which worked well some time ago but fails on master when trying to reorder channels:

```python-traceback

  File "C:\src\DiamSar\__init__.py", line 140, in read_raw
    raw.reorder_channels(chan_ord)

  File "c:\src\mne-python\mne\channels\channels.py", line 874, in reorder_channels
    return self._pick_drop_channels(idx)

  File "c:\src\mne-python\mne\channels\channels.py", line 938, in _pick_drop_channels
    self._read_picks = [r[idx] for r in self._read_picks]

  File "c:\src\mne-python\mne\channels\channels.py", line 938, in <listcomp>
    self._read_picks = [r[idx] for r in self._read_picks]

IndexError: index 63 is out of bounds for axis 0 with size 63
```
it seems that the problem arises because in my reader first a reference channel is added (`mne.add_reference_channels(raw, 'Cz', copy=False)`) and then `raw.reorder_channels()` is called. I can change the order of operations in my code, but thought I should let you know. :)

agramfort, 2020-09-24:
please share a full code snippet to replicate this

closed 2020-09-25 by larsoner

nbara, 2020-11-04:
I'm still having this issue on latest `develop`. 

```python
>> raw.reorder_channels(raw.ch_names)
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "~/mne-python/mne/channels/channels.py", line 874, in reorder_channels
    return self._pick_drop_channels(idx)
  File "~/mne-python/mne/channels/channels.py", line 938, in _pick_drop_channels
    self._read_picks = [r[idx] for r in self._read_picks]
  File "~/mne-python/mne/channels/channels.py", line 938, in <listcomp>
    self._read_picks = [r[idx] for r in self._read_picks]
IndexError: index 12 is out of bounds for axis 0 with size 12
```
Weirdly enough, I wanted to save the `raw` object to a fif file to share to you, and noticed the problem does not affect the newly saved file...  I do a whole bunch of operations to the `raw` instance before this happens, but I can't exactly pinpoint which is the culprit. 

Any idea how to go about this? 

I have traced it back in debug mode to these lines:
https://github.com/mne-tools/mne-python/blob/0e080a5a645b5934445ef4da6da744d7dfa7f68d/mne/channels/channels.py#L937-L938

There's something wrong with `self._read_picks`, which contains a duplicate list: 

[image]

larsoner, 2020-11-04:
We plan to release 0.21.1 this week, and the fix is in there. You can either wait for that and upgrade, or use the latest `master` branch

nbara, 2020-11-04:
Sorry I meant I'm using the latest `master` (not the pip version) in my previous post.

larsoner, 2020-11-04:
Which commit are you on? You need the fixes from #8303

nbara, 2020-11-04:
I litterally pulled an hour ago.

larsoner, 2020-11-04:
Okay, if you can share the original file plus a script that reproduces the issue we can take a look. It's best if you try to cut down the script to as few operations as possible. Things like filtering and such should not matter, but adding / removing / concatenating / loading data probably will.

nbara, 2020-11-04:
I will. Like I said, I'm doing a whole bunch of things before reordering channels, so it was not obvious to come up with a MWE. Will investigate more and let you know.

nbara, 2020-11-04:
@larsoner I am not super familiar with the `Raw` internals. Am I looking for the operation that causes `raw._pick_reads` to contain a duplicate list ? 

[image]

larsoner, 2020-11-04:
@nbara I'm not sure what to look for in the error until I can reproduce it locally and dig into it a bit. This section of code isn't the easiest to debug or immediately know what to do

nbara, 2020-11-04:
Can you confirm the bug with attached `.fif` file (zipped to bypass github restrictions)?

```python
raw = mne.io.read_raw_fif('/Users/nicolas/Desktop/test-raw.fif', preload=True)
raw = mne.add_reference_channels(raw, ['stim0', 'stim1', 'stim2', 'stim3'])
raw.reorder_channels(raw.ch_names[::-1])
```

[test-raw.fif.zip](https://github.com/mne-tools/mne-python/files/5489792/test-raw.fif.zip)

larsoner, 2020-11-04:
Yes and also without that particular file:
```
import mne
raw = mne.io.read_raw_fif(mne.datasets.testing.data_path() + '/MEG/sample/sample_audvis_trunc_raw.fif').crop(0, 1).load_data()
raw = mne.add_reference_channels(raw, ['stim0', 'stim1', 'stim2', 'stim3'])
raw.reorder_channels(raw.ch_names[::-1])
```
so the problem is really that `add_reference_channels` somehow does not update the picks I think. I'll look into it!

PR #8303 (2020-09-25): MRG, BUG: Fix add_reference_channels
Closes #8300

@mmagnuski can you see if this fixes your problem?