mne-tools/mne-python #8302: Error related to test_config_linux while trying to build documentation locally on Mac

danieltomasz, 2020-09-24:
While trying to build the documentation locally I got following error
MNE 21.0
MacOS Catalina 
```
generating gallery for auto_examples/preprocessing... [100%] plot_xdawn_denoising.py
WARNING: /Users/daniel/Downloads/mne-python-master/examples/visualization/plot_meg_sensors.py failed to execute correctly: Traceback (most recent call last):
  File "/Users/daniel/Downloads/mne-python-master/examples/visualization/plot_meg_sensors.py", line 31, in <module>
    'Magnes 3600wh': read_raw_bti(op.join(bti_path, 'test_pdf_linux'),
  File "<decorator-gen-175>", line 21, in read_raw_bti
  File "/Users/daniel/.pyenv/versions/3.8.5/lib/python3.8/site-packages/mne/io/bti/bti.py", line 1281, in read_raw_bti
    return RawBTi(pdf_fname, config_fname=config_fname,
  File "<decorator-gen-174>", line 21, in __init__
  File "/Users/daniel/.pyenv/versions/3.8.5/lib/python3.8/site-packages/mne/io/bti/bti.py", line 947, in __init__
    info, bti_info = _get_bti_info(
  File "/Users/daniel/.pyenv/versions/3.8.5/lib/python3.8/site-packages/mne/io/bti/bti.py", line 1054, in _get_bti_info
    raise ValueError('Could not find the config file %s. Please check'
ValueError: Could not find the config file /Users/daniel/.pyenv/versions/3.8.5/lib/python3.8/site-packages/mne/io/bti/tests/data/test_config_linux. Please check whether you are in the right directory or pass the full name
```

agramfort, 2020-09-24:
what command did you type in the terminal?

>

danieltomasz, 2020-09-24:
`make html_stable`

agramfort, 2020-09-25:
you run this from the doc folder?

>

danieltomasz, 2020-09-25:
Yes, everything was running almost smoothly till this moment 
(apart from one script from preprocessing folder which hangs on downloading one of the brainstorm datasets, but I solved this manually by running the same script manually, which downloaded this dataset)

agramfort, 2020-09-25:
ok I think I understand. The file test_config_linux is a data file that we
don't include in the source distribution and that
is not installed. Can you try to install mne with

cd mne-python
pip install -e .

danieltomasz, 2020-09-25:
After  local mne install, It didn't crash at this point, albeit the execution of the code seems stuck at this point ( 'generating gallery for auto_examples/visualization... [ 52%] plot_meg_sensors.py') more than 5min now

agramfort, 2020-09-25:
FYI building the entire doc is close to 2h ... :-/

>

danieltomasz, 2020-09-25:
EDIT: after over  10min it breaks through!  Thanks for your help, I am closing this issue!

closed 2020-09-25 by danieltomasz

agramfort, 2020-09-25:
it breaks?

>

danieltomasz, 2020-09-25:
No, it's good :) It finished generating the problematic script and start another one, thanks for the help again!