mne-tools/mne-python #8305: IS it a correct definition of PLV?

dahaiyu, 2020-09-24:
In the following page, I found the PLV definition is given as follows: PLV = |E[Sxy/|Sxy|]|. It is different from the definition in the reference. Are these two equivalent?

[https://mne.tools/stable/generated/mne.connectivity.spectral_connectivity.html?highlight=spectral_connectivity#mne.connectivity.spectral_connectivity](url)

agramfort, 2020-09-25:
yes it is equilavent

>

agramfort, 2020-09-25:
This appears to be a question about usage, not a bug report or a new feature request. Please use the [MNE mailing list](http://mail.nmr.mgh.harvard.edu/mailman/listinfo/mne_analysis) or the [gitter channel](https://gitter.im/mne-tools/mne-python) for usage questions.

closed 2020-09-25 by agramfort