mne-tools/mne-python #8309: Error: mri_make_bem_surfaces has been removed from this version of freesurfer

RickLicona, 2020-09-26:
Hi,

I was running the Python script "01-anatomy.py" of the GitHub project named mne-biomag-group-demo. However, the console throw the next message errors: 

[image]
[image]

I have the MNE version 0.20.8 and 7.1.1 in the case of Freesurfer. Should I update my MNE version (i.e., 0.21.0) or replace some code lines with one equivalent command of Freesurfer (https://surfer.nmr.mgh.harvard.edu/fswiki/FreeSurferCommands)? 

Best.

agramfort, 2020-09-26:
hum this is bad as this command is required by MNE

@buildqa @mshamalainen we need to see how to proceed as it seems freesurfer
broke mne flash bem :(

>

hoechenberger, 2020-09-26:
Interestingly this change does not seem to be listed in the release notes at https://surfer.nmr.mgh.harvard.edu/fswiki/ReleaseNotes

RickLicona, 2020-09-29:
It seems that I need to download an old version of Free Surfer. Can you suggest me the correct version to avoid this error message?. 

Thanks!

RickLicona, 2020-09-29:
Update: It is the same error with Free Surfer 7.1.0. Thus, 7.1.1 and 7.1.0 versions provided the same error message:
```
Processing sub001
  Skipping reconstruction (folder exists)
  Making BEM
('error: mri_make_bem_surfaces has been removed from this version of freesurfer. For further information, please post to the help list at freesurfer@nmr.mgh.harvard.edu\n', '')
```

```
-------- freesurfer-darwin-macOS-7.1.0-20200511-813297b --------
Setting up environment for FreeSurfer/FS-FAST (and FSL)
FREESURFER_HOME   /Applications/freesurfer/7.1.0
FSFAST_HOME       /Applications/freesurfer/7.1.0/fsfast
FSF_OUTPUT_FORMAT nii.gz
SUBJECTS_DIR      /Applications/freesurfer/7.1.0/subjects
MNI_DIR           /Applications/freesurfer/7.1.0/mni

```

hoechenberger, 2020-09-29:
You will need to use FreeSurfer 6 for now, which can be downloaded from https://surfer.nmr.mgh.harvard.edu/fswiki/rel6downloads

RickLicona, 2020-09-29:
> You will need to use FreeSurfer 6 for now, which can be downloaded from https://surfer.nmr.mgh.harvard.edu/fswiki/rel6downloads

Thanks! Now it is working. To avoid some problems in the case of **Mac Os X** , it is neccesary to **add a patch (https://drive.google.com/file/d/1ezjiNIbGNrNAiYD5ciYxMO2SK-R5vz9A/view)** for the **FreeSurfer 6**.  

See more regarding Mac Os X patch: https://www.mail-archive.com/freesurfer@nmr.mgh.harvard.edu/msg64333.html

closed 2020-09-29 by RickLicona

agramfort, 2020-09-30:
FYI we contacted freesurfer core dev team to try to fix this issue

cc @buildqa

thanks for the detailed report

>

agramfort, 2020-10-06:
FYI mri_make_bem_surfaces has been put back on the freesurfer source tree so it should be fixed in the next release.

thanks @RickLicona for reporting the issue.

jhouck, 2022-08-18:
FWIW mri_make_bem_surfaces is missing from freesurfer's Ubuntu 18 build, but support for that build is likely to be phased out soon.  Still present for all newer builds. See https://www.mail-archive.com/freesurfer@nmr.mgh.harvard.edu/msg73261.html