mne-tools/mne-python #8322: Add ICA.pca_explained_variance_ratio_

hoechenberger, 2020-09-30:
#### Describe the problem
When using `ICA.max_pca_components` to reduce data dimensionality before ICA, after fitting we have `ICA.pca_explained_variance_` containing the absolute variances of all retained principal components. However, it's impossible to reconstruct how much of the total variance **before** dimensionality reduction (i.e., leaving out PCs) each PC explained.

#### Describe your solution
The `_PCA` object has an attribute `explained_variance_ratio_`, which holds the individual value explained variance relative to **all** principal components, before leaving some out. We should expose these values after an ICA fit as `ICA.pca_explained_variance_ratio_`, so users can find out post-hoc just how much variance of the original data their PCA/ICA approach actually explains.

#### Describe possible alternatives
We could retain the variances of **all** components, not only those which were kept. But I don't like this idea: If I decide to keep 30 PCs, I want the `ICA.pca_explained_variance_` vector to be precisely 30 elements long.

larsoner, 2020-09-30:
Why not just adjust `pca_explained_variance_` such that it reflects the variance of the original data? To me this would qualify as a bug fix. The other quantity (what is currently given) is just `pca_explained_variance_ / pca_explained_variance_.sum()`, right?

hoechenberger, 2020-09-30:
> Why not just adjust `pca_explained_variance_` such that it reflects the variance of the original data?

You mean, for **all** components in the original data, including those that were not retained for ICA?

larsoner, 2020-09-30:
I mean let's say you have 30 channels but `max_pca_components=28`, `pca_explained_variance_` should be length 28 (it's how many PCs were kept) but the variance numbers correspond to those of the original data. So it will not sum to 1 (unless the last two PCs had variance zero in this example).

hoechenberger, 2020-09-30:
Oh but that's exactly my point with this issue: `pca_explained_variance_` contains absolute variance values. So if all you get is a subset, you don't know how much variance there was **in total**, so you cannot calculate which fraction of total variance each PC explains.

hoechenberger, 2020-09-30:
Therefore I'm suggesting to introduce `pca_explained_variance_ratio_`, where for each PC you'd have their relative contribution to total variance.

larsoner, 2020-09-30:
Ahh, I see. Personally I don't see too much value in the absolute variances anyway (?), I'd be up for just making them relative.

Another alternative would be to add `ica.pca_residual_variance_`. It gets you what you need to get the ratios but is less redundant than / more compact than the ratios I think.

hoechenberger, 2020-09-30:
> Ahh, I see. Personally I don't see too much value in the absolute variances anyway (?), I'd be up for just making them relative.

if that's an option, I'd be totally up for it. I always thought that absolute variances are kind of odd but then again what do **I** know about other users' needs :)

larsoner, 2020-09-30:
Given that we don't know if users are doing something with the absolute values, I think adding `ica.pca_residual_variance_` is my favorite option at this point...

hoechenberger, 2020-09-30:
Residual variance in this case being the relative variance not explained by the retained components, or…? I have the feeling this is over-the-top complicated? Users would always have to do `1-residual` to get to the figure they actually want?

larsoner, 2020-09-30:
No I'm saying keep `pca_explained_variance_` the way it is, don't add `pca_explained_variance_ratio_`. If people want this normalized quantity (which apparently they haven't until now?) it's just:
```
pca_explained_variance / (pca_explained_variance_ + pca_residual_variance)
```

hoechenberger, 2020-09-30:
Okay, got it. sounds like a plan.

hoechenberger, 2020-09-30:
> If people want this normalized quantity (which apparently they haven't until now?)

Well, I want it ;)

larsoner, 2020-09-30:
We could even make `pca_residual_variance_` an array of the remaining (unused) components. Then the expression above does not get much more complicated:
```
ratio = ica.pca_explained_variance_ / (ica.pca_explained_variance_ + ica.pca_residual_variance_.sum())
```
but you can for example decide how many more PCA components you should have kept by looking at individual terms of the `ica.pca_residual_variance_`.

hoechenberger, 2020-09-30:
> We could even make `pca_residual_variance_` an array of the remaining (unused) components.

Yes this is how I (obviously mis-)understood your proposal in the first place! 👍👍👍

agramfort, 2020-09-30:
just expose the explained_variance_ratio_ from sklearn PCA estimator

>