mne-tools/mne-python #8325: BUG?: ICA can blow up components

larsoner, 2020-09-30:
@hoechenberger not 100% sure this is related to #8313 and #8316 since those seem to have to do with ICA taking too long rather than doing something totally wrong. To me this code should work / be a no-op:
```
import numpy as np
import mne

data = np.random.RandomState(0).randn(10, 100000)
info = mne.create_info(10, 1000., 'eeg')
raw = mne.io.RawArray(data, info)
raw.set_eeg_reference()
assert np.linalg.matrix_rank(raw.get_data()) == 9

ica = mne.preprocessing.ICA(
    max_pca_components=10, n_pca_components=10, max_iter=1, verbose=True)
raw_new = ica.fit(raw).apply(raw.copy())
np.testing.assert_allclose(raw.get_data(), raw_new.get_data())
```
gives an `AssertionError`:
```
Mismatched elements: 1000000 / 1000000 (100%)
Max absolute difference: 4.53435389
Max relative difference: 70047.35032413
 x: array([[ 1.481445,  0.391694,  1.081697, ...,  0.370815, -1.556122,
        -1.275782],
       [-0.766405,  1.279594, -0.02692 , ...,  0.312448, -1.409486,...
 y: array([[ 6.455399e-05,  6.455399e-05,  6.455399e-05, ...,  6.455399e-05,
         6.455399e-05,  6.455399e-05],
       [ 3.581095e-03,  3.581095e-03,  3.581095e-03, ...,  3.581095e-03,...
```
It seems like this should work even if you get the number of components wrong. It suggests that something is wrong with the mixing/unmixing step that when the number of components equals the number of channels you don't get an identity transform here.

It also fails for `max_pca_components=10, n_pca_components=9` but passes for any value of `max_pca_components` (8, 9, 10) so long as `max_pca_components=9.

If I remove this call:
```
        self.unmixing_matrix_ /= np.sqrt(
            self.pca_explained_variance_[sel])[None, :]  # whitening
```
Then things work okay.

hoechenberger, 2020-09-30:
> @hoechenberger not 100% sure this is related to #8313 and #8316
...
> It also fails for `max_pca_components=10, n_pca_components=9` but passes for any value of `max_pca_components` (8, 9, 10) so long as `max_pca_components=9`.

I believe this is related, and in fact may be the explanation to super bad ICA results @agramfort and I witnessed when working with rank-deficient data (i.e., without setting `max_pca_components` to an appropriate value)

Reducing the rank always fixed things for us.

This is why I started working on #8321, to make it easier to select a relative number of components (explaining x% of variance).

hoechenberger, 2020-09-30:
I believe the problem occurs not during ICA itself but when reconstructing the signal

> . It suggests that something is wrong with the mixing/unmixing step that when the number of components equals the number of channels

I think it's really that your data is rank-deficient here thanks to the avg reference.

hoechenberger, 2020-10-01:
> I think it's really because your data is rank-deficient here thanks to the avg reference.

```python
import numpy as np
import mne

data = np.random.RandomState(0).randn(10, 100000)
info = mne.create_info(10, 1000., 'eeg')
raw = mne.io.RawArray(data, info)
# raw.set_eeg_reference()
# assert np.linalg.matrix_rank(raw.get_data()) == 9
assert np.linalg.matrix_rank(raw.get_data()) == 10

ica = mne.preprocessing.ICA(
    max_pca_components=10, n_pca_components=10, max_iter=1, verbose=True)
raw_new = ica.fit(raw).apply(raw.copy())
np.testing.assert_allclose(raw.get_data(), raw_new.get_data())
```
Commented-out the avg reference step, and the assertion holds…

larsoner, 2020-10-01:
> Commented-out the avg reference step, and the assertion holds…

Yes it's 100% because it's rank deficient. But my point is that if we can make it so that this *doesn't* blow up the data without any special intervention (such as trying to detect the rank), we should. That's what #8326 does.

hoechenberger, 2020-10-01:
Ok, got it

PR #8326 (2020-10-04): MRG, FIX: Test round-trip, fix EEGLAB ICA
Closes #8325 
Closes #8308
Closes #7657
closed 2020-10-04 by hoechenberger