mne-tools/mne-python #8332: Should we drop ICA.apply(..., n_pca_components)?

hoechenberger, 2020-10-02:
While looking into #8327 I discovered that `ICA.apply()` accepts an `n_pca_components` kwarg
https://github.com/mne-tools/mne-python/blob/74f6f41f827df552d258c4db4d4c99a5b71863d6/mne/preprocessing/ica.py#L1531-L1532

The idea here is obviously that the user shall be able to override the `n_pca_components` parameter passed to `ICA` on initialization.

However, the current documentation doesn't match the code:
https://github.com/mne-tools/mne-python/blob/74f6f41f827df552d258c4db4d4c99a5b71863d6/mne/preprocessing/ica.py#L1551-L1554

This is not true, because if `None` is passed, we will fall back to `self.n_pca_components`, i.e. the value passed on `ICA` instantiation.

Now of course one could fix the docstring accordingly: "If None, use the value passed on instantiation". But then, how would you select **all** components? The only way would be via `n_pca_components=1.0`, but then this API is kinda inconsistent with `ICA.__init__()`… So I'm wondering, how important is it actually to keep the parameter anyway? Could it be removed?

hoechenberger, 2020-10-02:
… or we could keep it in `apply()`, but remove it from `__init__()` …

agramfort, 2020-10-02:
apply is like transform. It should not alter the instance. makes sense?

>

hoechenberger, 2020-10-02:
> apply is like transform. It should not alter the instance. makes sense?

Yes, but that wasn't exactly my point :) I'm just wondering if anyone ever passes a different `n_pca_components` to `apply()` than they have passed to `__init__()`. But maybe the Juelich folks do :)

agramfort, 2020-10-02:
sorry I don't get what you suggest then

>

hoechenberger, 2020-10-02:
Ok let me restart:

- `ICA.__init__()` accepts `n_pca_components`, which is then stored in `self.n_pca_components`
    - If `n_pca_components is None`, we use all available components for signal reconstruction when invoking `apply()`
- but `ICA.apply()` **also** accepts an `n_pca_components` param
    - if this one `is not None`, it overrides `self.n_pca_components` (and currently actually **overwrites** it, but this will need to be addressed in #8327)
    - if it `is None`, `self.n_pca_components` is used (that's the default)

I find the API slightly inconsistent in the sense that `n_pca_components=None` passed to `__init__()` means "use all components"; whereas `n_pca_components=None` passed to `apply` means "use what was passed to `__init__()`"

But maybe it's not a problem, idk. Just irritated me.

The docstring of `apply()` will have to be updated in any case, because it claims `None` means "all components", which is not true.

agramfort, 2020-10-02:
+1 we have a bug. The self.n_pca_components should not be changed with
apply. It should just be used provided it's less than max_pca_components

>

hoechenberger, 2020-10-02:
> +1 we have a bug

Yep, that's being tracked in #8327

hoechenberger, 2020-10-02:
Pinging @jdammers, you probably have a strong opinion on this one 🧐🙂

jdammers, 2020-10-02:
@hoechenberger thanks for pinging me!
At the time of coding the ICA API (thanks to Denis Engemann) the motivation for the parameter n_pca_components was to have an additional setscrew for further denoising. Apart from some test it was rarely used. Finally, in real data analysis nobody is using it in Juelich.  
So if no one else is using it, why not removing the parameter?

larsoner, 2020-10-02:
In this case, +1 for removing `n_pca_components` and keep just `n_components` and `max_pca_components`. I think we really only need two of these.

If we really want to keep the option of additional denoising, we can still allow overriding `n_pca_components` in `apply`, so long as the `n_pca_components` there is less than or equal to the one used during init it should be fine. `max_pca_components` seems to currently be serving that role, but it's confusing.

hoechenberger, 2020-10-02:
@jdammers 
> Apart from some test it was rarely used. Finally, in real data analysis nobody is using it in Juelich.

Thanks for getting back so swiftly! This really seems to indicate we could (should) try to deprecate this kwarg. It has caused so much confusion 😅

Have a great weekend!

hoechenberger, 2020-10-02:
> In this case, +1 for removing `n_pca_components` and keep just `n_components` and `max_pca_components`. I think we really only need two of these three.
> 
> If we really want to keep the option of additional denoising, we can still allow overriding `n_pca_components` in `apply`, so long as the `n_pca_components` there is less than or equal to the one used during init it should be fine. `max_pca_components` seems to currently be serving that role, but it's confusing.

Sounds perfect.

jdammers, 2020-10-02:
Sounds good to me!

closed 2020-10-05 by hoechenberger