mne-tools/mne-python #8338: Error: Surface outer skull is not completely inside surface outer skin

RickLicona, 2020-10-03:
Dear all,

Regarding the DS117 dataset. When I employ the **conductivity for 3 layers (EEG)**,  I got the next error (subject 006):
[image]
[image]

However, employing the same subject, but with conductivity for 1 layer (MEG), there is not any error:

[image]

In the case of the subject 002, there is not problem with **conductivity for 3 layers (EEG)**:

[image]
[image]

I only require to use the EEG data (**I know that the 3-layer is unreliable**). Is there any solution for the first error message? or do you suggest me only take the users such as 002?. The idea is to continue with the forward estimation and the inverse stepts.  

By the way, thanks for your proposal MNE pipeline ;).

hoechenberger, 2020-10-03:
@RickLicona Can you try running this with MNE 0.21, which was just recently released? Which version of FreeSurfer are you using?

@agramfort Any idea what might be going on here?

RickLicona, 2020-10-03:
> @RickLicona Can you try running this with MNE 0.21, which was just recently released? Which version of FreeSurfer are you using?
> 
> @agramfort Any idea what might be going on here?

I am using MNE 0.21.0 and FreeSurfer 6.0 (due to problems with the most recent versions)

agramfort, 2020-10-04:
I think that I recall that for this dataset we only used a single layer BEM
to do source loc on MEG only.

Now I think you have the flash MRI images which should give much better BEM
surfaces.

is that what you tried? maybe @jasmainak remembers more as he did most of
the analysis for this paper.

>

RickLicona, 2020-10-04:
> I think that I recall that for this dataset we only used a single layer BEM to do source loc on MEG only. Now I think you have the flash MRI images which should give much better BEM surfaces. is [...]

Yes, that is right, only employed a single layer for MEG (here there is not any problem). However, I changed it for 3 layers (EEG) `conductivity = (0.3, 0.006, 0.3)  # for three layer (EEG)`. I want to use only EEG data. For the moment I could generate the 3-layer bem surfaces for the users 2, 14 and 18. For the rest of users, I got the error message: _**Surface outer skull is not completely inside surface outer skin**._ 

There is a comment in the project (12-make_forward.py) that said`# Here we only use 1-layer BEM because the 3-layer is unreliable`.
```
    # Here we only use 1-layer BEM because the 3-layer is unreliable
    fname_bem = op.join(subjects_dir, subject, 'bem',
                        '%s-5120-bem-sol.fif' % subject)
```
 Thus, I guess that for this dataset is more recommendable to use MEG data rather than EEG data. It makes sense because there are problems for generating the 3-layer bem surfaces in the majority of the subjects.

agramfort, 2020-10-05:
that's right

>

jasmainak, 2020-10-07:
If you want to use 3 layer BEM you should look at the advice here:

https://mne.tools/stable/overview/faq.html?highlight=mri_watershed#my-watershed-bem-meshes-look-incorrect

and also verify that the BEM looks good by plotting it