mne-tools/mne-python #8346: FR: add apply_function method to epochs and evoked objects

sappelhoff, 2020-10-06:
based on [gitter thread](https://gitter.im/mne-tools/mne-python?at=5f7b310b17377c6b5ddcf5a3). I don't have the time to make a PR right now, but I think an issue to track this feature is a good idea.

# problem

I want to modify the data stored in epochs in some arbitrary way, so currently I use:

- get the data `X = epochs.get_data()`
- modify it `X = ...`, for example making a matrix multiplication with a covariance matrix of shape (nchs, nchs)
- reassign it `epochs._data = X`

but this is unreliable, because the `._data` attribute is private and not guaranteed to work this way in all cases.

# solution

Adding an [`apply_function`](https://mne.tools/dev/generated/mne.io.RawArray.html#mne.io.RawArray.apply_function) method like in `raw` objects to `epochs` and `evoked` would solve this problem.

larsoner, 2020-10-06:
+1 sounds good to me, any volunteer to try implementing?

(@sappelhoff until it's implemented you can indeed do `epochs._data[:] = X` assuming dtype is the same -- the `[:]` ensures you don't mess up the shape at least. But as you say it's not guaranteed to work permanently so we should fix this properly concurrently.)

vpeterson, 2021-03-15:
I can take this

PR #9088 (2021-03-19): [ENH]: add apply_function method to epochs and evoked objects
#### Reference issue
Fixes #8346.

#### What does this implement/fix?
first proposed version on how apply_function can be added to epochs and evoked objects. At the moment, just implementing the function to epochs objects. This follows the apply_function implemented in raw objects.

#### Additional information
CC with @nordme
closed 2021-03-19 by drammock