mne-tools/mne-python #8349: Why delete then edf_info['physical_min'] and edf_info['digital_min'] in mne-python/mne/io/edf/edf.py

ddbbqqpp, 2020-10-07:
# physical dimension in µV
    edf_info['offsets'] = (
        edf_info['physical_min'] - edf_info['digital_min'] * edf_info['cal'])
    del edf_info['physical_min']
    del edf_info['digital_min']

ddbbqqpp, 2020-10-07:
What's the use of variables edf_info['cal'] and edf_info['offsets'] in /mne/io/edf/edf.py

cbrnr, 2020-10-07:
That's a good question, I actually don't know why `edf_info['physical_min']` and `edf_info['digital_min']` are deleted. In fact, their counterparts `edf_info['physical_max']` and `edf_info['digital_max']` remain in the dict.

Note that `_get_info` is a private function, and so is the attribute `raw._raw_extras`. This means that they are not meant for public use, so the simple answer to why these keys are deleted is that we don't need them in our internal functions. In fact, we try to map all EDF-specific information to the `Info` dict accessible as `raw.info`. Information regarding channel units and scaling can be found in `raw.info["chs"]`, so if you think that something is missing there let us know.

larsoner, 2020-10-07:
> What's the use of variables edf_info['cal'] and edf_info['offsets'] in /mne/io/edf/edf.py

https://github.com/mne-tools/mne-python/blob/master/mne/io/edf/edf.py#L249-L251
and
https://github.com/mne-tools/mne-python/blob/master/mne/io/edf/edf.py#L293-L295

> That's a good question, I actually don't know why edf_info['physical_min'] and edf_info['digital_min'] are deleted.

It makes things like `print(edf_info.keys())` easier to parse when debugging and also tell us internally that we shouldn't need / don't use these values anymore (they have been converted to some other variables that we do use). 

> Note that _get_info is a private function, and so is the attribute raw._raw_extras

Indeed, don't rely on these even existing as a `dict` or any other fixed type, let alone having fixed keys. If these are needed for something, we should expose them publicly somehow.

cbrnr, 2020-10-08:
> It makes things like `print(edf_info.keys())` easier to parse when debugging and also tell us internally that we shouldn't need / don't use these values anymore (they have been converted to some other variables that we do use).

I was just wondering where these other variables are if someone needs to know the physical/digital minimum values - any idea? Maybe they can be recovered, but I couldn't figure it out quickly because all we have are values that depend on these minimums/maximums, e.g. `cal = (phys_max - phys_min) / (dig_max - dig_min)` and `offsets = phys_min - dig_min * cal`. I'm not saying that we need to keep `phys_min` and `phys_max`, but if they are useful to someone then maybe it's not worth discarding them just because the dict gets more compact.

larsoner, 2020-10-08:
> I'm not saying that we need to keep phys_min and phys_max, but if they are useful to someone then maybe it's not worth discarding them just because the dict gets more compact.

I disagree -- this is private. My philosophy (and IIUC the design philosophy of MNE-Python in general) is that from the development side we should have no problem changing private attributes when it makes maintaining code better/easier for us, and we should provide public attributes for things when people need them and there is a good reason to have them (i.e., a compelling use case). And from the user side, users should not rely on these values ever being the same type, value, name, etc., and we should not suggest to them that they be used.

Assuming we're not changing that design philosophy, let's shift the discussion -- do we need to make the physical min/max stuff public? I don't see a reason to so far. My assumption was that that this issue is more a point of curiousity about the format and how we interact with it than it is about needing these values for some reason. @ddbbqqpp do you need these values for something?

cbrnr, 2020-10-08:
> Assuming we're not changing that design philosophy, let's shift the discussion -- do we need to make the physical min/max stuff public? I don't see a reason to so far. My assumption was that that [...]

Maybe I wasn't very clear, but that's what I meant. I wasn't challenging the private nature of "private" values. I wanted to point out that if people needed access to private values (i.e. if they provide a unique value), then we should consider making these values public.

ddbbqqpp, 2020-10-09:
In fact, I need to show a more detailed information about of EDF signals, Like https://www.teuniz.net/edfbrowser/ **File Info**
But, I can't get more Header Infos about of EDF signals, Through public methods @cbrnr @larsoner 
```
raw = mne.io.read_raw_edf(input_fname=fpath)
print(raw.info.keys())
```
Output:
`dict_keys(['acq_pars', 'acq_stim', 'ctf_head_t', 'description', 'dev_ctf_t', 'dig', 'experimenter', 'utc_offset', 'device_info', 'file_id', 'highpass', 'hpi_subsystem', 'kit_system_id', 'helium_info', 'line_freq', 'lowpass', 'meas_date', 'meas_id', 'proj_id', 'proj_name', 'subject_info', 'xplotter_layout', 'gantry_angle', 'bads', 'chs', 'comps', 'events', 'hpi_meas', 'hpi_results', 'projs', 'proc_history', 'custom_ref_applied', 'sfreq', 'dev_head_t', 'ch_names', 'nchan'])`

I think we should include as much header information as possible

agramfort, 2020-10-09:
can you list information that you cannot recover from public API so maybe
we add it? basically EDF is not specific it is just one format we aim to
expose in a consistent way.

>

cbrnr, 2020-10-09:
Adding to @agramfort's comment, if you need access to the header info you can already do that by using a private function:

```
import mne
mne.io.edf.edf._read_edf_header(edf_fname, [])
```

If that is not enough, you could try [PyEDFlib](https://pyedflib.readthedocs.io/en/latest/):

```
import pyedflib
f = pyedflib.EdfReader(edf_fname)
f.getHeader()
f.getSignalHeaders()
# f has many more methods that expose EDF properties
```

BTW, EDFbrowser uses the same underlying lib.

ddbbqqpp, 2020-10-09:
Can you remove these?
https://github.com/mne-tools/mne-python/blob/af671e78561d7eb9e05eb69170149142ae336855/mne/io/edf/edf.py#L488-L489

cbrnr, 2020-10-09:
Fine with me - what about you @agramfort and @larsoner?

cbrnr, 2020-10-09:
On the other hand, even if we do not delete these values they will still be in a private dict. Therefore, you could use the private function `_read_edf_header` I mentioned before to extract this information.

larsoner, 2020-10-09:
> Can you remove these?

I don't think we should, for the reasons listed above.

> if you need access to the header info you can already do that by using a private function:

>  Therefore, you could use the private function _read_edf_header I mentioned before to extract this information.

@cbrnr please any time you advise someone to use a private attribute or function append the reminder/warning that "do not expect this to work, it can change at anytime" or something similar, and make clear that it should only be a temporary solution to their problem. I know we mentioned it earlier in this thread and it may seem redundant, but someone getting these results from Google or quickly browsing the responses might not understand the instability/unsupported-ness of the approach you're suggesting.

In this case to me one solution is probably to make `read_edf_header` properly public and return a dict with all information. Then internally when we use it we can delete keys as we want.

drammock, 2020-10-09:
> In this case to me one solution is probably to make `read_edf_header` properly public and return a dict with all information

I don't object to that in principle, but I wonder why we should do the work to support a new public function, when as @cbrnr mentioned above there is an existing package (pyedflib) that provides the desired functionality. Why isn't that good enough?

cbrnr, 2020-10-09:
@larsoner I've edited my comment and included a note about the private nature of the function (I assumed it was clear because we've mentioned it in this thread before).

@drammock exactly, that's why I made two suggestions that solve the issue.

agramfort, 2020-10-10:
I think we offered solutions to the problem and that it does not require to
change mne code.

>

ddbbqqpp, 2020-10-10:
I think the return value of the function should be consistent，At least in terms of version compatibility. When you upgrade, You find that some variables are missing. It's frustrating. Maybe I can't convince you to change the code.I just want to return some necessary information through a function, I don't understand why I should delete some values @larsoner @drammock 
```
    def info(self, fpath):
        '''获取EDF文件信息

        Parameters
        ----------
        fpath : str
            文件路径

        Returns
        -------
        info : mne.io.meas_info.Info
            EDF测量信息
        edf_info : dict
            几乎所有的EDF文件头信息
        orig_units : dict
            通道对应的物理单位
        '''
        try:
            # EDF文件信息
            info, edf_info, orig_units = mne.io.edf.edf._get_info(fname=fpath, stim_channel='auto',
                                                                  eog=None, misc=None,
                                                                  exclude=(), preload=False)
            return info, edf_info, orig_units

        except Exception as e:
            return None, None, None
```

ddbbqqpp, 2020-10-10:
At least in the last version they exist. I also used them. Suddenly it's gone in, Just upgrade to the latest version

cbrnr, 2020-10-10:
Which function do you mean? What existed in 0.20 and doesn't exist in 0.21?

ddbbqqpp, 2020-10-10:
This is the difference between the two versions. I mean, can the original dict be kept, not deleted, Although they are private

```
tuple_info = Signal().info(fpath=file_path)  # 获取EDF文件信息
    if not tuple_info[0]:
        return jsonify(WM.DataJSON(100, '无法获取EDF文件信息'))
data_json = {}
info, edf_info, orig_units = tuple_info  # 获取EDF文件信息
# MNE v0.21.0
# line 488 del edf_info['physical_min']
# line 489 del edf_info['digital_min']
print(info)
print(edf_info)
print(orig_units)
```

**Disparity:**
units (data type Changed)
digital_min, physical_min(Deleted)
cal, offsets(New)

**0.20.8**

<Info | 7 non-empty values
 bads: []
 ch_names: F3, F4, C3, C4, M1, M2, O1, O2, E1, E2, Chin 1, Chin 2, Chin 3, ...
 chs: 39 EEG
 custom_ref_applied: False
 highpass: 0.0 Hz
 lowpass: 512.0 Hz
 meas_date: 2018-06-21 21:21:11 UTC
 nchan: 39
 projs: []
 sfreq: 1024.0 Hz
\>

{'events': [], 'units': [1e-06, 1e-06, 1e-06, 1e-06, 1e-06, 1e-06, 1e-06, 1e-06, 1e-06, 1e-06, 0.001, 0.001, 0.001, 0.001, 0.001, 0.001, 0.001, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1], 'ch_names': ['F3', 'F4', 'C3', 'C4', 'M1', 'M2', 'O1', 'O2', 'E1', 'E2', 'Chin 1', 'Chin 2', 'Chin 3', 'RIP ECG', 'Leg/L', 'Leg/R', 'Therm', 'Resp Imp', 'Thor', 'Abdo', 'Nasal Pressure', 'Light', 'PositionSen', 'Snore', 'SpO2', 'Pulse', 'Ox Status', 'Pleth', 'CPAP Press', 'CPAP Flow', 'EtCO2', 'EtCO2-W', 'Leak', 'TidalVol', 'Derived HR', 'RespRate', 'ManPosition', 'ManLights', 'Sum'], 'data_offset': 10240, 'digital_max': array([3.2767e+04, 3.2767e+04, 3.2767e+04, 3.2767e+04, 3.2767e+04,
       3.2767e+04, 3.2767e+04, 3.2767e+04, 3.2767e+04, 3.2767e+04,
       3.2767e+04, 3.2767e+04, 3.2767e+04, 3.2767e+04, 3.2767e+04,
       3.2767e+04, 3.2767e+04, 3.2767e+04, 3.2767e+04, 3.2767e+04,
       3.2767e+04, 1.0000e+00, 4.0000e+00, 3.2767e+04, 3.2767e+04,
       3.2767e+04, 2.0000e+00, 3.2767e+04, 3.2767e+04, 3.2767e+04,
       3.2767e+04, 3.2767e+04, 3.2767e+04, 3.2767e+04, 3.2767e+04,
       3.2767e+04, 3.0000e+00, 1.0000e+00, 3.2767e+04]), 'digital_min': array([-32768., -32768., -32768., -32768., -32768., -32768., -32768.,
       -32768., -32768., -32768., -32768., -32768., -32768., -32768.,
       -32768., -32768., -32768., -32768., -32768., -32768., -32768.,
            0.,      0., -32768., -32768., -32768.,      0., -32768.,
       -32768., -32768., -32768., -32768., -32768., -32768., -32768.,
       -32768.,      0.,      0., -32768.]), 'highpass': array([], dtype=float64), 'sel': array([ 0,  1,  2,  3,  4,  5,  6,  7,  8,  9, 10, 11, 12, 13, 14, 15, 16,
       17, 18, 19, 20, 21, 22, 23, 24, 25, 26, 27, 28, 29, 30, 31, 32, 33,
       34, 35, 36, 37, 38]), 'lowpass': array([], dtype=float64), 'meas_date': datetime.datetime(2018, 6, 21, 21, 21, 11, tzinfo=datetime.timezone.utc), 'n_records': 31245, 'n_samps': array([ 256,  256,  256,  256,  256,  256,  256,  256,  256,  256,  256,
        256,  256, 1024,  256,  256,   32,   32,   32,   32,  256,   32,
         16,  256,   16,   16,   16,  256,   16,   32,   16,   32,   32,
         16,  512,  512,   64,   64,   32]), 'nchan': 39, 'subject_info': {'id': '???', 'name': ''}, 'physical_max': array([1.00e+03, 1.00e+03, 1.00e+03, 1.00e+03, 1.00e+03, 1.00e+03,
       1.00e+03, 1.00e+03, 1.00e+03, 1.00e+03, 1.00e+00, 1.00e+00,
       1.00e+00, 2.50e+00, 1.00e+00, 1.00e+00, 1.00e+01, 1.00e+00,
       1.00e+00, 1.00e+00, 1.00e+01, 1.00e+00, 4.00e+00, 1.00e+00,
       1.00e+02, 3.00e+02, 2.00e+00, 1.00e+00, 3.00e+01, 1.25e+00,
       1.00e+02, 1.00e+02, 9.00e+01, 1.00e+03, 5.00e+02, 3.00e+01,
       3.00e+00, 1.00e+00, 5.00e-01]), 'physical_min': array([-1.00e+03, -1.00e+03, -1.00e+03, -1.00e+03, -1.00e+03, -1.00e+03,
       -1.00e+03, -1.00e+03, -1.00e+03, -1.00e+03, -1.00e+00, -1.00e+00,
       -1.00e+00, -2.50e+00, -1.00e+00, -1.00e+00, -1.00e+01, -1.00e+00,
       -1.00e+00, -1.00e+00, -1.00e+01,  0.00e+00,  0.00e+00, -1.00e+00,
        0.00e+00,  0.00e+00,  0.00e+00, -1.00e+00,  0.00e+00, -1.25e+00,
        0.00e+00,  0.00e+00,  0.00e+00,  0.00e+00,  0.00e+00,  1.00e+01,
        0.00e+00,  0.00e+00, -5.00e-01]), 'record_length': array([1., 1.]), 'subtype': 'edf', 'tal_idx': array([], dtype=int64), 'dtype_byte': 2, 'dtype_np': <class 'numpy.int16'>, 'stim_channel': None, 'max_samp': 1024, 'nsamples': 31994880}

{'F3': 'uV', 'F4': 'uV', 'C3': 'uV', 'C4': 'uV', 'M1': 'uV', 'M2': 'uV', 'O1': 'uV', 'O2': 'uV', 'E1': 'uV', 'E2': 'uV', 'Chin 1': 'mV', 'Chin 2': 'mV', 'Chin 3': 'mV', 'RIP ECG': 'mV', 'Leg/L': 'mV', 'Leg/R': 'mV', 'Therm': 'mV', 'Resp Imp': 'V', 'Thor': 'V', 'Abdo': 'V', 'Nasal Pressure': 'V', 'Light': '', 'PositionSen': '', 'Snore': 'V', 'SpO2': '%', 'Pulse': 'bpm', 'Ox Status': '', 'Pleth': 'V', 'CPAP Press': 'cmH2O', 'CPAP Flow': 'V', 'EtCO2': 'mmHg', 'EtCO2-W': 'mmHg', 'Leak': 'L/min', 'TidalVol': 'ml', 'Derived HR': 'bpm', 'RespRate': 'BPM', 'ManPosition': '', 'ManLights': '', 'Sum': 'V'}

Traceback (most recent call last):
[... 42 lines cut ...]
       3.05180438e-02, 3.05180438e-02, 3.05180438e-02, 3.05180438e-02,
       3.05180438e-02, 3.05180438e-02, 3.05180438e-05, 3.05180438e-05,
       3.05180438e-05, 7.62951095e-05, 3.05180438e-05, 3.05180438e-05,
       3.05180438e-04, 3.05180438e-05, 3.05180438e-05, 3.05180438e-05,
       3.05180438e-04, 1.00000000e+00, 1.00000000e+00, 3.05180438e-05,
       1.52590219e-03, 4.57770657e-03, 1.00000000e+00, 3.05180438e-05,
       4.57770657e-04, 3.81475547e-05, 1.52590219e-03, 1.52590219e-03,
[... cut ...] Nasal Pressure': 'V', 'Light': '', 'PositionSen': '', 'Snore': 'V', 'SpO2': '%', 'Pulse': 'bpm', 'Ox Status': '', 'Pleth': 'V', 'CPAP Press': 'cmH2O', 'CPAP Flow': 'V', 'EtCO2': 'mmHg', 'EtCO2-W': 'mmHg', 'Leak': 'L/min', 'TidalVol': 'ml', 'Derived HR': 'bpm', 'RespRate': 'BPM', 'ManPosition': '', 'ManLights': '', 'Sum': 'V'}

Traceback (most recent call last):
  File "./app/routes/file.py", line 105, in file
    data_json['digital_min'] = edf_info['digital_min'].tolist()
KeyError: 'digital_min'

ddbbqqpp, 2020-10-10:
only need to read the EDF once to get the information I want，It takes two times now. The intermediate variable I want to get was deleted

cbrnr, 2020-10-10:
Got it, I wasn't aware that this changed in 0.21 - apparently @larsoner deleted them in #8210. I still don't see a reason why we can't keep these values around if they are clearly useful, but I'll leave this up to @larsoner to decide.

larsoner, 2020-10-10:
>  When you upgrade, You find that some variables are missing. It's frustrating

>  I mean, can the original dict be kept, not deleted, Although they are private

>  I still don't see a reason why we can't keep these values around if they are clearly useful

I apologize for the frustration -- we try hard to ensure our software to be stable release to release. Based on these responses I think an important point is still being missed, though. Perhaps we should be clearer about this in the documentation somewhere with a statement like: "If you use private attributes, you should expect your code to break from version to version without any warning. If you want something stable, you need to stick to public functions or attributes."

In this light, statements like the three above are inconsistent with our policies and practices. This is because putting the values back without making the function public does not solve the problem of and frustration you've experienced about "I want this property available to me in a way that my code will work." Private attribute stability is not guaranteed under the MNE API contract, and just putting the attribute back -- which I agree is a trivial, two-line change! -- is thus *not actually a solution to the problem you're having*, unless we 1) make the function public, or 2) make a special exception that "in MNE private attributes can be changed at any time without warning, except for the dict you get from mne.io.edf._read_edf_info, we need to leave the key-value pairs stable" (which is a no-go).

Concretely @ddbbqqpp I would suggest:

1. Try the `pyedflib` lines @cbrnr [suggested above](https://github.com/mne-tools/mne-python/issues/8349#issuecomment-706021070) -- if these work for your use case then we're all set because there is already a public API you can use
2. If the `pyedflib` solution is not a good one (e.g., extra dependency, etc.) then we can work toward making the EDF info reading function public and its return value more stable.

(2) incurs some additional maintenance burden from our end, so I'd prefer (1) if possible.

larsoner, 2020-11-03:
FYI @ddbbqqpp it looks like @klacourse has good reasons to avoid `pyedflib` -- it's not flexible enough about the files it handles (they must conform to the spec pretty strictly). So I'm +1 on solution (2).

@klacourse do you want to try making a PR to make `_get_info` into a public `mne.io.read_edf_info` function or so?

klacourse, 2020-11-03:
I am not comfortable doing it. My experience in writing robust and bug-free python code is limited.  I still need practice!

larsoner, 2020-11-03:
> My experience in writing robust and bug-free python code is limited. I still need practice!

I would encourage you not to let this stop you -- PRs are a great way to learn better techniques. Experienced contributors / maintainers / reviewers will be happy to guide you on the process. The bonus of social coding is that it's not 100% the responsibility of the person writing the PR to write good code -- the reviewers equally share in the responsibility, and everyone benefits. But if you are unconvinced by this reasoning, I'm of course happy to open a PR instead!

klacourse, 2020-11-03:
OK.  I can try to make a PR, but what I really want is to read an edf with mne and write it back corrected and standardized with `pedflib`.
Therefore, the best would be to get all the fields specified in the edf specification (https://www.edfplus.info/specs/edf.html) and 
 call from `pedflib` : `EdfWriter.setHeader` (https://pyedflib.readthedocs.io/en/latest/_modules/pyedflib/edfwriter.html#EdfWriter.setHeader), `setSignalHeader` (https://pyedflib.readthedocs.io/en/latest/_modules/pyedflib/edfwriter.html#EdfWriter.setSignalHeader) and `setDatarecordDuration` (https://pyedflib.readthedocs.io/en/latest/_modules/pyedflib/edfwriter.html#EdfWriter.setDatarecordDuration)
 Would you agree with these changes?  Do you see any limitations?

larsoner, 2020-11-03:
> what I really want is to read an edf with mne and write it back corrected and standardized with pedflib.

FYI you might get this for free with MNELAB's routines. We have discussed before the possibility of writing formats other than FIF but so far have decided not to, but IIRC MNELAB allows you to do so.

> Therefore, the best would be to get all the fields specified in the edf specification

I would start just by taking our internal function `_get_info` and seeing how much of the stuff you need it already has. If the dict it returns is 90% of what you need, then we can figure out what we need to add to it (or not remove from it, which was what started this GitHub issue originally).

There are also probably some fields that are derived from the EDF header that we might want to factor out of the public function, but we can figure that part out later.

So +1 for you opening a PR to 1) rename `_get_info` to `read_edf_info`, 2) change our uses of `_get_info` to call this new public `read_edf_info`, and 3) modify `read_edf_info` to have the fields you actually need. If this allows you to get the info you need for your use case, then we can sort out the rest of the details later.

cbrnr, 2020-11-03:
Yes, see here:
https://github.com/cbrnr/mnelab/blob/master/mnelab/io/writers.py#L46-L89

If you take this as a starting point I'd be grateful if I could reuse any improvements you might make on the way.

klacourse, 2020-11-06:
@cbrnr thank you for your function.

I did modify `read_edf_info` from a fork.

I modified the prefiltering field in the edf_info because the last channel was dropped.
I added the transducer information into the edf_info dict.

Since I had issues reading one of my edf file (the data looks corrupted in mne but it is read properly in EDF browser and Polyman) and I realized that all the channels are re-sampled to the highest sampling rate I'm not sure to use those modifications anymore.

I plan to write a simple library to read and modify the edf header only (and keep the whole data set as it is) as edfBrowser allows it. 

Do you think it worth trying a PR for #8349 anyyway?

larsoner, 2020-11-06:
> I plan to write a simple library to read and modify the edf header only (and keep the whole data set as it is) as edfBrowser allows it.
>
> Do you think it worth trying a PR for #8349 anyyway?

Nice! Would having a public `read_edf_info` help you with this your reading and modification plan?

ddbbqqpp, 2020-11-08:
Thanks! @larsoner @klacourse Too important to me! With MNE-Python, Why use other ...

klacourse, 2020-11-13:
@larsoner 
> Nice! Would having a public `read_edf_info` help you with this your reading and modification plan?

No, I started from scratch (by looking at what you did in mne).  I'm happy with my personal edf library for reading/writing the edf header, and I will use MNE for the EEG signal processing.  Thanks!

closed 2020-11-13 by larsoner