mne-tools/mne-python #8353: BUG: 1. and 1 treated differently in ICA

larsoner, 2020-10-08:
From #8351.

My proposed solution is to disallow `1.`. We have `None` for "select all components", so we don't actually need `1.`.

larsoner, 2020-10-08:
cc @hoechenberger @cbrnr

agramfort, 2020-10-08:
+1 to disallow 1.

>

hoechenberger, 2020-10-08:
+1 to disallow

hoechenberger, 2020-10-08:
To be more precise, I think we should disallow 1 as int, and allow 1.0 as float. 

If we decide to keep allowing 1 as int, 1.0 float should not be allowed.

larsoner, 2020-10-08:
I don't think we should allow `1.` (float), I'm okay with `1` as an int -- maybe there is some use case for it

larsoner, 2020-10-08:
Okay let's just disallow both. ICA is a no-op when you set `n_components=1` so it doesn't really make sense

agramfort, 2020-10-08:
using 1 component does not make sense ! the mixing matrix is a float and
scale is not "estimable" in ICA so....

>

cbrnr, 2020-10-08:
Yes, let's either disallow both or allow both (and then both 1 and 1.0 refer to all components because one component doesn't make sense).

hoechenberger, 2020-10-08:
Uh please don't allow both then, for the sake of API consistency

cbrnr, 2020-10-08:
I'm fine with disallowing both, but IMO `1 == 1.0` is consistent.

PR #8351 (2020-10-12): MRG, API: Deprecate max_pca_components
Todo:

- [x] Deprecate `max_pca_components`
- [x] Get `test_ica.py` working
- [x] Add `n_pca_components` to `plot_ica_overlay`
- [x] Make keyword-only after #8350 is in
- [x] Update `plot_40_artifact_correction_ica.py`
- [x] Add warning when `n_components` leads to unstable pseudoinversion
- [x] Update `doc/_static/diagrams/ica.dot`
- [x] Decide if `n_components` as float selects based on the variance remaining of the `n_pca_components` components. This is how `max_pca_components` worked (it was relative to that)

Closes #8337
Closes #8342
Closes #8327
Cl [... cut]
closed 2020-10-12 by hoechenberger