mne-tools/mne-python #8355: Remove n_pca_components from ICA.__init__

cbrnr, 2020-10-12:
Following up the changes in #8351, I noted that because `n_pca_components` is only used in `ICA.apply` it should be removed from `ICA.__init__`. This would make ICA usage a lot easier and clearer, because right now one of the questions I frequently need to answer is about the two parameters `n_components` and `n_pca_components`. I'd prefer to have only `n_components` in `ICA.__init__` and `n_pca_components` only in `ICA.apply`. @hoechenberger @larsoner @agramfort

hoechenberger, 2020-10-12:
x-ref https://github.com/mne-tools/mne-python/issues/8337#issuecomment-703244103

hoechenberger, 2020-10-12:
> I noted that because `n_pca_components` is only used in `ICA.apply` it should be removed from `ICA.__init__`. This would make ICA usage a lot easier and clearer

I don't feel strongly here, but if you believe my removing the param from `__init__()` we can make MNE easier to use, I'm +1 for it!

larsoner, 2020-10-12:
@cbrnr did you look at the updated docs for these parameters in the PR?

https://22659-1301584-gh.circle-artifacts.com/0/dev/generated/mne.preprocessing.ICA.html#mne.preprocessing.ICA

I tried to make it clearer that `n_pca_components` in `__init__` is really now just a *default* PCA reconstruction value. See the link @hoechenberger posted above for some rationale, but the TL;DR is that keeping it the way it is after #8351 allows you to set a "default" amount of PCA denoising. I can live with it either way...

cbrnr, 2020-10-12:
> I tried to make it clearer that n_pca_components in __init__ is really now just a default PCA reconstruction value. See the link @hoechenberger posted above for some rationale, but the TL;DR is that [...]

Yes, I saw that, and it's definitely clearer than before. However, many users don't read the whole docstring (even though they should, but that's my point), so not having this parameter would benefit usability. And BTW, having this parameter in `__init__` to provide a default is also not a good argument, because we already have that default directly in `apply`.

larsoner, 2020-10-12:
> And BTW, having this parameter in __init__ to provide a default is also not a good argument, because we already have that default directly in apply.

It sounds like you're talking about the default value of the `n_pca_components` being the same (`None`) in `__init__` and `apply`? That's not quite the idea. The point is if the user wants to change it from this `None` default to some other default for their apply operations for their instance. For example if you have 10 channels and do:
```
ica = ICA(n_pca_components=8)
```
Then when you do `ica.apply(raw)` (even with a save-load round trip of the ICA instance in between) it will use 8 components for this operation because there was `n_pca_components=8` in the `__init__`, unless you override this instance's default like `ica.appy(raw, n_pca_components=9)` or whatever. This is the sense in which it "provides a default".

Make more sense?

hoechenberger, 2020-10-12:
After having a coffee, I'm now +1.5 for dropping the argument from init! Unless **I** would have to do it haha

cbrnr, 2020-10-12:
@larsoner I get the idea, but I don't see why providing this in `__init__` is useful. It seems that if they want e.g. `n_pca_components=8` in all subsequent `apply` calls all this does is save a few characters, because they can use `.apply()` instead of `.apply(n_pca_components=8)`, right? Usually, I'd apply this only once, but even if I wanted to play around with some values I think that a clearer API (i.e. no `n_pca_components` arg in `__init__`) would be worth the extra characters in repeated `apply` calls.

@hoechenberger I didn't have any recent coffee and it's too late for today, so I'm not sure how caffeine would change my view on this 🤣 .

hoechenberger, 2020-10-12:
@cbrnr never mind, I deleted this comment again, as it was posted under the influence of transient caffeine intoxication 😅💩

PR #8356 (2020-10-14): MRG, API: Deprecate n_pca_components in init
Closes #8355
closed 2020-10-14 by larsoner